node-opcua/node-opcua #1216: "build_address_space_for_conformance_testing is not a function" when trying to run the simple_server sample/demo

jacobq, 2022-11-21:
**Current behavior**

I am trying to run the demo/sample application on a Raspberry Pi with node 14.x per the instructions [in the `README.md`](https://github.com/node-opcua/node-opcua#installing-node-opcua-samples-as-a-node-package). However, when I run the `simple_server` I get an error stating that "`build_address_space_for_conformance_testing` is not a function."  

```
# node -v; npm -v; uname -s -r -v -m -o
v14.15.1
6.14.8
Linux 5.10.63-v8+ #1488 SMP PREEMPT Thu Nov 18 16:16:16 GMT 2021 aarch64 GNU/Linux
# jq .version node_modules/node-opcua/package.json
"2.84.0"
# node --trace-warnings --unhandled-rejections=warn ./node_modules/.bin/simple_server
FQDN =  <hostname of this device>
(node:94276) UnhandledPromiseRejectionWarning: TypeError: build_address_space_for_conformance_testing is not a function
    at post_initialize (/root/node-opcua-demo/node_modules/node-opcua-samples/bin/simple_server.js:239:9)
    at main (/root/node-opcua-demo/node_modules/node-opcua-samples/bin/simple_server.js:443:5)
    at emitUnhandledRejectionWarning (internal/process/promises.js:168:15)
    at processPromiseRejections (internal/process/promises.js:224:9)
    at processTicksAndRejections (internal/process/task_queues.js:94:32)
(node:94276) TypeError: build_address_space_for_conformance_testing is not a function
    at post_initialize (/root/node-opcua-demo/node_modules/node-opcua-samples/bin/simple_server.js:239:9)
    at main (/root/node-opcua-demo/node_modules/node-opcua-samples/bin/simple_server.js:443:5)
```

**Describe the bug**

It looks like the sample server is [trying to import a function from the main `node-opcua` module that is not there](https://github.com/node-opcua/node-opcua/blob/3c7743b73ccb9e24f80638b08e304b6b46f74379/packages/node-opcua-samples/bin/simple_server.js#L24). Perhaps it used to exposed as part of that interface and then was later moved? I'm not entirely sure what to make of the commented-out lines at the end of `packages/node-opcua/source/server-stuff.ts`. Should they be commented-out? Perhaps caused by commit 214d4724c0eb6b322b3f39de07985514ca5dccbe?

In any case, the sample app no longer runs.

```
> Object.keys(require('node-opcua')).filter(x => x.match(/address/i))
[
  'generateAddressSpace',
  'FilterContextOnAddressSpace',
  'AddressSpace',
  'resolveOpaqueOnAddressSpace',
  'generateAddressSpaceRaw'
]
> Object.keys(require('node-opcua-address-space-for-conformance-testing')).filter(x => x.match(/address/i))
[ 'build_address_space_for_conformance_testing' ]
```

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

Follow the same steps from the [documentation here](In any case, the sample app no longer works.):

1. `mkdir myserver`
2. `cd myserver`
3. `npm init`
4. `npm install node-opcua-samples --unsafe-perms`
5. `node ./node_modules/.bin/simple_server` 

**Expected behavior**

Server should start rather than terminate with uncaught exception. 

**Context**

~~- ( ) my request is related to node-opcua acting as an OPCUA CLIENT~~
- (X) my request is related to node-opcua acting as an OPCUA SERVER

~~- ( ) I have installed node-opcua from source ( using git clone)~~
- (X) I have installed node-opcua as a package ( using npm install )
~~- ( ) I am using an application that uses node-opcua~~

    ~~- ( ) node-red~~
    ~~- ( ) other : please specify~~

- Device: Raspberry Pi 4 Model B w/ 2GB of RAM
- OS version: 
```
# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

    ~~- ( ) Windows : version : ______~~
    ~~- ( ) Linux : version : ______~~
    ~~- ( ) MacOs : version : ______~~
    - (X) Raspbian: version : Bullseye (64-bit)
    ~~- ( ) Other : specify : ______~~

- Description of the other OPCUA system I am trying to connect to: N/A

- **node-opcua version:** :

`v2.84.0`

- **Node:**

  `node --version` = `v14.15.1`

jacobq, 2022-11-21:
Looks like there is a similar issue for `install_optional_cpu_and_memory_usage_node` (i.e., you'll see it if you correct this first problem)

erossignon, 2022-11-24:
It occurs that the install_optional_cpu_and_memory_usage_node  and build_address_space_for_conformance_testing have be intentioanaly removed from the node-opcua packages. 

the samples need to be fixed instead.

closed 2022-11-24 by erossignon