node-opcua/node-opcua #1219: OPC client connection error

top58525785, 2022-11-24:
Hello  OPCUA Team:

I am using another  OPCUA Server,  and using Node -opcua client to connect,but failed

and error as bellow
08:27:01.698Z :client_base_impl :534 Internal error : re-entrancy in createDefaultCertificate!
2022-11-19T08:27:01.700Z [ERR] 'opc' connect failure! Error: Cannot find private key C:\Users\Administrator\AppData\Roaming\node-opcua-default-nodejs\Config\PKI\own\private\private_key.pem

So how to deal with the error  Thanks

AndreasHeine, 2022-11-24:
you can use openssl to create your own cert and key

top58525785, 2022-11-24:
Thanks AndreasHeine 
Is there any info about Openssl how to create
Thanks again

AndreasHeine, 2022-11-24:
ssl.conf
```
[ req ]
default_bits = 2048
default_md = sha256
distinguished_name = subject
req_extensions = req_ext
x509_extensions = req_ext
string_mask = utf8only
prompt = no

[ req_ext ]
basicConstraints = CA:FALSE
nsCertType = client, server
keyUsage = nonRepudiation, digitalSignature, keyEncipherment, dataEncipherment, keyCertSign
extendedKeyUsage= serverAuth, clientAuth
nsComment = "OpenSSL Generated Certificat"
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid,issuer
subjectAltName = URI:urn:opcua:server,IP: 127.0.0.1

[ subject ]
countryName = DE
stateOrProvinceName = HE
localityName = HE
organizationName = AndreasHeine
commonName = OpcUaServer
```

Step 1: Change ssl.conf (subjectAltname, country, organizationName, ...)
Step 2: `openssl genrsa -out key.pem 2048`
Step 3: `openssl req -x509 -days 365 -new -out certificate.pem -key key.pem -config ssl.conf`

erossignon, 2022-11-24:
alternatively  node-opcua-pki provides  an easy way to generate PKI, certificates and private

erossignon, 2022-11-25:
@top58525785!: can you provide the source code of your client for review? I guess something is wrong here.

NodeOPCUA creates automatically a default certificate when a client is first started.

In your case there is a Internal error : re-entrancy in createDefaultCertificate!  which we would need to investigate....

top58525785, 2022-11-25:
Thanks erosignon
I just using [node-opcua](https://github.com/node-opcua/node-opcua) library to connect
Now the erros shows as bellow
2022-11-25T03:10:41.873Z [INF]  socket.io client connected
2022-11-25T03:10:44.342Z [ERR]  'test' connect failure! TypeError: Cannot read properties of undefined (reading 'position')
2022-11-25T03:10:44.342Z [ERR]  'test' try to connect error! TypeError: Cannot read properties of undefined (reading 'position')
2022-11-25T03:10:49.343Z [ERR]  'test' connect failure! TypeError: Cannot read properties of undefined (reading 'position')
2022-11-25T03:10:49.343Z [ERR]  'test' try to connect error! TypeError: Cannot read properties of undefined (reading 'position')
2022-11-25T03:10:54.343Z [ERR]  'test' connect failure! TypeError: Cannot read properties of undefined (reading 'position')

erossignon, 2022-11-25:
There isn't enough detail here to provide a way to reproduce. 
I'll close the issue and suggest you contact sterfive for professional support.

closed 2022-11-25 by erossignon