node-opcua/node-opcua #1224: Coerce arrays

robertsLando, 2022-12-14:
What is the suggested way to coerce internal array values to JS arrays?

Actually when I add a var like this: 

```
namespace.addVariable({
			componentOf: device,
			browseName: 'ArrayVar',
			dataType: 'Double',
			value: {
				get: () =>
					new Variant({
						dataType: DataType.Double,
						arrayType: VariantArrayType.Array,
						value: [10, 15, 20]
					})
			}
		})
```

When I subscribe to it the value I get is a Float64Array that is converted to:

```
{ "0": 10, "1": 15, "2": 20 }
```

What's the proper way to convert them to arrays so?

erossignon, 2022-12-15:
Float64Array  is used by design to improve the encoding when exchanging the array between the client & server. it also provides more compact storage.

One easy way to convert back to a Javascript array is

```` 
const v = new Variant({dataType: "Float", value: [1,2,3,]);
console.log(v.value);
// => Float32Array(3) [ 1, 2, 3 ]

console.log([...v.value]);
// [ 1, 2, 3 ]

````

erossignon, 2022-12-16:
YEs, matrice are internaly stored as flat arrays anyway.
 you need to check the dimension property to get the  size of each axis.

erossignon, 2022-12-16:
> Could I suggest a toJS()
There is already a toJSON() method but very limited as it is not reversible and do not cover all cases.

I would suggest you use the dedicated node-opcua-json module to do this. as it converts the Variant or DataValue into the exact JSON OPCUA encoding as specified by OPCUA. It will take care of all Variant combinations and is capable of producing reversible JSON object ( a JS object that can be use to regenerate the exact native NodeOPCUA Variant or DataValue)

erossignon, 2022-12-16:
```
npm install node-opcua-json
```

```javascript
const { Variant, VariantArrayType } = require("node-opcua");
const { opcuaJsonEncodeVariant } = require("node-opcua-json");

const a = new Variant({
  dataType: "Float",
  arrayType: VariantArrayType.Array,
  value: [1, 2, 3],
});
console.log(opcuaJsonEncodeVariant(a, true));

const m = new Variant({
  dataType: "Float",
  arrayType: VariantArrayType.Matrix,
  value: [1, 2, 3, 4],
  dimensions: [2, 2],
});
console.log(opcuaJsonEncodeVariant(m, true));
```` 

 output:
````
{ Type: 10, Body: [ 1, 2, 3 ] }
{ Type: 10, Body: [ 1, 2, 3, 4 ], Dimensions: [ 2, 2 ] }
````

robertsLando, 2022-12-16:
Cool, by simply removing the second arg `true` I get what I want. Thanks @erossignon 🙏🏼

robertsLando, 2023-01-31:
I'm getting this error:

```
 10:02:02.615Z :extension_object              :171   WARNING => decodeExtensionObject: Extension object decoding error on  DataChangeNotification  expected size was 4294967295 but only this amount of bytes have been read : 31 
  ...                                                           encoding nodeId =  ns=0;i=811 encodingType =  1
 10:02:02.615Z :message_builder               :477   Decode message error :  The value of "offset" is out of range. It must be >= 0 and <= 101. Received 4_294_967_361
 10:02:02.615Z :message_builder               :229   cannot decode message  for valid object of type ns=0;i=829 PublishResponse
 { /*PublishResponse*/
  responseHeader                /*  ResponseHeader                            */: {
    timestamp                   /*  DateTime                                  */: 2023-01-31T11:01:15.383Z
    requestHandle               /*  UInt32                                    */: 7               0x7
    serviceResult               /*  StatusCode                                */: Good (0x00000000)
    serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
  namespaceUri                  /*  Int32                                     */: null
  symbolicId                    /*  Int32                                     */: -1
  locale                        /*  Int32                                     */: -1
  localizedText                 /*  Int32                                     */: -1
  additionalInfo                /*  String                                    */: null
  innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
  innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 };
    stringTable                 /*  String                                []  */: [ /* empty*/ ]
    additionalHeader            /*  ExtensionObject                           */: null
  }
  subscriptionId                /*  UInt32                                    */: 2               0x2
  availableSequenceNumbers      /*  UInt32                                []  */: [ /* length =1*/
    1
  ]
  moreNotifications             /*  Boolean                                   */: false
  notificationMessage           /*  NotificationMessage                       */: {
    sequenceNumber              /*  UInt32                                    */: 1               0x1
    publishTime                 /*  DateTime                                  */: 2023-01-31T11:01:15.382Z
    notificationData            /*  ExtensionObject                       []  */: [ /* length =1*/
      
      { /*DataChangeNotification*/
       monitoredItems                /*  MonitoredItemNotification             []  */: [
         { /* 0 - MonitoredItemNotification*/
           clientHandle              /*  UInt32                                    */: 1               0x1
           value                     /*  DataValue                                 */: { /* DataValue */
         value: Variant(Scalar<Boolean>, value: false)
         statusCode:      Good (0x00000000)
         serverTimestamp: 2023-01-31T11:01:15.383Z $ 000.000.000
         sourceTimestamp: 2023-01-31T11:01:15.383Z $ 000.000.000
      }
         }
       ]
       diagnosticInfos               /*  DiagnosticInfo                        []  */: [ /* empty*/ ]
      };
    ]
  }
  results                       /*  StatusCode                            []  */: [ /* empty*/ ]
  diagnosticInfos               /*  DiagnosticInfo                        []  */: [ /* empty*/ ]
 };
```

Not sure if it's caused by coerce or something else but I have no idea what to look for as there is no other context

robertsLando, 2023-02-03:
@erossignon Kindly ping

erossignon, 2023-02-03:
Please reopen an different issue with full description of how to reproduce with some sample code if possible by filling the template

closed 2023-02-03 by robertsLando