node-opcua/node-opcua #1225: Server is accessible with wrong Security Policy

SebsSa, 2022-12-16:
**Current behavior**

I created a server with :  **SecurityPolicy.Basic256Sha256** and  one **user** (abc/123).
I´m able to log on the server with user abc/123 and **NONE** (SecurityPolicy) 
Server runs with allowAnonymous: false

**Describe the bug**

I created a server with :

`            allowAnonymous: false,
        securityPolicies: [SecurityPolicy.Basic256Sha256],`
User/PW: abc/123

So i tried to connect : Via

None | Anonymous - Like expected did not work
Basic256Sha256 | Anonymous - Like expected did not work
Basic256Sha256 | abc/123 - Like expected did work

But now i tried :

**None | abc/123 - And i got logged in - expected it to NOT work**

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1.  Create Server with SecurityPolicy.Basic256Sha256 and one user ( and allowAnonymous: false)
2.  Try to connect with NONE SecurityPolicy + user
3.  You are able to browse on the server

**Expected behavior**

I should not be able to access the server if my client has NONE SecurityPolicy 

**Context**

- (x) my request is related to node-opcua acting as an OPCUA SERVER

- **node-opcua version:** :

"node-opcua": "^2.75.0",

matthiasfeurer, 2023-01-11:
Is somebody already working this?

erossignon, 2023-01-13:
No. this type of work is adressed through consulting work or under the term of NodeOCPUA Support membership. We encourage you to subscribe at https://support.sterfive.com for the paid subscription.

iliareshetov, 2023-12-17:
Hi @erossignon I'd be willing to help resolve this issue. Could you verify that it hasn't been resolved and that no one is working on it right now?

iliareshetov, 2024-01-28:
I tried to reproduced the issue with the latest version ([v2.120.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.120.0)) and got a different result. Instead of getting in, I got an error 
 > Error =  Cannot find an Endpoint matching  security mode: 2 policy: http://opcfoundation.org/UA/SecurityPolicy#None
 
**Steps taken to reproduce:**

1. Created a [server](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-samples/bin/simple_server.js) with SecurityPolicy.Basic256Sha256, allowAnonymous: false, and a user (user1/password1).
2. Attempted to connect with SecurityPolicy: None and user credentials. ```node packages/node-opcua-samples/bin/client_example.js -e opc.tcp://localhost:4840 -P=None -s=Sign -u user1 -p password1```
3. Received the error: ```Cannot find an Endpoint matching  security mode: 2 policy: http://opcfoundation.org/UA/SecurityPolicy#None```

**The server correctly rejected the connection with SecurityPolicy: None, indicating the intended behaviour.**
The underlying problem could have been with the configuration of the server rather than the node-opcua source code.

erossignon, 2024-02-02:
Thank you @iliareshetov,

 Just to make sure , can you test again with node-opcua@2.75.0 and verify that the problem is visible in that older version? 
This will confirm that the problem is fixed in the most recent version.

iliareshetov, 2024-02-02:
Hi @erossignon , I tried node-opcua@2.75.0, 
```javascript 
    const server_options = {
        allowAnonymous: false,
        securityPolicies: [SecurityPolicy.Basic256Sha256]
       ...
  }
```
```
 node simple_client.js -e opc.tcp://local:26543 -P=None -s=Sign -u user1 -p password1 🔴 
```

but got same error and was not able to connect to sever:
```
Cannot find an Endpoint matching security mode: 2 policy: http://opcfoundation.org/UA/SecurityPolicy#None
```
I'm thinking that this could not be the library's failure, but rather the server configuration. What do you think ?

Connecting with correct security policy went successfully : 
```
 node simple_client.js -e opc.tcp://local:26543 -P=Basic256Sha256 -s=Sign -u user1 -p password1 🟢 
```

erossignon, 2025-05-03:
Further investigations show that the securityModes options need to be set too. OPCUAServer works as expected. 

```
 const server = new OPCUAServer({
        serverCertificateManager,
        userManager: {
            isValidUser(username: string, password: string) {
                return true;
            }
        },
         allowAnonymous: false,
        securityPolicies: [
            SecurityPolicy.Basic256Sha256
        ],
        securityModes: [
            MessageSecurityMode.Sign,
            MessageSecurityMode.SignAndEncrypt,
        ]
    });

```

closed 2025-05-03 by erossignon

matthiasfeurer, 2025-11-19:
> [@erossignon](https://github.com/erossignon) thanks for the comments and the test script. However can you please explain why in r2 it is expected that user/pw does not work for policy "none", securityMode "none"? In other servers (like the test server from Unified Automation) this is a valid setting and works.

some answer on this would be appreciated - thanks!