node-opcua/node-opcua #1231: SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND

Chonthee, 2023-01-12:
I try to create a client side of opcua application.  I have only one monitoring group to subscribe data from opcua server  but I don't understand why this error occur. 

```
04:29:07.043Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
04:29:07.044Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  4
04:29:07.045Z :client_publish_engine         :271    => nbMaxPublishRequestsAcceptedByServer = 4
04:29:07.047Z :client_publish_engine         :269    WARNING : SERVER TELLS THAT TOO MANY PUBLISH REQUEST HAS BEEN SEND ...
04:29:07.049Z :client_publish_engine         :270    On our side nbPendingPublishRequests =  3
04:29:07.050Z :client_publish_engine         :271    => nbMaxPublishRequestsAcceptedByServer = 3
```
`
07:03:34.674Z :client_secure_channel_layer   :884   Invalid socket => Communication has been lost, cannot renew token`

Here is my code

```
const { TimestampsToReturn, OPCUAClient, ClientMonitoredItemGroup,ClientSubscription, DataType, AttributeIds, DiagnosticInfo} = require('node-opcua');
const opcuaConfig = require("../config/opcua-config");

let statusConnectOPC = false;
let itemsToMonitor = [
    {
        nodeId: "ns=2;s=Simulation.Device1.TEST1",
        attributeId: AttributeIds.Value
    },
    {
        nodeId: "ns=2;s=Simulation.Device1.TEST2",
        attributeId: AttributeIds.Value
    },
    {
        nodeId: "ns=2;s=Simulation.Device1.TEST3",
        attributeId: AttributeIds.Value
    },
    {
        nodeId: "ns=2;s=Simulation.Device1.TEST4",
        attributeId: AttributeIds.Value
    },
    {
        nodeId: "ns=2;s=Simulation.Device1.TEST5",
        attributeId: AttributeIds.Value
    },
    {
        nodeId: "ns=2;s=Channel1.Device1.TEST",
        attributeId: AttributeIds.Value
    }
];
let result = [];

let session,client;

async function initOPCUA(){
    client = await OPCUAClient.create(opcuaConfig.opcuaClientOptions);

    client.on("backoff", (number) => {
        statusConnectOPC = false;
        console.log("[opcua-client] Backoff attemp #" + number + " retrying . . . ");        
    })
    client.on("connected", () => {
        statusConnectOPC = true;
        console.log("[opcua-client] Connected successfully");
    })
    client.on("connection_reestablished", () =>{
        console.log("[opcua-client] Re-connected successfully");
    })

    await client.connect(opcuaConfig.endpointURL);
    
    session = await client.createSession();

    session.on("keepalive", (lastKnowServerState) => {
        console.log('opcua keep alive');
        console.log(lastKnowServerState);
    });

    session.on("keepalive_failure", (err) => {
        console.log(err);
    });

    const subscription = await session.createSubscription2(
        {
            maxNotificationsPerPublish: 9000,
            publishingEnabled: true,
            requestedLifetimeCount: 10,
            requestedMaxKeepAliveCount: 10,
            requestedPublishingInterval: 1000
        }
    );

    const parameters = {
        discardOldest: true,
        queueSize: 10,
        samplingInterval: 2000
    };

    const monitoringGroup = await ClientMonitoredItemGroup.create(subscription, itemsToMonitor,parameters,TimestampsToReturn.Both)
    
    monitoringGroup.on("changed",(monitoringItem,value,index)=>{
        result[index] = value.value.value
    })

    monitoringGroup.on("err", (err) =>{
        console.log("monitoring group error");
    })
}

initOPCUA();

async function WriteDataOPC(opcuaAddress,value){
    if(!statusConnectOPC&&!session){
        initOPCUA();
        return;
    }

    let nodeToWrite = {
        nodeId: opcuaAddress,
        attributeId: AttributeIds.Value,
        indexRange: null,
        value:{
            value: {
                dataType: DataType.String,
                value: value
            }
        }
    };

    try {
        await session.write(nodeToWrite, (err,statusCode,diagnosticInfo) => {
            if(!err){

            }
        })
    }
    catch(err){
        console.log(err);
    }
};

module.exports = { result, WriteDataOPC}

```

erossignon, 2023-04-11:
try not to mix async/await and callback form 

```` 
 const nodeToWrite = {
        nodeId: opcuaAddress,
        attributeId: AttributeIds.Value,
        indexRange: null,
        value:{
            value: {
                dataType: DataType.String,
                value: value
            }
        }
    };

    try {
        const statusCode = await session.write(nodeToWrite);
    }
    catch(err){
        console.log(err);
    }
````

erossignon, 2023-05-01:
closing due to inactivity

closed 2023-05-01 by erossignon