node-opcua/node-opcua #1232: Receiving ExtensionObject arrays

nakr94, 2023-01-12:
Hello,
i encountered a problem while using node-opcua as a client for receiving OPC UA messages containing ExtensionObject arrays. In particular, the received ExtensionObject array is empty
[image]
The length of the received array however is correct.

Simultaneously the client throws the exception
```
resolve_dynamic_extension_obje:75 err Error: Internal Error: we are not expecting this dataType to be processed already 
... at /snapshot/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:46:19
```

I already verified that my server modell is working as intended by viewing it with UAExpert and the FreeOpcuaClient. I also created an [issue ](https://github.com/FreeOpcUa/opcua-asyncio/issues/1172) on the used server library. So i am relativly certain, that my data structures on the server side are correctly defined and transmitted. The issue mentioned also contains a short description of how i setup the server side.

My Question: Is this fixable from a server perspective because i can not alter the client software or is this issue related to the client and i have to contact the developers of the client. Can this maybe caused by a version conflict between the client and the server?

Thanks

nakr94, 2023-01-12:
Edit: The first value of the array is received correctly on startup. After this the exception is thrown and all following entries are empty. 
[image]

AndreasHeine, 2023-01-12:
you should at least post the node-opcua version!?

nakr94, 2023-01-12:
I contacted the developers, ill let you know if i have the version.

nakr94, 2023-01-13:
The version ist: 2.62.7

mikakaraila, 2023-01-13:
And latest one is 2.88.0 here are releases:
https://github.com/node-opcua/node-opcua/releases

I expect there are a lot fixes related to extension object handling.

nakr94, 2023-01-16:
I created a short client program following the [example](https://github.com/node-opcua/node-opcua-sample/blob/master/simple_client_es8.js):

```
const {
    OPCUAClient,
    AttributeIds,
    TimestampsToReturn,
    StatusCodes,
    DataType
} = require("node-opcua");

const endpointUrl = "opc.tcp://localhost:10000/Tes/";
const nodeId = "ns=2;s=MyVar";

async function main() {
    try {

        const client = OPCUAClient.create({
            endpoint_must_exist: false,
            connectionStrategy: {
                maxRetry: 2,
                initialDelay: 2000,
                maxDelay: 10 * 1000
            }
        });
        client.on("backoff", () => console.log("retrying connection"));

        await client.connect(endpointUrl);
        const session = await client.createSession();
        const dataValue = await session.read({ nodeId, attributeId: AttributeIds.Value });
        console.log(` value = ${dataValue.value.value.toString()}`);

        const subscription = await session.createSubscription2({
            requestedPublishingInterval: 1000,
            requestedLifetimeCount: 100, // 1000ms *100 every 2 minutes or so
            requestedMaxKeepAliveCount: 10,// every 10 seconds
            maxNotificationsPerPublish: 10,
            publishingEnabled: true,
            priority: 10
        });

        subscription
            .on("started", () => console.log("subscription started - subscriptionId=", subscription.subscriptionId))
            .on("keepalive", () => console.log("keepalive"))
            .on("terminated", () => console.log("subscription terminated"));

        const monitoredItem = await subscription.monitor(
        {
            nodeId: nodeId,
            attributeId: AttributeIds.Value
        },
        {
            samplingInterval: 1000,
            discardOldest: true,
            queueSize: 10
        }, TimestampsToReturn.Both
        );

        monitoredItem.on("changed", (dataValue) =>console.log(` value = ${dataValue.value.value.toString()}`));
        await new Promise((resolve) => setTimeout(resolve, 5000));
        await subscription.terminate();

        console.log(" closing session");
        await session.close();
        await client.disconnect();
    }
    catch (err) {
        console.log("Error !!!", err);
        process.exit();
    }
[... 44 lines cut ...]

},/* OpaqueStructure */ { 
nodeId ns=2;i=2
buffer = 
00000000: 00 00 00 00 00 00 05 00 00 00 52 45 41 44 59 02 00 00 00 00 00 00 00                               ..........READY........

},/* OpaqueStructure */ { 
nodeId ns=2;i=2
buffer = 
00000000: 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00                                              ..................

},/* OpaqueStructure */ { 
nodeId ns=2;i=2
buffer = 
00000000: 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00                                              ..................

},/* OpaqueStructure */ { 
nodeId ns=2;i=2
buffer = 
00000000: 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00                                              ..................

}
```

Are there any idea how to solve this? Because i suspect that the original client is canceling the read operation right after the error occurs. So i would guess that if the error is resolved, the rest should work fine because the actual data is received correctly.

nakr94, 2023-01-23:
I am still stuck with this issue. Is there anyone who has at least a direction i can move to in order to solve this?

mikakaraila, 2023-01-23:
I expect Etienne fixed this to latest (new version):
https://github.com/node-opcua/node-opcua/releases/tag/v2.89.0

nakr94, 2023-01-23:
Unfortunately the error still occurs in version 2.89.0 with the same error message produced by the client code above:
```
Connecting...
Reading...
09:04:40.513Z :datatype_factory              :165   warning  ns=2;i=1 name= RobotDataEntry  do not 
have binary encoding
09:04:40.630Z :resolve_dynamic_extension_obje:147   err Error: Internal Error: we are not expecting this dataType to be processed already ns=2;i=1
 ...                                                    at {...}\js\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:45:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at fulfilled ({...}\js\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:5:58)
 ...                                                    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
09:04:40.633Z :resolve_dynamic_extension_obje:148   opaqueStructure =  ns=2;i=2
09:04:40.635Z :resolve_dynamic_extension_obje:149   opaqueStructure =  0x00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

                                                                                      ................................
 ...
09:04:40.637Z :resolve_dynamic_extension_obje:150   00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................     
 ...
09:04:40.638Z :resolve_dynamic_extension_obje:151   ExtraDataTypeManager
 ...                                                ------------- namespace: http://opcfoundation.org/UA/
 ...                                                ------------- namespace: urn:freeopcua:python:server
 ...
 ...                                                ------------- namespace: Test
 ...                                                structureTypeName = RobotDataEntry
 ...                                                name            RobotDataEntry
 ...                                                dataType        ns=2;i=1
 ...                                                binaryEncoding  ns=0;i=0
 ...                                                           name                           false String
 ...                                                           dataType                       false Int64
 ...                                                           stringValue                    false String
 ...                                                           intValue                       false Int64
 ...                                                           floatValue                     false Double
value = /* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................ 

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................ 

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................ 

}
Closing session
```
I verified that i used the version 2.89.0 by looking into the package.json.

erossignon, 2023-01-30:
It looks like the OPCUA Server is not exposing the BinaryEncoding for the dataType. Please make sure that the NodeSETXML.XMl injected to the Python Server provides a binary encoding.

Can you provide more information about how the data model of the remote server has been created ?

nakr94, 2023-01-30:
I am using a library for that purpose so i have no direct idea how the data model is setup internally. I created this [issue](https://github.com/FreeOpcUa/opcua-asyncio/issues/1172) where i tried to verify that my code is not missing any initialization step which seemed to be the case.

erossignon, 2023-05-02:
I found the root cause of this:
* Decoding Extension Object requires a long sequence of background operation to extract the Extension Object definition.
* this step is done as the first DataValue containing a ExtensionObject is retrieve.
* In the case of a monitoring of a Extension Object variable this will be done as the first PublishResponse is received.
* However the flow of data change notification could be faster that the time to retrieve the Extension Object definition causing the extraction routine to be entered once again as it is already extracting the definition. This re-entrancy causes a clash sometime.

erossignon, 2023-05-04:
ExtensionObject object decoding in the context of a subscription has been fixed in [version @2.102.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.102.0)

erossignon, 2023-05-04:
@nakr94, can you give it a try and confirm that the issue is now fixed on your end as well with this version ?

nakr94, 2023-05-08:
With the most recent Version, ill get the Data, but the Exception still occurs. I am fine with that though, i can just filter the output. Just want to let you know. Thank you very much for your efforts :)
```
Connecting...
Reading...
13:57:40.113Z :datatype_factory              :165   warning  ns=2;i=1 name= RobotDataEntry  do not have binary encoding
13:57:40.120Z :datatype_factory              :165   warning  ns=2;i=3 name= ErrorStruct  do not have binary encoding
13:57:40.219Z :resolve_dynamic_extension_obje:147   err Error: Internal Error: we are not expecting this dataType to be processed already ns=2;i=1
 ...                                                    at {...}\js\node_modules\node-opcua-client-dynamic-extension-object\dist\resolve_dynamic_extension_object.js:45:19
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    ...
 ...                                                    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
13:57:40.221Z :resolve_dynamic_extension_obje:148   opaqueStructure =  ns=2;i=2
13:57:40.222Z :resolve_dynamic_extension_obje:149   opaqueStructure =  0x00000000: 00 00 00 00 00 00 00 00 00 00 00 00 05 00 00 00 52 45 41 44 59 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

                                                                ................READY................
 ...
13:57:40.224Z :resolve_dynamic_extension_obje:150   00000000: 00 00 00 00 00 00 00 00 00 00 00 00 05 00 00 00 52 45 41 44 59 02 00 00 00 00 00 00 00 00 00 00    ................READY...........
 ...                                                00000020: 00 00 00 00 00                                                                                     .....
 ...
13:57:40.226Z :resolve_dynamic_extension_obje:151   ExtraDataTypeManager
 ...                                                ------------- namespace: http://opcfoundation.org/UA/
 ...                                                ------------- namespace: urn:freeopcua:python:server
 ...
 ...                                                ------------- namespace: Test
 ...                                                structureTypeName = RobotDataEntry
 ...                                                name            RobotDataEntry
 ...                                                dataType        ns=2;i=1
 ...                                                binaryEncoding  ns=0;i=0
 ...                                                           name                           false String
 ...                                                           dataType                       false Int64
 ...                                                           stringValue                    false String
 ...                                                           intValue                       false Int64
 ...                                                           floatValue                     false Double
 ...                                                structureTypeName = ErrorStruct
 ...                                                name            ErrorStruct
 ...                                                dataType        ns=2;i=3
 ...                                                binaryEncoding  ns=0;i=0
 ...                                                           number                         false Int64
 ...                                                           message                        false String
value = /* OpaqueStructure */ { 
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 05 00 00 00 52 45 41 44 59 02 00 00 00 00 00 00 00 00 00 00    ................READY...........
00000020: 00 00 00 00 00                                                                                     .....

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 4f 34 32 34 32 39 02 00 00 00 00 00 00 00 00 00    ................O42429..........
00000020: 00 00 00 00 00 00                                                                                  ......

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 50 33 30 38 38 38 02 00 00 00 00 00 00 00 00 00    ................P30888..........
00000020: 00 00 00 00 00 00                                                                                  ......

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 54 5a 55 5a 02 00 00 00 00 00 00 00 00 00 00 00    ................TZUZ............
00000020: 00 00 00 00                                                                                        ....

},/* OpaqueStructure */ {
nodeId ns=2;i=2
buffer =
00000000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................................

}
Closing session
```

erossignon, 2023-05-09:
@nakr94  it looks like the server is not specifying a BinaryEncoding node for the DataType.
Can you tell me more about the Server ? Is there a way I can access it for investigation ?

nakr94, 2023-05-09:
Sadly i can not give you the entire Server, but i prepared a snippet, that reproduces the same error.
```
import asyncio

from asyncua import ua, Server
from asyncua.common.structures104 import new_struct, new_struct_field

N = 5

def created_logged_task(routine: callable) -> asyncio.Task:
    task = asyncio.create_task(routine)

    def task_callback(task: asyncio.Task) -> None:
        try:
            task.result()
        except asyncio.CancelledError:
            pass
        except Exception as e:
            print(f"Exception '{e}' in task '{task}'!")

    task.add_done_callback(task_callback)
    return task

def construct_opcua_array(a: str, b: str, c: str, d: str) -> list:
    result = []
    for i in range(0, N, 1):
        entry = ua.RobotDataEntry()
        entry.name = ""
        entry.dataType = 0
        entry.stringValue = ""
        entry.intValue = 0
        entry.floatValue = 0.0
        result.append(entry)

    code = 0
    result[1].stringValue = a
    result[1].intValue = 0

    return result

async def run(endpoint: str, namespace: str) -> None:
    print("Initializing...")
    server = Server()
    server.set_security_IDs(["Anonymous"])
    server.set_security_policy([ua.SecurityPolicyType.NoSecurity])

    await server.init()
    server.set_endpoint(endpoint)
    idx = await server.register_namespace(namespace)

    print("Creating Types...")
    var_struct_node, var_struct_encoding = await new_struct(
        server,
        idx,
        "RobotDataEntry",
        [
            new_struct_field("name", ua.VariantType.String),
            new_struct_field("dataType", ua.VariantType.Int64),
            new_struct_field("stringValue", ua.VariantType.String),
            new_struct_field("intValue", ua.VariantType.Int64),
            new_struct_field("floatValue", ua.VariantType.Double),
        ],
    )
    await server.load_data_type_definitions()

    print("Creating Variables...")
    name = "TestVar"
    node_id = f"ns={idx};s={name}"
    default_val = construct_opcua_array("", "", "", "")
    node = await server.nodes.objects.add_variable(node_id, "TestVar", ua.Variant(default_val, ua.VariantType.ExtensionObject))

    await node.write_value_rank(1)
    await node.write_array_dimensions([N])
    await node.add_reference(var_struct_encoding[0], 38);

    print("Running...")
    created_logged_task(server.start())
    
    i = 0
    while True:
        await asyncio.sleep(3.0)
        
        print("Incrementing...")
        string = f"{i}";
        await node.write_value(ua.Variant(construct_opcua_array(string, string, string, string), ua.VariantType.ExtensionObject))
        i += 1
        
if __name__ == "__main__":
    endpoint = "opc.tcp://localhost:10000/Foo/Bar"
    namespace = "Testing"

    try:
        asyncio.run(run(endpoint, namespace), debug=True)
    
    except KeyboardInterrupt:
        print("Detected keyboard interrupt, aborting...")
    print("Bye.")
```
With this i am getting this Variable in UAExpert (which looks good for me):
[image]

The error is the same as mentioned in my previous comment.

erossignon, 2024-01-14:
@nark94, stricly speaking, there is a bug in the python asyncua in the way the dataType is specified;

[image]

We have
```
  DataType  RobotDataEntry (ns=2;i=0)
         +-- hasEncoding -> "Default Binary"  (nodeClass=ObjectType)
```

We should have
```
  DataType  RobotDataEntry (ns=2;i=0)
         +-- hasEncoding -> "Default Binary"  (nodeClass=Object)  << HERE
```

erossignon, 2024-01-21:
fixed in node-opcua@2.200.0

closed 2024-01-21 by erossignon