node-opcua/node-opcua #1234: setPublishingMode() and setMonitoringMode are not accessible?

JoelKrec, 2023-01-26:
**Context**

- ( X ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

The [specification part 4](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.13.4) states that there is a service that disables publishing of notifications of a subscription. I remembered that one or two years ago I already used it and in a issue from 2021 you mentioned it too. 
I found a implementation in code in the ClientSessionImpl and ClientSubscriptionImpl. But neither of them are accessible from ClientSession or ClientSubscription objects. It is the same with setMonitoringMode from the monitoredItem.

Is this intended and why is this not accessible but implemented? Are there other specifications that state a client is not allowed to use these services?

If this is intended, is there a quick workaround you know of to disable publishing or sampling of a monitoredItem or the whole subscription or session?

Lastly, the book node-opcua by Example, neither the old nor the new version, does not give an example of setPublishingMode or setMonitoringMode. It just lists them in the specification service list at the end.

AndreasHeine, 2023-01-26:
Hey,

have you tested:

```ts
const item = await this.subscription!.monitor({
                nodeId: v.nodeId,
                attributeId: AttributeIds.Value
            }, {
                samplingInterval: 1000
            },
            TimestampsToReturn.Both
            )
item.setMonitoringMode(MonitoringMode.Sampling)
```

regards

JoelKrec, 2023-01-26:
Hey and thank you!

Apparently, that is something I have overlooked. I was to fixed on the subscription and session as it seems.

Still, do you know why setPublishing is not accessible even if a specification mentions it? 

But thank you again for the info!

Best regards
Joel

AndreasHeine, 2023-01-26:
it looks like the interface is not complete!?

https://github.com/node-opcua/node-opcua/blob/93d5502a1b71afbb5e5fa56846615c412b63cc1b/packages/node-opcua-client/source/client_subscription.ts#L54

JoelKrec, 2023-01-26:
Okay, thank you.
Is there a list or something like that where people can see what is incomplete or still missing? I sometimes find it difficult to contribute to big projects when I don't really know what needs to be done or is still incomplete. Maybe that would help new and old contributors. But this is not a place for this issue.

As my problem is solved, I will close this issue and thank you again for your help!

closed 2023-01-26 by JoelKrec