node-opcua/node-opcua #1235: BadTooManyMonitoredItems

cz-1999, 2023-01-28:
Hi,
I wan't to limit the number of monitored items per subscription in the server and return a BadTooManyMonitoredItems if too many monitored items are added. Is this possible using node-opcua?

AndreasHeine, 2023-01-28:
sure! you can do that by setting the server configuration -> ServerCapabilities -> OperationalLimits

erossignon, 2023-01-30:
duplicates #616

erossignon, 2023-01-30:
More precisely, this aspect is controled by

*    `serverCapabilities.maxMonitoredItemsPerSubscription` 
*     `serverCapabilities.maxMonitoredItems`

You can initialize them this way:

```typescript
const server = new OPCUAServer({
  ... 
  serverCapabilities: {
      maxMonitoredItemsPerSubscription: 10000,
      maxMonitoredItems: 100000
  }
});

Note: Those 2 variables are now exposed (readonly) in the standard address space with OPCUA 1.05.

closed 2023-03-01 by erossignon