node-opcua/node-opcua #1236: Set variable to a specific value when starting the server

Newinthistopic, 2023-01-28:
Hello all,

I want a certain variable "actTemp" to be set to a certain value at startup.

I have the variables in an extra js file and then call it in the server file with the require function.

I now want to set the variable actTemp to 20 in my file where I have the getter and setter function when the server is launched.

Looking forward to any suggestions.

Attached the code:

`module.exports = {
    
    run: function (namespace, device, opcua, verbose, serverValues) {
        
        var actset = namespace.addObject({
                                              componentOf: device,
                                              browseName: "actset",
                                              nodeId: "ns=1;s=" + "actset"        
                                            });
        
        var actTemp           = 'actTemp'; 
        serverValues[actTemp]=20;
        namespace.addVariable(
            {
                componentOf: actset,
                nodeId     : "ns=1;s=" + actTemp,
                browseName : actTemp,
                dataType   : opcua.DataType.Double,
                value      : {
                    get: function () {
                        ;
                        return new opcua.Variant({dataType: opcua.DataType.Double, value: serverValues[setTemp]});
                    },                        
                                            // return new opcua.Variant({dataType: opcua.DataType.Double, value:  serverValues[setTemp]});
                    
                    set: function (variant) {
                        
                            serverValues[setTemp] = parseFloat(variant.value);
                            return opcua.StatusCodes.Good;
                                               
                    }                    
                    
                }

            }
             
             );
        var setTemp  = 'setTemp';
        serverValues[setTemp] = 0;
        namespace.addVariable(
            {
                componentOf: actset,
                nodeId     : "ns=1;s=" + setTemp,
                browseName : setTemp,
                dataType   : opcua.DataType.Double,
                value      : {
                    get: function () {
                         
                        return new opcua.Variant({dataType: opcua.DataType.Double, value: serverValues[setTemp]});
                       
                    },
                   set: function (variant) {
                   
                    if (parseFloat(variant.value) > 20) {
                        serverValues[actTemp] = 20;
                        serverValues[setTemp] = parseFloat(variant.value);
                        serverValues[actTemp] = parseFloat(variant.value);
                        return opcua.StatusCodes.Good;
                    }else {  
                        return opcua.StatusCodes.Bad  
                                              
                        } 
                    } ,
                                   
                                  
                }   
            }   
            
            

        );
     
}

}`

AndreasHeine, 2023-01-29:
initialise serverValues! or set the value from source!

Newinthistopic, 2023-01-29:
Hello Mr. Heine. Thanks for u reply. I Tried ot. But it did not work. Can u Show me how u would do ?

AndreasHeine, 2023-01-29:
> Hello Mr. Heine. Thanks for u reply. I Tried ot. But it did not work. Can u Show me how u would do ? 

just set the values before running the function so the values are present before the server is initialised!

Newinthistopic, 2023-01-29:
Thanks for the answer. Do I need to add the variable before the module.run function or in the js. File where the server is initialized . Thanks in advance.

AndreasHeine, 2023-01-29:
```js
get: function () {
                     
                    return new opcua.Variant({dataType: opcua.DataType.Double, value: serverValues[setTemp]});
                   
                },
```

as soon as a opcua client connects and reads the variable value the get function is called -> means serverValues[setTemp] must have a value before the server initialises!

i mean it looks like that you did it right!

```js
    var setTemp           = 'setTemp';
    serverValues[setTemp]=20;
    namespace.addVariable(
```

the only thing i would guess is that addVariable initilaises the variable...
you can try to set the value after calling addVariable:

```js
const setTemp = namespace.addVariable({<VariableOptions>})
setTemp.setValueFromSource(<Variant>)    
```   

like:
https://github.com/node-opcua/node-opcua/blob/1be94dc098d2972bbb3b301f5ba5a62b37d68c65/packages/node-opcua-samples/bin/simple_server.js#L275

Newinthistopic, 2023-02-01:
Thanks for your help

closed 2023-02-01 by Newinthistopic