node-opcua/node-opcua #1238: Creating a simple server crashes in an electron app

ptillemans, 2023-02-06:
**Current behavior**

When creating a simple server running in electron crashes the server thread. This happens too in our real application which is a frontend for an ATE.

A reproduction repo is made at 

**Describe the bug**

I created a simple OPCUA server based on the provided example and started it from the main.js of my electron app.

When starting the app we get the following error:

```
➜  electron-opcua git:(main) npm start                                                                              >>>

> electron-opcua@1.0.0 start
> electron .

App threw an error during load
Error: error:06000066:public key routines:OPENSSL_internal:DECODE_ERROR
    at createPrivateKey (node:internal/crypto/keys:619:12)
    at Object.<anonymous> (/home/pti/Projects/electron-opcua/node_modules/node-opcua-secure-channel/dist/source/message_builder.js:37:59)
    at Module._compile (node:internal/modules/cjs/loader:1141:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1196:10)
    at Module.load (node:internal/modules/cjs/loader:1011:32)
    at Module._load (node:internal/modules/cjs/loader:846:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1035:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/pti/Projects/electron-opcua/node_modules/node-opcua-secure-channel/dist/source/index.js:25:14)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: error:06000066:public key routines:OPENSSL_internal:DECODE_ERROR
    at createPrivateKey (node:internal/crypto/keys:619:12)
    at Object.<anonymous> (/home/pti/Projects/electron-opcua/node_modules/node-opcua-secure-channel/dist/source/message_builder.js:37:59)
    at Module._compile (node:internal/modules/cjs/loader:1141:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1196:10)
    at Module.load (node:internal/modules/cjs/loader:1011:32)
    at Module._load (node:internal/modules/cjs/loader:846:12)
    at f._load (node:electron/js2c/asar_bundle:2:13330)
    at Module.require (node:internal/modules/cjs/loader:1035:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/pti/Projects/electron-opcua/node_modules/node-opcua-secure-channel/dist/source/index.js:25:14)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[9483:0206/135159.535359:ERROR:gpu_memory_buffer_support_x11.cc(44)] dri3 extension not supported.
https://github.com/ptillemans/electron-opcua
```

on line 90 of *message-builder.ts* an invalid key is created. Since electron is compiled against BoringSSL which is stricter than OpenSSL, creating this key throws the error. 

Replacing this key with the same key reformatted by pumping it through *openssl* cli tool changed the error to  key length too short.

Replacing the key with a dummy generated 1024 bit RSA key fixes the issue and the application starts up normally.

**Step by step instruction to reproduce**

check out the repo at https://github.com/ptillemans/electron-opcua

run 

```
$ npm start
```
and the electron app will start, throw the error and hang there without a visible window till Ctrl-C is pressed in the terminal.

When _fixing_ the invalid key to a valid dummy key, the app throws no errors and shows a friendly hello message.
Steps to reproduce the behavior:

**Expected behavior**

**Screenshots**

**Gist**

gist:

https://github.com/ptillemans/electron-opcua

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Intel based PS (various)
- OS version: various Linux

    - (x) Windows : version : Win 11 Pro 22H2
    - (x) Linux : version : Arch recent, Ubuntu-14.04, Ubuntu-22.04
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:
    not applicable.

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :

  
  

```
electron-opcua@1.0.0 C:\Users\PeterTillemans\Projects\electron-opcua
+-- electron@22.2.0
`-- node-opcua@2.90.0
```

- **Node:**

  `node --version` = 19.6.0

Mansko, 2023-02-07:
I have the exact same issue in my electron app. "Fixed" it by commenting out line 37:39 inside "_node-opcua-secure-channel/dist/source/message_builder.js_". Needless to say that I'm also interested in a correct fix.

EirikVea, 2023-02-16:
I am experiencing the same issue from and including v2.86.0 and onwards. Electron v22.0.0.
[image]

erossignon, 2023-03-01:
fixed in 2.91.1