node-opcua/node-opcua #1240: Missing nodeClass crawling an address space

MattiaMarzano-Eng, 2023-02-08:
Hello,
I'm currently developing a mapping tool for OPCUA Agent in the Fiware catalogue able to map all the infomation in whatevere OPCUA Server once stablished a connection.
Until now no issues or impediments, I'm able to navigate every OPCUA servers using the "node-opcua-client-crawler": "^2.90.0".
What I would like to have invoking the method crawler.read(nodeId) is the field nodeClass allowing me to discriminate if a node I'm crawling is an object, a method, a variable...

Is there any specific option to retrieve this kind of information? Or is it something can be added in the crawler.read(nodeId) response?

Thanks

iliareshetov, 2023-03-15:
Hi @MattiaMarzano-Eng,
Could you provide some more for info ? 
Code example of what you're trying to do?

MattiaMarzano-Eng, 2023-03-20:
Hello,
what I'm trying to do is a "simple" crawling in recursive way. Below the code I'm using to crawl a nodeID once a session is established.
`const nodeCrawler = require('node-opcua-client-crawler');
const crawler = new nodeCrawler.NodeCrawler(the_session);
const browseSubLev = await crawler.read(nodeId);`

What I would like to have in the response stored in the _browseSubleve_ is the nodeClass associated to the specific nodeID I'm crawling. It is fundamental for me to proceed with the implementation of the tool I mentioned before. Is there a way to add it in the response?

iliareshetov, 2023-03-27:
Hi, 

I added one unit test to file : packages\node-opcua-client-crawler\test\test_crawler_on_addresss_space.ts

In the test, I used NodeCrawlerBase to fetch nodeClass information. 
All of this was accomplished without making any actual changes to the code.

Is this what you were looking for, or is there anything else you need ?

```typescript
       const crawler = new NodeCrawlerBase(session);

        interface CNode {
            browseName: string;
            nodeClass: string;
        }

        const browseResults: CNode[] = [];

        const data = {
            onBrowse(this: UserData, crawler1: NodeCrawlerBase, cacheNode: CacheNode) {
                const cnode: CNode = {
                    browseName: cacheNode.browseName.toString(),
                    nodeClass: cacheNode.nodeClass.toString()
                };
                browseResults.push(cnode);
            }
        };

        await crawler.crawl($.groupNodeId, data);

        browseResults.forEach((cnode) => {
            console.log(`INFO: browseName: ${cnode.browseName}, nodeClass: ${cnode.nodeClass}`);
        });
``` 

Output:
```
2023-03-27T18:42:26.847Z - INFO: Elapsed time: 3ms
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Group, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: Organizes, nodeClass: 32
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object0, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object1, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object2, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object3, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object4, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object5, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object6, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object7, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object8, nodeClass: 1
2023-03-27T18:42:26.848Z - INFO: browseName: 1:Object9, nodeClass: 1
```

erossignon, 2023-05-07:
resolve by example provided by @iliareshetov

closed 2023-05-07 by erossignon