node-opcua/node-opcua #1242: Node-opcua logs "cannot find fuCount" when it is mangled

tetanw, 2023-02-09:
**Current behavior**
When node-opcua is mangled as part of a bundling procedure it will produce a lot of errors that show up in the console.

**Describe the bug**
The following messages are spammed in my logs:
[image]

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Copy the files from my gist into an empty directory
2. **Important**: copy the standard nodeset from node-opcua (the one called ```Opc.Ua.NodeSet2.xml```) over into this directory as well
3. Do a ```npm install```
4. Do a ```npx webpack --config webpack.config.js``` to create a bundle
5. Run ```node dist/main.js``` to run the bundle
6. Start a OPCUA client and connect with the server (In my case, I used UaExpert as a client)
7. See all kinds of similar messages show up like the one above

**Expected behavior**

To not have any error messages show up in my log.

**Suggested fix**

Everywhere were ```session.incrementRequestErrorCounter(ResponseClass.name.replace("Response", ""));``` shows up change it to ```session.incrementRequestErrorCounter(ResponseClass.schema.name.replace("Response", ""));```. This means that the counter is now sourced from the schema instead of the class name, because the schema name is not mangled and has the same name as the class name anyway. I quickly tried this solution in the ```_apply_on_SessionObject``` method in ```node-opcua-server\source\opcua_server.js``` and it seems to fix the problem.

**Gist**

gist: https://gist.github.com/tetanw/92284239c744f11acaa23ff60b074ca6

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Lenovo Thinkstation P700
- OS version: Windows 10 Enterprise, 21H2

    - ( ) Windows : version : Windows 10 Enterprise, 21H2
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: UaExpert
    - Version: 1.5.1
    - Manufacturer/Software vendor: Unified automation
    - link to the manufacturer or product site: [UA Homepage](https://www.unified-automation.com/products/development-tools/uaexpert.html)

- **node-opcua version:** 2.90.0
- **Node:** 16.13.1

AndreasHeine, 2023-02-09:
why are you bundling it with webpack?

tetanw, 2023-02-09:
Because we use the tree shaking mechanism of webpack to keep bundle size down and as a result this saves disk space. The mangling is done as an obfuscation step.

erossignon, 2023-02-10:
thank for reporting, this will be fixed in 3ebaaef0d12640c645257db3367743864fad3790

erossignon, 2023-02-15:
Should now be fixed in node-opcua@2.90.1 (https://github.com/node-opcua/node-opcua/releases/tag/v2.90.1).

@tetanw, please confirm

tetanw, 2023-02-16:
Just upgraded, version 2.90.1 does indeed fix the issue.

closed 2023-02-16 by erossignon