node-opcua/node-opcua #1244: Stucture inheritence not working as expected.

rfx77, 2023-02-16:
Hi!

I have a problem with creating a structure inheritence. I have a base type AAABaseDataType and a child-type AAAChildDataType
i define both schemas and create the types which works well and i can see both types in the Types-Node with the opcua-commander.
In the types-tree AAABaseDataType iss under Structure and AAAChildDataType is under AAABaseDataType.

The problem is no that AAAChildDataType does not inherit the fields from AAABaseDataType. I can see the fields in the property "$extensionFields" property but not in the instance.

i have attached the full code of my test.

Can someone please tell me what i am doing wrong.

Greetings,
Franz

```

let os = require("os");

let opcua = require("node-opcua");
const {addExtensionObjectDataType, addVariableTypeForDataType} = require("node-opcua-modeler")

let server = new opcua.OPCUAServer({
    port: 4334,
    resourcePath: "/UA/MYServer",
    buildInfo: {
        productName: "TestServer",
        buildNumber: "1",
        buildDate: new Date()
    }
});

server.initialize(async () => {
    const addressSpace = server.engine.addressSpace;
    const namespace = addressSpace.getOwnNamespace();

    let device = namespace.addObject({
        organizedBy: addressSpace.rootFolder.objects,
        browseName: "MyDevice"
    });

    const opt_base = {
        browseName: "AAABaseDataType",
        structureDefinition: {
            baseDataType: "",
            fields: [
                {name: "basefield1", dataType: opcua.DataType.String,valueRank:-1},
                {name: "basefield2",dataType: opcua.DataType.String,valueRank:-1},
            ]
        },
    };
    const tp_base = await addExtensionObjectDataType(namespace, opt_base);

    const opt_child = {
        browseName: "AAAChildDataType",
        subtypeOf: tp_base.nodeId,
        structureDefinition:{
            baseDataType: tp_base.nodeId,
            fields: [
                {name: "childfield1",dataType: opcua.DataType.String,valueRank:-1}
            ]
        },

    };
    const tp_child = await addExtensionObjectDataType(namespace, opt_child);

    const vt_base = addVariableTypeForDataType(namespace, tp_base)
    const vt_child = addVariableTypeForDataType(namespace, tp_child)

    const myStructureType2 = namespace.findVariableType("AAAChildType");

    let inst1 = myStructureType2.instantiate({
        browseName: "MyVar3",
        componentOf: device,
    });
    console.log(JSON.stringify(inst1, null, 2))
    server.start();
});

```

erossignon, 2023-02-17:
addExtensionObjectDataType requires a fully defined structureDefinition, including redefinition of field defined in base structure.

You could adapt your code like this: 

```typescript
const opt_child = {
        browseName: "AAAChildDataType",
        subtypeOf: tp_base.nodeId,
        structureDefinition:{
            baseDataType: tp_base.nodeId,
            fields: [
                 // Add this line
                 .... opt_base.structureDefinition.fields,
                {name: "childfield1",dataType: opcua.DataType.String,valueRank:-1}
            ]
        },

    };
 ````

erossignon, 2023-02-17:
You may also check [Sterfive's OPCUA Modeler](https://www.sterfive.com/product/modeler)  provided  by the author of NodeOPCUA to produce XML models.

rfx77, 2023-02-18:
Thanks very much. I will givebit a try. Greetings, Franz

closed 2023-02-24 by erossignon