node-opcua/node-opcua #1247: Changing namespaceindex on an opc ua server in node.js

Newinthistopic, 2023-03-02:
Hello everyone,

I have set up an OPC UA server in node.js and generated variables in another js file. Now, I would like to change the Namespace index to organize my variables in a separate Namespace.

Unfortunately, I am not familiar enough with OPC UA and node.js to implement this change on my own. Can someone explain how I can change the Namespace index?

Attached are the code for my variables and the code for the OPC UA server.

I would be very grateful for your help!

Best regards.

module.exports = {
    run: function (namespace, device, opcua, verbose, serverValues) {
    
        
        
        // create ppC_Compound_V3 object
        var ppC_Compound_V3 = namespace.addObject({
            organizedBy: device,
            browseName: "ppC_Compound_V3",
            nodeId: "ns=1;s=PLC",
            description: "The PLC instance which supports you with OPC UA functionality" 
        });

        // create DatablocksGlobal object
        var datablocksGlobal = namespace.addObject({
            organizedBy: ppC_Compound_V3,
            browseName: "DatablocksGlobal",
            nodeId: "ns=1;s=DatablocksGlobal"
        });

        // create ZEEX_3111_Hmi object
        var prefix = namespace.addObject({
            organizedBy: datablocksGlobal,
            browseName: "ZEEX_3111_Hmi",
            nodeId: "ns=1;s=" + "\"ZEEX_3111_Hmi\""
        });

         // create udtEmPz variable undefined Datatype Equipment Module Process Zone
        var udtEmPz = namespace.addVariable({
            componentOf: prefix,
            browseName : "udtEmPz",
            dataType   : opcua.DataType.Double,
            nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""  ,         
        });
I want to set the namespace index of the variable "udtEmPz" to 3.

console.log("<<< WebIQ OPC UA Server without MQTT >>>");
console.log("(c) 2019 Smart HMI GmbH");
console.log("-----------------------------------------------------------");
console.log('Starting... please wait!');

const fs = require('fs');

const {
          OPCUAServer,
          OPCUACertificateManager
      } = require("node-opcua");

const certificateManager = new OPCUACertificateManager({
                                                           automaticallyAcceptUnknownCertificate: true,
                                                           rootFolder                           : "./certs",
                                                       });

// Select profile
var profile = 'default';
if (process.argv.length > 2) {
    profile = process.argv[2];
}
var profileDir = __dirname + '/profiles/' + profile;

if (!fs.existsSync(profileDir)) {
    console.log('ERROR: Profile does not exist: "' + profile + '" dir: ' + profileDir);
    process.exit(1);
}

console.log('Using profile "' + profile + '"');

// Functions
const opcua   = require("node-opcua");
var config    = require(profileDir + '/config.json');
const jp      = require('jsonpath');
var batchVars = null;

//console.log(batchVars);

// Settings / vars
var verbose             = config.verbose;
var serverValues        = {}; // Holds all server values except MQTT values
const methodConfigDir   = profileDir + '/methods';
const varConfigDir      = profileDir + '/variables';

// Parse config files (JS)
let methodFiles = [];
let varFiles    = [];
if (fs.existsSync(methodConfigDir)) {
    var files = fs.readdirSync(methodConfigDir);
    files.forEach(file => {
        if (file.match(/\.js$/)) {
            methodFiles.push(file);
        }
    });
}

if (fs.existsSync(varConfigDir)) {
    files = fs.readdirSync(varConfigDir);
    files.forEach(file => {
        if (file.match(/\.js$/)) {
            varFiles.push(file);
        }
    });
}

(async () => {

    try {
        /*******************/
        /*** SERVER INIT ***/
        /*******************/
        const server = new opcua.OPCUAServer(
            {
                alternateHostname       : config.opcua.serverHostname,
                port                    : config.opcua.serverPort,
                resourcePath            : config.opcua.resourcePath,
                buildInfo               : {
                    productName: config.opcua.productName,
                    buildNumber: config.opcua.buildNumber,
                    buildDate  : new Date(config.opcua.buildDate)
                },
                serverCertificateManager: certificateManager
            }
        );

        await server.initialize();
        if (verbose) console.log("[OK] OPC UA: Server created");

 

        const addressSpace = server.engine.addressSpace;
        const namespace    = addressSpace.getOwnNamespace();

        const device = namespace.addObject(
            {
                organizedBy: addressSpace.rootFolder.objects,
                browseName : config.opcua.browseName
            }
        );
      
         

        /*****************/
        /*** VARIABLES ***/
        /*****************/
        if (varFiles.length) {
            if (verbose) console.log("[..] OPC UA: Creating variables from " + varFiles.length + ' external files');
            varFiles.forEach(function (file) {
                var module            = require(profileDir + '/variables/' + file);
                var varName           = file.replace(/\.js$/);
                serverValues[varName] = module.run(namespace, device, opcua, verbose, serverValues);
            });
            if (verbose) console.log("[OK] OPC UA: Variables created");
        }
       
           
        /********************/
        /*** START SERVER ***/
        /********************/
        await server.start();
        let endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;

        // Replace host name by local ip
        endpointUrl = endpointUrl.replace(/^opc.tcp:\/\/([^:]+)/, "opc.tcp://127.0.0.1");

        if (verbose) console.log("[OK] OPC UA: Listening on port " + server.endpoints[0].port);
        console.log('\nPlease use these values to connect to this server via the IO Manager in WebIQ Designer:');
        console.log("Endpoint URL: ", endpointUrl);
        console.log("Namespace:     " + namespace.namespaceUri);

    } catch (err) {
        console.log("[ERROR]");
        console.log(err);
    }
  

})();

AndreasHeine, 2023-03-02:
if you want ns=3 you need to create two additional namespaces because the index is some sort of fluid so the idx is always the next free idx in the namespace array
the ns-uri (http://mynamespace.com/UA) is the actual const and can be resolved via the namespacearray which each server exposes. its a better practice to not think ns-idx is something const which you can choose...

Newinthistopic, 2023-03-03:
Hello Andreas,

many thanks for your feedback.
I understood that I have to build all namespaces before 3 first, so that I can use namespace 3.

Can you give me a code where to insert it?
I had already used a lot of codes from this forum and none worked.

I am happy about feedback.

Many greetings 

> if you want ns=3 you need to create two additional namespaces because the index is some sort of fluid so the idx is always the next free idx in the namespace array the ns-uri (http://mynamespace.com/UA) is the actual const and can be resolved via the namespacearray which each server exposes. its a better practice to not think ns-idx is something const which you can choose...
> 
> edit: ns=0 is always OPC Core ns=1 is always Serverown-Namespace ns=... imported xml or other registered namespaces depend on the order in which they will be registered

Newinthistopic, 2023-03-03:
.

Newinthistopic, 2023-03-03:
Hello Andreas,

many thanks for your feedback.
I understood that I have to build all namespaces before 3 first, so that I can use namespace 3.

Can you give me a code where to insert it?
I had already used a lot of codes from this forum and none worked.

I am happy about feedback.

Many greetings

closed 2023-03-03 by Newinthistopic

AndreasHeine, 2023-03-03:
its rather simple:

// ns=0 and ns=1 already exist!

const namespace2 = addressSpace.registerNamespace('http://mynamespace-2/UA/')

Newinthistopic, 2023-03-03:
Hello i put it here now: 
[image]

but it does not work. When I set the variable ns=2 then come this error message: 

[image]

AndreasHeine, 2023-03-03:
remove the nodeid from addVariable...

currently you try to add to "namespace" (which is ns=1) a variable with ns=2 which is impossible ^^ 

alternatively you can do 
```ts
`ns=${addressSpace.getNamespaceIndex('http://mynamespace-2/UA/')};s=DataBlocksGlobalsdsfsfs`
```

Newinthistopic, 2023-03-03:
super thanks, that's how it worked :)  

[image]

Now I have a new problem.

I have my variables in another file. See here:

[image]

I would like to change the namespace of the variable udtEmPz, how can I do that ?

Newinthistopic, 2023-03-03:
> I expect leave nodeId without "ns=1;" then it should use your string based nodeId.

Unfortunately this does not work

AndreasHeine, 2023-03-03:
in your second screenshot "namespace2" is undefined!? your run function just gets namespace as param! maybe you should pass addressSpace instead so you can access all namespaces while invokin run function?

Newinthistopic, 2023-03-03:
i now have it that way, but it doesn't work.

[image]

AndreasHeine, 2023-03-03:
you need to look more careful in the examples i posted...

addressSpace.getNamespaceIndex('http://mynamespace-3/UA/').addObject()

Newinthistopic, 2023-03-03:
danke für deine Antworten. Ich habe das jetzt so:

[image]

UA Experts zeigt auch die Namespaces an. 

[image]

ich möchte nun das bestimmte Variablen den Index 2 oder 3 haben. freue mich auf deine Hilfe :)

AndreasHeine, 2023-03-03:
weird translation ^^ anyways my last attempt -> never hardcode a namespaceindex!

use:
```
const nsUri_xyz = "http://mynamespace-xyz"
const xyzNamespace = addressSpace.registerNamespace(nsUri_xyz)
const xyzIndex = addressSpace.getNamespaceIndex(<namespacename>)
```

you can use xyzNamespace to add stuff to xyzIndex (nsUri_xyz)!

Newinthistopic, 2023-03-03:
Hello, thank you for your efforts. All this does not work. This is my last stand.

The test object is also displayed as index 3. that fits so f

[image]

[image]

[image]

I now have variables and objects in another file and want to change the namespace index there. but it doesn't work.

[image]

AndreasHeine, 2023-03-03:
https://github.com/AndreasHeine/SampleServer-node-opcua/blob/82d614b43cd79d29b8f888ffd732473a32255317/src/serveraddressspace/serveraddressspace.ts#L43

you can replace `const namespace = addressSpace.getOwnNamespace()` in L43 (link above) with `const namespace = addressSpace.getNamespace("http://mynamespaceuri/UA")`

and then just namespace.add... and they will all have the namespaceindex of "http://mynamespaceuri/UA" in the NamespaceArray

Newinthistopic, 2023-03-03:
i have now this in my server.js 
[image]

i get this error

TypeError: Cannot read properties of undefined (reading 'addObject')
    at C:\Users\fischerfk\Desktop\webiq-opcua-demo-server-mqtt\opcua-demo-server\server.js:123:34

AndreasHeine, 2023-03-03:
you need to use the vscode debugger to go line by line and look what happens in the objects thats the pain with plain js i mentioned! using typescript would prevent you from that hassle you experience now...

AndreasHeine, 2023-03-03:
send me the code in the chat so i can show you what i mean!

Newinthistopic, 2023-03-03:
> send me the code in the chat so i can show you what i mean!

here is my code:

module.exports = {
    run: function (namespace, device, opcua, verbose, serverValues) {
        
        
       
        // create ppC_Compound_V3 object
        var ppC_Compound_V3 = namespace.addObject({
            organizedBy: device,
            browseName: "ppC_Compound_V3",
            nodeId: "ns=2;s=PLC",
            description: "The PLC instance which supports you with OPC UA functionality" 
        })

      /*  // create DatablocksGlobal object
        var datablocksGlobal = namespace.addObject({
            organizedBy: ppC_Compound_V3,
            browseName: "DatablocksGlobal",
            nodeId: "ns=1;s=DatablocksGlobal"
        });

        // create ZEEX_3111_Hmi object
        var prefix = namespace.addObject({
            organizedBy: datablocksGlobal,
            browseName: "ZEEX_3111_Hmi",
            //nodeId: "ns=1;s=" + "\"ZEEX_3111_Hmi\""
        });

         // create udtEmPz variable undefined Datatype Equipment Module Process Zone
        var udtEmPz = namespace.addVariable({
            componentOf: prefix,
            browseName : "udtEmPz",
            dataType   : opcua.DataType.Double,
            nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""  ,         
        });
        
        
        for (var i = 0; i<14; i++){
            var nr = namespace.addVariable({
            componentOf: udtEmPz,
            browseName : i.toString(),
            dataType   : opcua.DataType.Double,
            nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""+ "[" + i + "]",
             
            
        }
        ) 
        
        var rPzTemp = namespace.addVariable({
            componentOf: nr,
            browseName : "rPzTemp",
            dataType   : opcua.DataType.String,
            nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""+ "[" + i + "]." + "\"rPzTemp\""
           
        }
        )
        
        
        var rSet= namespace.addVariable({
            componentOf: rPzTemp,
            browseName : "rSet",
            dataType   : opcua.DataType.Float,
            nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""+ "[" + i + "]." + "\"rPzTemp\".\"rSet\"",
            value      : {
                get: function () {
                    return new opcua.Variant({dataType: opcua.DataType.Float, value: serverValues[rSet]});
                    
                },
                set: function (variant) {
                    serverValues[rSet] = parseFloat(variant.value);
                    return opcua.StatusCodes.Good;
                    }
                                     
                },
            }
        )
        var rAct= namespace.addVariable({
            componentOf: rPzTemp,
            browseName : "rAct",
            dataType   : opcua.DataType.Double,
            nodeId     : "ns=1;s=" + "ZEEX_3111_Hmi.udtEmPz"+ "[" + i + "]" + "rPzTemp.rAct",
        }
        )

    }*/
              
    }
}

AndreasHeine, 2023-03-03:
### reformatted + comments

```js
module.exports = {
run: function (namespace, device, opcua, verbose, serverValues) {

// at this point you have a namespace as input param and the namespace-index of that instance you cant change! if you want to add to different namespaces within the run function you need to pass addreassSpace instead of namespace

/*
function (addressSpace, device, opcua, verbose, serverValues) { 

// get the correct namespace instance associated with 'http://mynamespace-2/UA/':
const namespace2 = addressSpace.registerNamespace('http://mynamespace-2/UA/')
// if namespace2 already exists use getNamespace instead of registerNamespace:
// const namespace2 = addressSpace.getNamespace('http://mynamespace-2/UA/')

var objInNs2 = namespace2.addObject({
        organizedBy: device,
        browseName: "ObjInNs2",
        nodeId: "s=ObjInNs2", // what mika mentioned!
    })

// get the correct namespace instance associated with 'http://mynamespace-3/UA/':
const namespace3 = addressSpace.registerNamespace('http://mynamespace-3/UA/')
// if namespace3 already exists use getNamespace instead of registerNamespace:
// const namespace3 = addressSpace.getNamespace('http://mynamespace-3/UA/')

var objInNs3 = namespace3.addObject({
        organizedBy: device,
        browseName: "ObjInNs3",
        nodeId: "s=ObjInNs3", // what mika mentioned!
    })

}
*/

    // create ppC_Compound_V3 object
    var ppC_Compound_V3 = namespace.addObject({
        organizedBy: device,
        browseName: "ppC_Compound_V3",
        nodeId: "ns=2;s=PLC",
        description: "The PLC instance which supports you with OPC UA functionality" 
    })

  /*  // create DatablocksGlobal object
    var datablocksGlobal = namespace.addObject({
        organizedBy: ppC_Compound_V3,
        browseName: "DatablocksGlobal",
        nodeId: "ns=1;s=DatablocksGlobal"
    });

    // create ZEEX_3111_Hmi object
    var prefix = namespace.addObject({
        organizedBy: datablocksGlobal,
        browseName: "ZEEX_3111_Hmi",
        //nodeId: "ns=1;s=" + "\"ZEEX_3111_Hmi\""
    });

     // create udtEmPz variable undefined Datatype Equipment Module Process Zone
    var udtEmPz = namespace.addVariable({
        componentOf: prefix,
        browseName : "udtEmPz",
        dataType   : opcua.DataType.Double,
        nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""  ,         
    });
    
    
    for (var i = 0; i<14; i++){
        var nr = namespace.addVariable({
        componentOf: udtEmPz,
        browseName : i.toString(),
        dataType   : opcua.DataType.Double,
        nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""+ "[" + i + "]",
         
        
    }
    ) 
    
    var rPzTemp = namespace.addVariable({
        componentOf: nr,
        browseName : "rPzTemp",
        dataType   : opcua.DataType.String,
        nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""+ "[" + i + "]." + "\"rPzTemp\""
       
    }
    )
    
    
    var rSet= namespace.addVariable({
        componentOf: rPzTemp,
        browseName : "rSet",
        dataType   : opcua.DataType.Float,
        nodeId     : "ns=1;s=" + "\"ZEEX_3111_Hmi\".\"udtEmPz\""+ "[" + i + "]." + "\"rPzTemp\".\"rSet\"",
        value      : {
            get: function () {
                return new opcua.Variant({dataType: opcua.DataType.Float, value: serverValues[rSet]});
                
            },
            set: function (variant) {
                serverValues[rSet] = parseFloat(variant.value);
                return opcua.StatusCodes.Good;
                }
                                 
            },
        }
    )
    var rAct= namespace.addVariable({
        componentOf: rPzTemp,
        browseName : "rAct",
        dataType   : opcua.DataType.Double,
        nodeId     : "ns=1;s=" + "ZEEX_3111_Hmi.udtEmPz"+ "[" + i + "]" + "rPzTemp.rAct",
    }
    )

}*/
          
}
}
```

### cleaned up:

```js
function (addressSpace, device, opcua, verbose, serverValues) { 

// get the correct namespace instance associated with 'http://mynamespace-2/UA/':
const namespace2 = addressSpace.registerNamespace('http://mynamespace-2/UA/')
// if namespace2 already exists use getNamespace instead of registerNamespace:
// const namespace2 = addressSpace.getNamespace('http://mynamespace-2/UA/')

var objInNs2 = namespace2.addObject({
        organizedBy: device,
        browseName: "ObjInNs2",
        nodeId: "s=ObjInNs2", // what mika mentioned!
    })

// get the correct namespace instance associated with 'http://mynamespace-3/UA/':
const namespace3 = addressSpace.registerNamespace('http://mynamespace-3/UA/')
// if namespace3 already exists use getNamespace instead of registerNamespace:
// const namespace3 = addressSpace.getNamespace('http://mynamespace-3/UA/')

var objInNs3 = namespace3.addObject({
        organizedBy: device,
        browseName: "ObjInNs3",
        nodeId: "s=ObjInNs3", // what mika mentioned!
    })

}
```

### hope its now clear!

Newinthistopic, 2023-03-03:
[image]

Newinthistopic, 2023-03-03:
[image]