node-opcua/node-opcua #1249: CAN I USE NI OPC SERVER to Connect?

top58525785, 2023-03-10:
HI 
I AM USING another NI OPC SERVER SOFTWARE
DOES IT SUPPORT NODE-OPCUA TO CONNECT?

[image]

i meet such problem

2023-03-10T06:44:45.092Z [ERR]  'OPC01' connect failure! TypeError: Cannot read properties of undefined (reading 'position')
2023-03-10T06:44:45.092Z [ERR]  'OPC01' try to connect error! TypeError: Cannot read properties of undefined (reading 'position')

What does it mean?

AndreasHeine, 2023-03-10:
should be possible! might be a bug in your code (position is undefined)

top58525785, 2023-03-10:
Thanks for your reply ! what do you mean position is undefined?

AndreasHeine, 2023-03-10:
`2023-03-10T06:44:45.092Z [ERR] 'OPC01' connect failure! TypeError: Cannot read properties of undefined (reading 'position')`

nodejs tries to read the variable 'position' but it looks like its not existing or not available at the point reading...

https://stackdiary.com/guides/typeerror-cannot-read-property-of-undefined/
```
The Cannot read property '<property>' of undefined error is thrown when you try to access a property of an undefined value. This typically happens when you try to access a property of a variable that hasn't been assigned a value.

A few other common scenarios include:

    You are trying to access a property of an object that is undefined.
    You are trying to access an index of an array that is out of bounds.
    You are trying to call a function that is undefined.
    You are trying to access a variable that is undefined.

```

erossignon, 2023-03-12:
NI OPC SERVER uses old OPC DA protocol based on Microsoft DCOM.

You would need to purchase the LabView OPC UA Option from Lab view to get the OPC UA  feature.

NodeOPCUA doesn't support OPC , but only OPCUA.

endpoint for opcua are in the form :    opc.tcp://opcuademo.sterfive.com:26543  ,

I suspect you are trying to pass to the connect method a OPC endpoint instead, hence the error.

closed 2023-03-12 by erossignon