node-opcua/node-opcua #1255: How to limit number of sessions created for opcuaclient and what is the default count of sessions we can create with one connection

raptorsj, 2023-03-21:


AndreasHeine, 2023-03-21:
default 100, iguess!?

if you are the opc ua server you can set it in the operational limits in OPCUAServerOptions -> ServerCapabilities -> OperationLimits -> "maxSessions"

e.g:
https://github.com/AndreasHeine/SampleServer-node-opcua/blob/32bebf1c08e5a04c5f0dc4ee6bb809f7c859bb55/src/config.ts#L112

if you are the client you cant do anything about it ^^

erossignon, 2023-03-24:
BadTooManySessions error returned by the server usually indicates that a client application is crashing  or stopping the connection without properly closing the connection (calling disconnect) and hence helping the serv er to free resources.

A server is designed to keep the session resources up to date until the negotiated session timeout elapse, in case the client is able to recover the connection.

erossignon, 2023-03-24:
1 connection =  0 or  1 session

raptorsj, 2023-03-28:
Thank you @erossignon and @AndreasHeine . It worked fine with your solution