node-opcua/node-opcua #1258: No session_closed when session expire

pankleks, 2023-03-26:
Hello,

I'm wondering if this is designed behaviour.

My connection settings are:
```
keepSessionAlive: false,
requestedSessionTimeout: 10000
```

So I would expect following behaviour:
1. session gets timeout after 10s
2. I get "session_closed" event
3. I can then decide to reopen session if needed

However today I have no means to determine if session has timedout.

I seen #289 but - I don't want keep alive, I want idle session to timeouts.

Thanks:)

pankleks, 2023-03-27:
thanks for the answer, but my problem is that, even though session is closed by server, I don't receive following event:

```
session.on("session_closed", () => {
})
```

AndreasHeine, 2023-03-28:
hm the thing is you don't have any communication via the session or in other words how should the server inform the client that he has closed if there is no request to answer...

erossignon, 2023-05-05:
@pankleks I would recommand that you simply close the session once you finish using it.
Some servers limits the number of concurrent sessions that can be opened, and it is always a good idea to free them up when done.

Alternatively, you can create a subscription to keep the session permanently active.

closed 2023-05-05 by erossignon