node-opcua/node-opcua #1262: The connection may have been rejected by server, Err = (Invalid message header detected)

raptorsj, 2023-04-04:
i'm getting this error while connecting to the device. what is the meaning of this error

ahermosa-monom, 2023-04-14:
I'm getting the same error after everything was working before an electrical shutdown.
```
./node_modules/.bin/simple_client  -e "opc.tcp://192.168.150.150:4840" --securityMode SignAndEncrypt --securityPolicy Basic256Sha256 -n "ns=2;s=S7_343.50"
securityMode        =  3
securityPolicy      =  http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
timeout             =  20000
 monitoring node id =  NodeId { identifierType: 2, value: 'S7_343.50', namespace: 2 }
 connecting to  opc.tcp://192.168.150.150:4840
    strategy {
  initialDelay: 2000,
  maxDelay: 10000,
  maxRetry: 10,
  randomisationFactor: 0
}
12:24:59.678Z :client_secure_channel_layer   :522   requestData not found for requestId =  1 try with  2
 Cannot connect to opc.tcp://192.168.150.150:4840
 Error =  The connection may have been rejected by server,
Err = (Invalid message header detected)

```
UAExpert still works with the same config.

This is the captured packet from running simple client to browse a specific node:
[image]

ahermosa-monom, 2023-05-04:
@erossignon can you specify which additional info is required? 
I can provide the needed information

erossignon, 2023-05-04:
*Describe the bug**

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. ....
2. ....
3. ....

**Expected behavior**

**Screenshots**

**Gist**

 
     Whenever possible, provide a self-contain script file that reproduces the issue. 
     You can create a gist here: https://gist.github.com/ and share the link

gist:

     Please, provide a full script or program that can help reproduce the issue.

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) docker/K8s environment : under OS .....
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :

      provide the semver version numero of node-opcua

- **Node:**

       `node --version` =

ahermosa-monom, 2023-05-05:
Just a quick update, I ran npm update for this library and the error log now shows a new line:
```
$ ./node_modules/.bin/simple_client  -e "opc.tcp://192.168.150.150:4840" --securityMode SignAndEncrypt --securityPolicy Basic256Sha256 -n "ns=2;s=S7_343.140"  -x ~/private_key.pem -c ~/cert.pem --debug
securityMode        =  3
securityPolicy      =  http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
timeout             =  20000
 monitoring node id =  NodeId { identifierType: 2, value: 'S7_343.140', namespace: 2 }
 connecting to  opc.tcp://192.168.150.150:4840
    strategy {
  initialDelay: 2000,
  maxDelay: 10000,
  maxRetry: 10,
  randomisationFactor: 0
}
10:39:05.988Z :client_secure_channel_layer   :522   requestData not found for requestId =  1 try with  2
10:39:06.007Z :client_base_impl              :702   Inner create secure channel has failed Invalid message header detected
 Cannot connect to opc.tcp://192.168.150.150:4840
 Error =  The connection may have been rejected by server,
Err = (Invalid message header detected)

```

Vjb80, 2023-09-07:
I am also facing this problem and have done few testing using different option.

Client 1: Linux with node red version 3.0.2 and node-red-contrib-opcua version 0.2.306
Client 2: windows with node red version 3.0.2 and node-red-contrib-opcua version 0.2.313

OPC UA Server 1: Security: None
OPC UA Server 2: Security: Basic128Rsa15 (Sign, Sign & Encrypt), Basic256 (Sign, Sign & Encrypt)
OPC UA Server 3: Security: None, Basic128Rsa15 ( Sign & Encrypt), Basic256Sha256 ( Sign & Encrypt)
OPC UA Server 4: Security: None, Basic128Rsa15 (Sign, Sign & Encrypt), Basic256 (Sign, Sign & Encrypt)

I am able to establish connection with Security policy None, Basic256Sha256.   

The problem seem to be only with Basic128Rsa15 and Basic256.  I could see the client produces the certificate to the server, which I can see and trust it.  I don't see the server certificate accepted / rejected.

client_secure_channel_layer   :234   requestData not found for requestId =  1 try with  2
Client connect error: The connection may have been rejected by server,
Err = (Invalid message header detected)

erossignon, 2023-09-07:
THis forum is for node-opcua not node-red-contrib-opcua; 

if you have any doubt, you need to provide a way to reproduce the issue with node-opcua only:
 
   Whenever possible, provide a self-contain script file that reproduces the issue. 
   You can create a gist here: https://gist.github.com/ and share the link

Please post  node-red-contrib-opcua related issue to  node-red-contrib-opcua

Vjb80, 2023-09-09:
Hi,
Thank you for your script.  I posted here as I thought the problem was not necessarily library dependent.  Every small cent helps.

I managed to identify the problem.  The issue was caused due to the time synchronization.   Client and server times drifted and after the restart causing issue.  Synchronizing the time between the server and helped solved the problem.

closed 2024-03-06 by erossignon

EirikVea, 2024-03-07:
Here is some log output with NODEOPCUADEBUG=CLIENT{REQUEST-RESPONSE-TRACE-STATS-PERF} leading up to the crash:

`23:29:18:042   >>>>>> ------ C   2 3357704583      880 PublishRequest                           77500ms                       s=    46
23:29:18.043Z :utils                         :212   ------------------------------------- Client Sending a request   PublishRequest h= 880  channel id  3357704583  securityToken= 1
23:29:18.043Z :utils                         :214   { /*PublishRequest*/
 ...                                                 requestHeader                 /*  RequestHeader                             */: {
 ...                                                   authenticationToken         /*  NodeId                                    */: ns=0;i=2346899826
 ...                                                   timestamp                   /*  DateTime                                  */: 2024-03-07T23:29:18.042Z
 ...                                                   requestHandle               /*  UInt32                                    */: 880               0x370
 ...                                                   returnDiagnostics           /*  UInt32                                    */: 0               0x0
 ...                                                   auditEntryId                /*  String                                    */:
 ...                                                   timeoutHint                 /*  UInt32                                    */: 77500               0x12ebc
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                 subscriptionAcknowledgements  /*  SubscriptionAcknowledgement           []  */: [
 ...                                                   { /* 0 - SubscriptionAcknowledgement*/
 ...                                                     subscriptionId            /*  UInt32                                    */: 3357535158               0xc81febb6
 ...                                                     sequenceNumber            /*  UInt32                                    */: 1618               0x652
 ...                                                   }
 ...                                                 ]
 ...                                                };
23:29:18:255   <<<<<< ------ C   2 3357704583      876 PublishResponse                          DataChangeNt° 2  seq#=1619    s=   130  Good (0x00000000) 
23:29:18.256Z :utils                         :220   { /*PublishResponse*/
 ...                                                 responseHeader                /*  ResponseHeader                            */: {
 ...                                                   timestamp                   /*  DateTime                                  */: 2024-03-07T23:28:03.472Z
 ...                                                   requestHandle               /*  UInt32                                    */: 876               0x36c
 ...                                                   serviceResult               /*  StatusCode                                */: Good (0x00000000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32                                     */: null
 ...                                                 symbolicId                    /*  Int32                                     */: -1
 ...                                                 locale                        /*  Int32                                     */: -1
 ...                                                 localizedText                 /*  Int32                                     */: -1
 ...                                                 additionalInfo                /*  String                                    */: null
 ...                                                 innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 ...                                                };
 ...                                                   stringTable                 /*  String                                []  */: [ /* empty*/ ]
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                 subscriptionId                /*  UInt32                                    */: 3357535158               0xc81febb6
 ...                                                 availableSequenceNumbers      /*  UInt32                                []  */: [ /* length =2*/
 ...                                                   758,
 ...                                                       1619
 ...                                                 ]
 ...                                                 moreNotifications             /*  Boolean                                   */: false
[... cut ...]
    at MessageBuilder._append (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:267:25)
    at MessageBuilder._feed_messageChunk (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:318:22)
    at PacketAssembler.<anonymous> (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:155:66)
    at PacketAssembler.emit (node:events:514:28)
    at PacketAssembler.emit (node:domain:488:12)
    at PacketAssembler.feed (C:\my-project\node_modules\node-opcua-packet-assembler\source\packet_assembler.ts:127:18)`

EirikVea, 2024-03-07:
Attached is some console output leading up to the crash.
[node-opcua-crash.log](https://github.com/node-opcua/node-opcua/files/14531235/node-opcua-crash.log)

And a stack trace (from another incident):
C:\my-project\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:534
                const callback = requestData.callback;
                                             ^
TypeError: Cannot read properties of undefined (reading 'callback')
    at MessageBuilder.<anonymous> (C:\my-project\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:534:46)
    at MessageBuilder.emit (node:events:514:28)
    at MessageBuilder.emit (node:domain:488:12)
    at MessageBuilder._report_error (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:228:14)
    at MessageBuilder._append (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:267:25)
    at MessageBuilder._feed_messageChunk (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:318:22)
    at PacketAssembler.<anonymous> (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:155:66)
    at PacketAssembler.emit (node:events:514:28)
    at PacketAssembler.emit (node:domain:488:12)
    at PacketAssembler.feed (C:\my-project\node_modules\node-opcua-packet-assembler\source\packet_assembler.ts:127:18)

erossignon, 2024-03-08:
[EirikVea](https://github.com/EirikVea) this isssue is closed, please open a new one if necessary a,d don't forget to fill the template with required information.