node-opcua/node-opcua #1263: [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 8192 nb chunks 1

ultrafab, 2023-04-05:
Trying to connect to an industrial HMI with an embedded opcua server :

`securityMode        =  1
securityPolicy      =  http://opcfoundation.org/UA/SecurityPolicy#None
timeout             =  20000
 monitoring node id =  NodeId { identifierType: 1, value: 2258, namespace: 0 }
 connecting to  opc.tcp://10.10.1.59:63840
    strategy {
  initialDelay: 2000,
  maxDelay: 10000,
  maxRetry: 10,
  randomisationFactor: 0
}
12:00:32.829Z :message_chunker               :170   [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 8192 nb chunks 1
12:00:32.832Z :client_secure_channel_layer   :854   { /*ServiceFault*/
 ...                                                 responseHeader                /*  ResponseHeader                            */: {
 ...                                                   timestamp                   /*  DateTime                                  */: 2023-04-05T12:00:32.829Z
 ...                                                   requestHandle               /*  UInt32                                    */: 0               0x0
 ...                                                   serviceResult               /*  StatusCode                                */: BadInternalError (0x80020000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32                                     */: null
 ...                                                 symbolicId                    /*  Int32                                     */: -1
 ...                                                 locale                        /*  Int32                                     */: -1
 ...                                                 localizedText                 /*  Int32                                     */: -1
 ...                                                 additionalInfo                /*  String                                    */: null
 ...                                                 innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 ...                                                };
 ...                                                   stringTable                 /*  String                                []  */: [ /* length =1*/
 ...                                                     message size exceeds the negotiated message size
 ...                                                   ]
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                };
12:00:32.834Z :client_secure_channel_layer   :855   xxxxx  <<<<<< _on_message_received  ERROR requestId= 1 Object response.responseHeader.requestHandle= 0                   ServiceFault
 Cannot connect to opc.tcp://10.10.1.59:63840
 Error =  The connection may have been rejected by server,
Err = ( serviceResult = BadInternalError (0x80020000))`

Using the Python opcua-client app or some other clients (es. uaexpert) the connection works and the server is browsable
Changing security type, anonymous, username, etc doesnt change

AndreasHeine, 2023-04-05:
Siemens OPC UA Server?

ultrafab, 2023-04-05:
Hello Andreas,

no, it should be an ASEM HMI30 device

ultrafab, 2023-04-05:
While connecting with python opcua-client the first rows in the log are:

> 
> asyncua.uaprotocol - WARNING - updating client limits to: TransportLimits(max_recv_buffer=8192, max_send_buffer=65536, max_chunk_count=1, max_message_size=8192)')
> asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
> asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
> asyncua.uaprotocol - WARNING - updating client limits to: TransportLimits(max_recv_buffer=8192, max_send_buffer=65536, max_chunk_count=1, max_message_size=8192)')
> uaclient.uaclient - INFO - Endpoint 1:')
> uaclient.uaclient - INFO -   Endpoint URL: opc.tcp://HMI30:63840')
> uaclient.uaclient - INFO -   Application URI: urn:HMI30:PremiumHMI OPCUA Server')
> uaclient.uaclient - INFO -   Product URI: urn:ASEMAutomation:PremiumHMI')
> uaclient.uaclient - INFO -   Application Name: LocalizedText(Locale='en', Text='PremiumHMI@HMI30')')
> uaclient.uaclient - INFO -   Application Type: ApplicationType.Server')
> uaclient.uaclient - INFO -   Discovery URL: opc.tcp://HMI30:63840')
> uaclient.uaclient - INFO -   Server Certificate: domainComponent=HMI30, localityName=LocationName, organizationName=ASEM, organizationalUnitName=Unit, commonName=PremiumHMI OPCUA Server (self-signed), 2023-02-14 08:35:49 - 2028-02-13 08:35:49')

AndreasHeine, 2023-04-05:
usually the "MaxMessageSize" is negotiatd in the Hello/Ack Message-Handshake right at the start (first message after connect)

which means the client requests a MaxMessageSize he can support and the server checks if he can support that as well if not (e.g. smaller) the server revices the "MaxMessageSize" in the Ack-Message as response to the Hello-Message

reference:
https://reference.opcfoundation.org/Core/Part6/v104/docs/7.1.2.3#Table51
https://reference.opcfoundation.org/Core/Part6/v105/docs/7.1

from your python log:
> asyncua.uaprotocol - WARNING - updating client limits to: TransportLimits(max_recv_buffer=8192, max_send_buffer=65536, max_chunk_count=1, max_message_size=8192)')

it tells you exactly that the requested limits ("MaxMessageSize") have been revised to 8192 (requested 65535)

AndreasHeine, 2023-04-05:
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-secure-channel/source/message_chunker.ts

https://github.com/node-opcua/node-opcua/blob/4b1dbcc0e682379cd9132d08e2413412bfbfc973/packages/node-opcua-secure-channel/source/message_chunker.ts#L169

ultrafab, 2023-04-05:
Ok, I see message in the python log. But so the problem is the opc ua client that cannot comply with the server response?

AndreasHeine, 2023-04-05:
can you adjust your node-opcua-client code

```js
client.on("send_request", console.log)
client.on("receive_response", console.log)
```

so the console output will show the requests and responses for further investigation!?

mikakaraila, 2023-04-05:
Or use environment variable:
set NODEOPCUADEBUG=CLIENT{REQUEST-RESPONSE-TRACE-STATS-PERF}

ultrafab, 2023-04-05:
Setting the environment variable the output complete is:

```
mes@server-mes:~/opcua-samples/node_modules/.bin$ ./simple_client -e "opc.tcp://10.10.1.59:63840"
securityMode        =  1
securityPolicy      =  http://opcfoundation.org/UA/SecurityPolicy#None
timeout             =  20000
 monitoring node id =  NodeId { identifierType: 1, value: 2258, namespace: 0 }
 connecting to  opc.tcp://10.10.1.59:63840
    strategy {
  initialDelay: 2000,
  maxDelay: 10000,
  maxRetry: 10,
  randomisationFactor: 0
}
17:05:00:480   >>>>>> ------ C   0   0        1 OpenSecureChannelRequest                 Issue None lt:600000ms        s=    49 
17:05:00.505Z :client_secure_channel_layer   :1782  ------------------------------------- Client Sending a request   OpenSecureChannelRequest h= 1  channel id  0  securityToken= x
17:05:00.507Z :client_secure_channel_layer   :1794  { /*OpenSecureChannelRequest*/
 ...                                                 requestHeader                 /*  RequestHeader                             */: {
 ...                                                   authenticationToken         /*  NodeId                                    */: ns=0;i=0
 ...                                                   timestamp                   /*  DateTime                                  */: 2023-04-05T17:05:00.477Z
 ...                                                   requestHandle               /*  UInt32                                    */: 1               0x1
 ...                                                   returnDiagnostics           /*  UInt32                                    */: 0               0x0
 ...                                                   auditEntryId                /*  String                                    */: null
 ...                                                   timeoutHint                 /*  UInt32                                    */: 60000               0xea60
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                 clientProtocolVersion         /*  UInt32                                    */: 0               0x0
 ...                                                 requestType                   /*  SecurityTokenRequestType                  */: 0 /*(Issue)*/
 ...                                                 securityMode                  /*  MessageSecurityMode                       */: 1 /*(None)*/
 ...                                                 clientNonce                   /*  ByteString                                */: null
 ...                                                 requestedLifetime             /*  UInt32                                    */: 600000               0x927c0
 ...                                                };
17:05:00.511Z :message_chunker               :170   [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 8192 nb chunks 1
17:05:00.513Z :client_secure_channel_layer   :854   { /*ServiceFault*/
 ...                                                 responseHeader                /*  ResponseHeader                            */: {
 ...                                                   timestamp                   /*  DateTime                                  */: 2023-04-05T17:05:00.512Z
 ...                                                   requestHandle               /*  UInt32                                    */: 0               0x0
 ...                                                   serviceResult               /*  StatusCode                                */: BadInternalError (0x80020000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32                                     */: null
 ...                                                 symbolicId                    /*  Int32                                     */: -1
 ...                                                 locale                        /*  Int32                                     */: -1
 ...                                                 localizedText                 /*  Int32                                     */: -1
 ...                                                 additionalInfo                /*  String                                    */: null
 ...                                                 innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 ...                                                };
 ...                                                   stringTable                 /*  String                                []  */: [ /* length =1*/
 ...                                                     message size exceeds the negotiated message size
 ...                                                   ]
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                };
[... cut ...]
--------------------------------------------------------------------->> Stats
   request                   :  OpenSecureChannelRequest  /  ServiceFault  -  1 / 0 BadInternalError (0x80020000)
   Bytes Read                :             0  bytes
   Bytes Written             :             0  bytes
---------------------------------------------------------------------<< Stats
 Cannot connect to opc.tcp://10.10.1.59:63840
 Error =  The connection may have been rejected by server,
Err = ( serviceResult = BadInternalError (0x80020000))
```

erossignon, 2023-04-11:
something goes wrong during the HEL/ACK transaction when the connection is established.

You'll need to set the environment variable  `NODEOPCUADEBUG` to be `TRANSPORT{HELACK-CHUNK}`

```
08:43:39.007Z :client_tcp_transport          :284   sending Hello
 ...                                                 protocolVersion   = 0
 ...                                                receiveBufferSize = 655360
 ...                                                sendBufferSize    = 655360
 ...                                                maxMessageSize    = 0
 ...                                                maxChunkCount     = 0
 ...                                                endpointUrl       = opc.tcp://opcuademo.sterfive.com:26543
 ...                                                
08:43:39.023Z :client_tcp_transport          :258   receiving Ack
 ...                                                 protocolVersion   = 0
 ...                                                receiveBufferSize = 524288
 ...                                                sendBufferSize    = 524288
 ...                                                maxMessageSize    = 16777216
 ...                                                maxChunkCount     = 32
 ...                                                
08:43:39:025    MESSGAE BUILDER LIMITS maxMessageSize =  16777216 maxChunkCount =  32 maxChunkSize =  524288 ( 16777216 )
08:43:39:031    <$$  OPN nbChunk =   1 totalLength =      132 l=     53 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:043    >$$  OPN nbChunk =   1 totalLength =      135 l=    135 s= 1    r= 1   
08:43:39:049    <$$  MSG nbChunk =   1 totalLength =      107 l=     83 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:070    >$$  MSG nbChunk =   1 totalLength =    15137 l=  15137 s= 2    r= 2   
08:43:39:076    <$$  MSG nbChunk =   1 totalLength =     1524 l=   1500 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:089    >$$  MSG nbChunk =   1 totalLength =     5520 l=   5520 s= 3    r= 3   
08:43:39:096    <$$  MSG nbChunk =   1 totalLength =      124 l=    100 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:130    >$$  MSG nbChunk =   1 totalLength =       96 l=     96 s= 4    r= 4   
08:43:39:137    <$$  MSG nbChunk =   1 totalLength =      100 l=     76 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:161    >$$  MSG nbChunk =   1 totalLength =       72 l=     72 s= 5    r= 5   
08:43:39:166    <$$  MSG nbChunk =   1 totalLength =       82 l=     58 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:166    <$$  MSG nbChunk =   1 totalLength =       82 l=     58 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:167    <$$  MSG nbChunk =   1 totalLength =       82 l=     58 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:168    <$$  MSG nbChunk =   1 totalLength =       82 l=     58 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:169    <$$  MSG nbChunk =   1 totalLength =       82 l=     58 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:172    <$$  MSG nbChunk =   1 totalLength =      146 l=    122 maxChunkCount= 32 maxMessageSize= 16777216
08:43:39:185    >$$  MSG nbChunk =   1 totalLength =       83 l=     83 s= 6    r= 11  

```

please use this minimalistic javascript code to just test the connection

```
const { OPCUAClient } = require("node-opcua");

const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";
;(async()=>{
    const client = OPCUAClient.create({});
    await client.connect(endpointUrl);
    await client.disconnect();
})();

```

AlbertoPilatoDerga, 2023-04-21:
Hello, I have the same issue

```
08:48:21.521Z :message_chunker               :120   [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 8192 nb chunks 1
08:48:21.523Z :client_secure_channel_layer   :494   { /*ServiceFault*/
 ...                                                 responseHeader                /*  ResponseHeader                            */: {
 ...                                                   timestamp                   /*  DateTime                                  */: 2023-04-21T08:48:21.522Z
 ...                                                   requestHandle               /*  UInt32                                    */: 0               0x0
 ...                                                   serviceResult               /*  StatusCode                                */: BadInternalError (0x80020000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32                                     */: null
 ...                                                 symbolicId                    /*  Int32                                     */: -1
 ...                                                 locale                        /*  Int32                                     */: -1
 ...                                                 localizedText                 /*  Int32                                     */: -1
 ...                                                 additionalInfo                /*  String                                    */: null
 ...                                                 innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 ...                                                };
 ...                                                   stringTable                 /*  String                                []  */: [ /* length =1*/
 ...                                                     message size exceeds the negotiated message size
 ...                                                   ]
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                };
08:48:21.526Z :client_secure_channel_layer   :495   xxxxx  <<<<<< _on_message_received  ERROR requestId= 1 Object response.responseHeader.requestHandle= 0                   ServiceFault
C:\Sorgenti\Derga\node-opcua-sample\node_modules\node-opcua-client\dist\private\client_base_impl.js:709
                    err = new Error("The connection may have been rejected by server,\n" + "Err = (" + err.message + ")");
                          ^

Error: The connection may have been rejected by server,
Err = ( serviceResult = BadInternalError (0x80020000))
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\node-opcua-client\dist\private\client_base_impl.js:709:27
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\node-opcua-client\dist\private\client_base_impl.js:504:25
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:375:13
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\async\dist\async.js:3681:19
    at wrapper (C:\Sorgenti\Derga\node-opcua-sample\node_modules\async\dist\async.js:271:20)
    at replenish (C:\Sorgenti\Derga\node-opcua-sample\node_modules\async\dist\async.js:441:29)
    at iterateeCallback (C:\Sorgenti\Derga\node-opcua-sample\node_modules\async\dist\async.js:430:21)
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\async\dist\async.js:327:20
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\async\dist\async.js:3679:17
    at C:\Sorgenti\Derga\node-opcua-sample\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:371:21

Node.js v18.15.0
```

closed 2023-08-17 by erossignon

szelski, 2024-01-10:
dsfgd

szelski, 2024-01-10:
I have a similar issue:

adjusting transportSettings did not help:

10:17:38:524    MESSAGE BUILDER LIMITS maxMessageSize =  16384 maxChunkCount =  1 maxChunkSize =  16384 ( 16384 )
10:17:38.527Z :message_chunker               :117   [NODE-OPCUA-E11] message size 16385 exceeds the negotiated message size 16384 nb chunks 1

erossignon, 2024-01-10:
For those who are facing this issue, we encourage you to use [`@opcua/for-node-red`](https://flows.nodered.org/node/@opcua/for-node-red),  a industrial-grade OPC UA node for node-red developed and supported by the NodeOPCUA team at Sterfive.

erossignon, 2024-01-22:
fixed in node-opcua@2.200.0