node-opcua/node-opcua #1266: OPC UA client read data from Kepware Enterprise UA server

richardlrzjw, 2023-04-11:
I am using the node-opcua to create client app to read the tag in the Kepware enterprise UA server. I am using  foillowing format to connect with server:
    function(){
        setInterval(()=>{
            let nodeIDstr="ns=2;s=Stelco.DeviceSP.Execute";
            const maxAge=0;
            const nodeToRead={
                NodeID:nodeIDstr,
                attributeId:AttributeIds.DataType
            };

            the_session.read(nodeToRead,maxAge,(err,dataValue)=>{
                if (!err){
                    console.log(`ReadValue: ${dataValue.toString()}`)
                }
            })
        },1000)
    }
  I always get  following error
$ npm start

> opcuacli@1.0.0 start
> node client.js

Connected !
ReadValue: { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}

I have checked the nodeID by using OPC Watch and copied the nodeid pasted to the code I have, still get error, what is the correct opcua read format?

full code is as below:

const {OPCUAClient,AttributeIds, TranslateBrowsePathsToNodeIdsRequest, NodeId}=require('node-opcua');
const async=require('async');

const endpointurl="opc.tcp://ftedgegw:49370/";

const client=OPCUAClient.create({endpointMustExist:false});

let the_session;

client.on("backoff",(retry,delay)=>{
    console.log(`try to connect to ${endpointurl}, retry ${retry} next attempt in $(delay/1000) seconds`);
})
//connection
async.series([
    function(callback){
        client.connect(endpointurl,(err)=>{
            if (err){
                console.log(`Cannot connect to endpoint: ${endpointurl}`);
            } else{ console.log("Connected !");}
            callback();
        })
    },
    //create session
    function(callback){
        client.createSession((err,session)=>{
            if (err) {return};
            the_session=session;
            callback();
        })
    },
    //reading from OPC UA server
    function(){
        setInterval(()=>{
            let nodeIDstr="ns=2;s=Stelco.DeviceSP.Execute";
            const maxAge=0;
            const nodeToRead={
                NodeID:nodeIDstr,
                attributeId:AttributeIds.DataType
            };

            the_session.read(nodeToRead,maxAge,(err,dataValue)=>{
                if (!err){
                    console.log(`ReadValue: ${dataValue.toString()}`)
                }
            })
        },1000)
    }
])  

OPC watch screen shot is as below:
[image]

I am using the copy nodeID function to copy the nodeid and paste in my code, always get the error  statusCode:      BadNodeIdUnknown (0x80340000)

Any idea?

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT

AndreasHeine, 2023-04-11:
@richardlrzjw 

API-Reference: https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.clientsession.html#read

"BadNodeIdUnknown" is a error response from the OPC UA Server (Kepware)!
at least the NodeId ("ns=2;s=Stelco.DeviceSP.Execute") looks like the correct format. but the error is pretty accurate. not sure if OPC Watch copies it correctly....

can you provide screenshot where the nodeid is shown in the addressspace? preferable with UaExpert the no. 1 opc ua reference client?

erossignon, 2023-04-11:
there is a typo : 

```
const nodeToRead={
                NodeID:nodeIDstr,                  <<<<< SHOULD BE nodeid: nodeIDstr

                attributeId:AttributeIds.DataType
            };
```

NodeID => nodeId

erossignon, 2023-04-11:
Also I encourage you to use the async/await technic and move away from the old callback coding style.

richardlrzjw, 2023-04-11:
How can I tell the StatusCode give me 0? I use console.log(`Write to server ${statuscodes.toString()}`); but I only see [Object Promise].

I don't have to use call back, just don't get the proper way , so I used my old way.

AndreasHeine, 2023-04-11:
write(nodesToWrite: [WriteValueOptions](https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.writevalueoptions.html)[], callback: ResponseCallback<[StatusCode](https://node-opcua.github.io/api_doc/2.32.0/classes/node_opcua.statuscode.html)[]>): void

```js

session.write(nodesToWrite, (statusCodes) => {
    // statusCodes is a Array of StatusCode-Class
    console.log(statusCodes[0].toJSON())
};
```

richardlrzjw, 2023-04-11:
it thrown a call back error.  Since I create a session called the_session, so I have to change to the session.write to the session.write otherwise I got error as 
[image]

but after I changed I got callback error:
[image]

Full code is below 

const {OPCUAClient,AttributeIds, TranslateBrowsePathsToNodeIdsRequest, NodeId}=require('node-opcua');
const async=require('async');

const endpointurl="opc.tcp://ftedgegw:49370/";

const client=OPCUAClient.create({endpointMustExist:false});

let the_session;

client.on("backoff",(retry,delay)=>{
    console.log(`try to connect to ${endpointurl}, retry ${retry} next attempt in $(delay/1000) seconds`);
})
//connection
async.series([
    function(callback){
        client.connect(endpointurl,(err)=>{
            if (err){
                console.log(`Cannot connect to endpoint: ${endpointurl}`);
            } else{ console.log("Connected !");}
            callback();
        })
    },
    //create session
    function(callback){
        client.createSession((err,session)=>{
            if (err) {return};
            the_session=session;
            callback();
        })
    },
    //write to OPC UA server
    function(){
        setInterval(()=>{
            const nodesToWrite = [{
                nodeId: "ns=2;s=Stelco.DeviceSP.Execute",
                attributeId: AttributeIds.Value,
                value: ({
                    value:({
                        value: true,
                        dataType: "Boolean"
                    }),
                    sourceTimestamp: new Date()
                })
            }];
            
            /* https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.clientsession.html#write */
            the_session.write(nodesToWrite,(statusCodes)=>{
                console.log(`Write to server ${statusCodes[0].toString()}`);
            })

        },2000)
    }
])

richardlrzjw, 2023-04-11:
@AndreasHeine Great thanks, that works , only thing I changed is comment out the sourceTimestamp: new Date(), seems the Kepware server doesn't like the combination of the value, status and timestamps provided, after I removes the timestamps, it give feedback Good and I can see all other value changed.

Thank you again.

richardlrzjw, 2023-04-14:
@AndreasHeine and @erossignon , Thanks for your past help and I have another issue popped out  which is as follow:

I tried to put the read subscription and write together to implement function read a value and decide how to write the value back, once I put them together, no matter I use same client or two different client variables, I always get error as below:
withSubscriptionAsync inner function failed  TimestampsToReturn is not defined
[image]

Here is my full code  actually basically just combined your previous code with little adjustment.

const {OPCUAClient, AttributeIds, DataType, DataValue, Variant} = require('node-opcua');

const endpointUrl="opc.tcp://ftedgegw:49370/";

const nodeId="ns=2;s=Stelco.LastestTimeStampeRecordCountSP.input_output__Count";   //Read count of the records at lastest time stampe
const nodeId2="ns=2;s=Stelco.DeviceSP.Execute";
const nodeId1="ns=2;s=Stelco.LastestTimeStampeRecordCountSP.Execute";

;(async ()=>{
 
    const clientR = OPCUAClient.create({endpointMustExist:false});

    clientR.on("backoff",(retry,delay)=>{
       console.log(`try to connect to ${endpointurl}, retry ${retry} next attempt in $(delay/1000) seconds`);
    });

    const subscriptionParameters = {
        requestedPublishingInterval: 1000,
        requestedLifetimeCount: 100,
        requestedMaxKeepAliveCount: 10,
        maxNotificationsPerPublish: 100,
        publishingEnabled: true,
        priority: 10
    };
    
    await clientR.withSubscriptionAsync(endpointUrl, subscriptionParameters, async (session, subscription) => {

        const monitoredItem = await subscription.monitor({ nodeId, attributeId: AttributeIds.Value}, { samplingInterval: 1000}, TimestampsToReturn.Both);
        monitoredItem.on("changed", (dataValue) => {
            console.log(dataValue.toString());
        });

        console.log("CTRL+C to stop");
        await new Promise((resolve)=> process.once("SIGINT",resolve));
    }); 

})();

;(async ()=>{
    
    const client = OPCUAClient.create({endpointMustExist:false});
    client.on("backoff",(retry,delay)=>{
       console.log(`try to connect to ${endpointurl}, retry ${retry} next attempt in $(delay/1000) seconds`);
    });
    

    await client.connect(endpointUrl);

    const session = await client.createSession({});  //original code is .createSession2({}) but looks like the createSeesion({}) also works
    

    setInterval(async () => {
        const nodesToWrite = [{
            nodeId: nodeId2,
            attributeId: AttributeIds.Value,
            value: new DataValue({
                value: new Variant({
                    value: true,
                    dataType: DataType.Boolean
                })
                //sourceTimestamp: new Date()
            })
        }];

        const nodesToWrite1 = [{
            nodeId: nodeId1,
            attributeId: AttributeIds.Value,
            value: new DataValue({
                value: new Variant({
                    value: true,
                    dataType: DataType.Boolean
                })
                //sourceTimestamp: new Date()
            })
        }];
        const statuscodes = await session.write(nodesToWrite)
        console.log(`statuscodes _value is  ${statuscodes[0]._value} : _name is ${statuscodes[0]._name}`);
        const statuscodes1 = await session.write(nodesToWrite1)
        console.log(`statuscodes1 _value is  ${statuscodes1[0]._value} : _name is ${statuscodes1[0]._name}`);
    }, 3000)

})();

AndreasHeine, 2023-04-14:
looks like TimestampsToReturn is not imported!

richardlrzjw, 2023-04-14:
@AndreasHeine If I run the subscription I don't have any issue which also has no TimestampsToReturn importing issue, an idea?
Below is the code only with read subscription and no issue at all.

const {OPCUAClient, AttributeIds, TimestampsToReturn} = require('node-opcua');

const endpointUrl="opc.tcp://ftedgegw:49370/";
//const nodeId="ns=2;s=Stelco.DeviceSP.Execute";
const nodeId="ns=2;s=Stelco.LastestTimeStampeRecordCountSP.input_output__Count";

;(async ()=>{
 
    const client=OPCUAClient.create({endpointMustExist:false});
    client.on("backoff",(retry,delay)=>{
       console.log(`try to connect to ${endpointurl}, retry ${retry} next attempt in $(delay/1000) seconds`);
    });

    const subscriptionParameters = {
        requestedPublishingInterval: 1000,
        requestedLifetimeCount: 100,
        requestedMaxKeepAliveCount: 10,
        maxNotificationsPerPublish: 100,
        publishingEnabled: true,
        priority: 10
    };
    
    await client.withSubscriptionAsync(endpointUrl, subscriptionParameters, async (session, subscription) => {

        const monitoredItem = await subscription.monitor({ nodeId, attributeId: AttributeIds.Value}, { samplingInterval: 1000}, TimestampsToReturn.Both);
        monitoredItem.on("changed", (dataValue) => {
            console.log(dataValue.toString());
        });

        console.log("CTRL+C to stop");
        await new Promise((resolve)=> process.once("SIGINT",resolve));
    }); 

})();

AndreasHeine, 2023-04-14:
const {OPCUAClient, AttributeIds, TimestampsToReturn} = require('node-opcua');

you need to import it in your combined code as well!

hint:
you should use Typescript, it prevents a lot of little errors!

richardlrzjw, 2023-04-14:
@AndreasHeine , good eyes. yes I  missed, after I changed to 
const {OPCUAClient, AttributeIds, DataType, DataValue, Variant,TimestampsToReturn} = require('node-opcua');
it works.
I will definitely try typescript. :-)  Thank you again.

richardlrzjw, 2023-04-14:
@erossignon @AndreasHeine , Sorry about this back and forth and reopen this again, hopefully this is the last, I just noticed there is some thing interesting things for subscription more than two points, I have tried to add one more nodeId says nodeId1, not matter how do I switch over the nodeId information , I can always see only nodeId one works, Am I doing something wrong or there is some tricky I missed? I search the API guide, could not find any info, 
[image]
[image]
After I switch the nodeId definition to another point
[image]
no matter how I can only see one point rather two, any idea?

AndreasHeine, 2023-04-14:
not sure if i understood you right... what did you expect should happen?

richardlrzjw, 2023-04-14:
I expect to get both nodeId and nodeId1 result pop out in the terminal but I always get only one which is signed with nodeId, whichever signed with nideId1 not working, I have two monitoritem, monitoritrem1, they suppose to work same.

const nodeId1="ns=2;s=Stelco.LatestRecordInfo.RecordCount.RecordCount__1_";
const nodeId="ns=2;s=Stelco.LatestRecordInfo.LastReadTime.LastReadTime__1_";

one is number and bottom one is time stamp string,  like I said, only the one signed with nodeId works not working on the same time.

Thank you for your reply first.

AndreasHeine, 2023-04-14:
maybe you should name them nodeId1 and nodeId2 and assign them like:
`.monitor({nodeId: nodeId1, attributeId: AttributeIds.Value},
                {
                    samplingInterval: 1000
                },
                TimestampsToReturn.Both)
`

just a guess to be honest...

richardlrzjw, 2023-04-14:
That actually works, great guess. now I think I am able to safely close it and take your recommend for type script.
[image]

closed 2023-04-14 by richardlrzjw

AndreasHeine, 2023-04-14:
another example where typescript would had saved you time and nerves 😜

richardlrzjw, 2023-04-14:
I will definitely try the new  journey. :-)