node-opcua/node-opcua #1268: Unsupported Openssl version 1.0.2u

faljse, 2023-04-12:
Openssl version used by node-opcua-pki is unsupported.
(https://github.com/node-opcua/node-opcua-pki/releases/download/2.14.2/openssl-1.0.2u-i386-win32.zip)

This is probably not a security issue because in this case it is only used for pki stuff.
Nevertheless, we do not want to ship unsupported software on the customers system.

Which openssl versions are compatible with node-opcua-pki?

erossignon, 2023-04-12:
*  OpenSSL 1.01 , 1.1.0 or 3 is supported by  node-opcua-pki
*  node-opcua-pki only requires opensll to generate certificate and certificate signing request. 
*  the embedded nodejs crypto module is used for encryption and security check by the opuca security 
*  on linux on mac we use the OpenSSL version already installed
*  on windows, we install a default OpenSSL exe if no OpenSSL.exe is found in the path. This OpenSSL may not be the latest openssl version.

erossignon, 2023-07-17:
newest version of node-opcua>=[v2.104.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.104.0) do not rely anymore on external openssl.exe being installed. However node-opcua needs node >=16.0

closed 2023-07-17 by erossignon