node-opcua/node-opcua #1275: Callback function for write requests on server

nicolacalgaro89, 2023-05-19:
**Is your feature request related to a problem? Please describe.**

Having the possibility to execute a callback function when a write request is received by the server from a client could be a nice feature useful for triggering actions for external processes that are linked to opcua server.

**Describe the solution you'd like**

The idea is to extend server functionality in order to emit an event every time a client write the value of a variable. Callback args should have informations related to node id of the variable involved on write request and the new value received.

**Describe alternatives you've considered**

**Additional context**

erossignon, 2023-05-20:
@nicolacalgaro89 this is a core feature that already exists. It can be achieved by using the variable binding.

DStr276, 2023-05-22:
Could the write binding be realized using a timestamped_set? Are there code examples on how to implement this? Thanks in advance!

nicolacalgaro89, 2023-05-30:
Thanks for your answer @erossignon, but I'm searching something more generic to manage write requests: in my case I've got a server that loads a large address space from an xml file and I'm interested to intercept a write call done by clients on server variables without having to recreate in memory al the variables and binding to them a setter as shown on the book "Node-OPCUA by example" at page 45. It could be more efficient if it is possible to manage the feature I want as an event in order to define a listener like this:
`    server.on("WRITE", (nodeid, value) => {
        console.log(nodeid);
        console.log(value);
    });
`
Is my approach totally wrong? In my opinion this feature avoid to write a lot of workaround code and if there is nothing already done related to this functionality I'm also interested to take advantage of professional support.

AndreasHeine, 2023-05-30:
there is a request event but it is really noisy if you have a lot of clients...

```ts
server.on('request', (request: Request, channel: ServerSecureChannelLayer) => {
    // filter for the writerequest
})
```

nicolacalgaro89, 2023-05-30:
This could be very interesting! I have to try it! Thank you very much!

closed 2024-07-16 by erossignon