node-opcua/node-opcua #1276: How to check whether client is connected or not using Noce opcua client.

raptorsj, 2023-05-24:


raptorsj, 2023-05-24:
const client = new opcua.OPCUAClient();
await client.connect(endpointurl);

if (client.isConnected) {
   console.log("client is connected"); 
} else {
   console.log("client may be trying to  reconnected"); 
}

raptorsj, 2023-05-24:
isConnected field is not available for connected client

raptorsj, 2023-05-24:
@AndreasHeine This is when we connected initially i mean for first time. what if there is an existing  connection then how can we check whether it active or not

AndreasHeine, 2023-05-24:
do you mean connection or session there is a difference 

https://reference.opcfoundation.org/Core/Part4/v104/docs/6.7

dont know for what you need the info but there are multiple layers which can break e.g. tcp connection, but session and subscription are still present and can be reused... its up to your usecase

AndreasHeine, 2023-05-26:
i usually create a Class ("OpcUaDeviceClass" or "ProxyClass") which containes a client-instance with the connection, the session and the subscription/s to that opc ua server. its rather simple to add a boolean prop. to it e.g. connected which gets written/overwritten from the client and session events.

closed 2024-07-16 by erossignon