node-opcua/node-opcua #1279: Don't pollute stdout with warnings and errors

everhardt, 2023-06-09:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I'd love for node-opcua to not pollute stdout. I think dependencies should not do that (by default). Listening to [`messageLogger`](https://github.com/node-opcua/node-opcua/blob/853d291f740cd0cb6e57c962c160c79436b62f68/packages/node-opcua-debug/source/make_loggers.ts#L142) and deal with logs there seems to be a fine alternative. This requires:
1. A setting in the `OPCUAClientOptions` and the likes or `process.env` to _enable_ writing to stdout. The setting should be adhered to in `make_errorLog` and `make_warningLog`.
2. No direct `console.log` calls (like https://github.com/node-opcua/node-opcua/blob/853d291f740cd0cb6e57c962c160c79436b62f68/packages/node-opcua-client/source/private/client_base_impl.ts#L759)

**Additional context**

N/A

everhardt, 2023-06-09:
I'm open to making a PR along the lines of the desired solution if you agree to it.

erossignon, 2023-06-10:
yes, please, use a warningLog

everhardt, 2023-06-11:
Many thanks! Are you also open to the first part of my question:

> 1. A setting in the `OPCUAClientOptions` and the likes or `process.env` to _enable_ writing to stdout. The setting should be adhered to in `make_errorLog` and `make_warningLog`.

I think the env variable is the easiest solution, I can make a PR for that if you like.

erossignon, 2023-06-30:
@everhardt debugLog are now cleaned-up. 

Sure, please  add a mechanism to enable or disable warningLog and errorLog.
warningLog and errorLog should remain active by default, as it helps developers to troubleshoot common issues and highlight areas that requires attention during development phase.

everhardt, 2023-07-17:
I'm sorry for my late reply, in the end I did not need to use this package anymore, so for now I won't be contributing.

erossignon, 2023-07-17:
@everhardt thank you for joining the initiative, and thank your for you contribution to the code. 
Let's close the issue for the time being as the bulk of the initial issue is solved.

closed 2023-07-17 by erossignon

mb-software, 2024-01-25:
Since this has been closed as completed I assume it should now be possible to disable stdout output? If so, how to do it? I couldn't find it in the code. Is there an env variable to control the output?
Thanks

GusakIurii, 2024-04-27:
Greetings.
I don't quite understand, is there a solution to disable WarningLog? 
If so, can you provide a short example?

amelielue, 2024-08-28:
Hello, I would also like to know if it is possible to disable stdout output. This package creates too many logs which I would like to avoid.

erossignon, 2024-08-28:
```typescript
import { setLogLevel, LogLevel } from "node-opcua-debug";
setLogLevel(LogLevel.Error); // default is LogLevel.Warning;
```

or 
```
// linux
NODEOPCUA_LOG_LEVEL=Error  node my-node-opcua-application.js
```

erossignon, 2024-08-28:
Please do not disregard Warning messages  as they can highlight issues you have to fix in your code.

amelielue, 2024-08-28:
Thank you for the quick response! This helped me a lot, now I can set the log-level through the env variable NODEOPCUA_LOG_LEVEL when I need it!