node-opcua/node-opcua #1281: Not able to add a device using Security Policy-  Aes256_Sha256_RsaPss

raptorsj, 2023-06-12:
When i'm trying to add a device using this security policy - **http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss**  for a device with **UserTokenType.UserName**. It is throwing the below error while creating the session with usercredential

**Error: The connection may have been rejected by server,
Err = (ClientTCP_transport26: socket has been disconnected by third party)**

raptorsj, 2023-06-12:
Here is the debug log

11:36:15.851Z :client_base_impl              :256     Client 4 EndpointFetcher_TmpClient : _internalState from     uninitialized to disconnected
11:36:15.852Z :client_base_impl              :1329  connecting to TmpClient
11:36:15.862Z :client_base_impl              :256     Client 4 EndpointFetcher_TmpClient : _internalState from     disconnected to connecting
11:36:15.862Z :client_base_impl              :178   installPeriodicClockAdjustment  0
11:36:15.863Z :client_base_impl              :678   __connectStep2
11:36:15.865Z :client_base_impl              :445   _internal_create_secure_channel creating new ClientSecureChannelLayer _internalState = connecting
11:36:15.865Z :client_base_impl              :468   _internal_create_secure_channel before secureChannel.create
11:36:15.866Z :tcp_transport                 :121   Setting socket ClientTCP_transport3 timeout =  60000
11:36:15.867Z :client_secure_channel_layer   :322   ClientSecureChannelLayer#create creating ClientTCP_transport with  transport.timeout =  60000
11:36:15.871Z :client_secure_channel_layer   :848   max Retry === 0 =>  No backoff required -> call the _connect function directly
11:36:15.878Z :client_tcp_transport          :124   ClientTCP_transport#connect(endpointUrl = opc.tcp://192.168.2.40:4840)
11:36:15.881Z :tcp_transport                 :232     TCP_transport#install => setting ClientTCP_transport3 _socket.setTimeout to  60000
11:36:15.881Z :tcp_transport                 :236     TCP_transport#_install_socket  ClientTCP_transport3
11:36:15.883Z :client_tcp_transport          :35    connected to server! 192.168.2.40:4840 timeout:60000
11:36:15.886Z :client_tcp_transport          :159   entering _on_socket_connect
11:36:15.887Z :client_tcp_transport          :288   entering _perform_HEL_ACK_transaction
11:36:15.894Z :client_tcp_transport          :243   entering _send_HELLO_request
11:36:15.897Z :client_tcp_transport          :291   before  _on_ACK_response
11:36:15.897Z :client_tcp_transport          :265   entering _on_ACK_response
11:36:15.898Z :client_secure_channel_layer   :799   Connection => err null
11:36:15.907Z :client_secure_channel_layer   :777   Client now sending OpenSecureChannel
11:36:15.911Z :client_secure_channel_layer   :962   Adjusted timeout =  15000
11:36:15.913Z :message_builder               :153    Sequence Header SequenceHeader { sequenceNumber: 1, requestId: 1 }
11:36:15.914Z :message_builder               :278   client3 Sequence Number =  1
11:36:15.918Z :message_builder               :217   client3 message size = 136       body size   = 57        requestHandle =  1 OpenSecureChannelResponse
11:36:15.919Z :client_secure_channel_layer   :969   ------------------------------------- Client receiving response  OpenSecureChannelRequest 1 OpenSecureChannelResponse err= null securityTokenId= x
11:36:15.926Z :client_secure_channel_layer   :749   Server has send a new security Token
11:36:15.927Z :message_builder               :106   id= client3 Pushing new token with id  1 [ 1 ]
11:36:15.929Z :client_secure_channel_layer   :659    time until next security token renewal =  450000 ( lifetime =  600000 -  tokenRenewalInterval =0
11:36:15.930Z :client_base_impl              :470   _internal_create_secure_channel after secureChannel.create
11:36:15.939Z :client_secure_channel_layer   :962   Adjusted timeout =  15000
11:36:15.940Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  send_request
11:36:15.941Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  send_chunk
11:36:15.944Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  receive_chunk
11:36:15.944Z :message_builder               :153    Sequence Header SequenceHeader { sequenceNumber: 2, requestId: 2 }
11:36:15.945Z :message_builder               :278   client3 Sequence Number =  2
11:36:15.946Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  receive_chunk
11:36:15.947Z :message_builder               :153    Sequence Header SequenceHeader { sequenceNumber: 3, requestId: 2 }
11:36:15.948Z :message_builder               :278   client3 Sequence Number =  3
11:36:15.949Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  receive_chunk
11:36:15.950Z :message_builder               :153    Sequence Header SequenceHeader { sequenceNumber: 4, requestId: 2 }
11:36:15.951Z :message_builder               :278   client3 Sequence Number =  4
11:36:15.952Z :message_builder               :217   client3 message size = 20876     body size   = 20804     requestHandle =  2 GetEndpointsResponse
11:36:15.953Z :client_secure_channel_layer   :969   ------------------------------------- Client receiving response  GetEndpointsRequest 2 GetEndpointsResponse err= null securityTokenId= 1
11:36:15.954Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  receive_response
11:36:15.954Z :client_base_impl              :608    Connected successfully  to  opc.tcp://192.168.2.40:4840
11:36:15.955Z :client_base_impl              :262     Client 4 EndpointFetcher_TmpClient emitting  connected
11:36:15.956Z :client_base_impl              :256     Client 4 EndpointFetcher_TmpClient : _internalState from     connecting to connected
11:36:15.957Z :client_secure_channel_layer   :962   Adjusted timeout =  15000
[... cut ...]
11:36:16.291Z :client_base_impl              :596   The connection may have been rejected by server,
 ...                                                Err = (ClientTCP_transport4: socket has been disconnected by third party)
11:36:16.300Z :client_base_impl              :262     Client 3 ClientSession emitting  connection_failed
11:36:16.301Z :client_base_impl              :256     Client 3 ClientSession : _internalState from     connecting to disconnected
11:36:16.303Z :client_secure_channel_layer   :358   abortConnection already aborting!**

AndreasHeine, 2023-06-13:
does the server support that Policy?

AndreasHeine, 2023-06-13:
another thing in the log is ->  BadCertificateUntrusted !!!

erossignon, 2023-06-13:
simple answer : [Aes256_Sha256_RsaPss](http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss) is not yet implemented nor supported in node-opcua.  you can fall back on a different security policy in in the meantime.

raptorsj, 2023-06-13:
Does this support  -  **Aes128-Sha256-RsaOaep** security policy

erossignon, 2023-06-13:
SecurityPolicy.Aes128_Sha256_RsaOaep should be supported.

raptorsj, 2023-06-13:
It is throwing an error ..  **invalid message header detected** for  security policy -SecurityPolicy.Aes128_Sha256_RsaOaep for security mode - sign and sign&encrypt

erossignon, 2023-06-14:
I just checked SecurityPolicy.Aes128_Sha256_RsaOaep connecttion between a node-opcua client and Prosys DemoServer and it works fine. 
Please contact sterfive support (contact@sterfive.com) for deeper private investigation on your setup.

raptorsj, 2023-06-16:
Thankyou @erossignon.. SecurityPolicy.Aes128_Sha256_RsaPss is working

closed 2023-07-27 by erossignon