node-opcua/node-opcua #1283: Write of Uint64/Int64  Array results in exception

Graos, 2023-06-22:
Write of Uint64 Array or subscription of Uint64 array to/from node-opcua server results in exception. Below is the snippet of the exception thrown. Uint64 and Int64 array support is not available in node-opcua server ? However Uint64 and Int64 scalar works.

**Node-opcua version used: 2.105.0**

node:internal/process/promises:288
            triggerUncaughtException(err, true /* fromPromise */);
            ^

TypeError: Cannot convert a BigInt value to a number
    at writeU_Int32LE (node:internal/buffer:693:11)
    at Buffer.writeUInt32LE (node:internal/buffer:707:10)
    at BinaryStream.writeUInt32 (\\node_modules\node-opcua-binary-stream\dist\binaryStream.js:137:21)
    at encodeUInt64 (\\node_modules\node-opcua-basic-types\dist\integers.js:151:16)
    at encodeGeneralArray (\\node_modules\node-opcua-variant\dist\variant.js:502:9)
    at encodeVariantArray (\\node_modules\node-opcua-variant\dist\variant.js:509:12)
    at encodeVariant (\\node_modules\node-opcua-variant\dist\variant.js:214:9)
    at Variant.encode (\\node_modules\node-opcua-variant\dist\variant.js:112:9)
    at encodeDataValue (\\node_modules\node-opcua-data-value\dist\datavalue.js:59:25)
    at encodeArray (\\node_modules\node-opcua-basic-types\dist\array.js:22:9)

Node.js v18.16.0

erossignon, 2023-06-30:
This is on the roadmap and currently waiting for sponsors:

* https://opencollective.com/node-opcua/projects/add-bigint-support

Graos, 2023-06-30:
Thanks for the quick update.

erossignon, 2024-11-11:
Update: we are still expecting sponsors to manifest themselve at https://opencollective.com/node-opcua/projects/add-bigint-support