node-opcua/node-opcua #1284: Update variable description dynamically

ptorrent, 2023-06-26:
Hello there !

Just a simple question, is there a way to update dynamically the description of a variable ?
```
var variable = namespace.addVariable({
		description 	: description || '',
})

```

THanks for your support !

erossignon, 2023-06-29:
In javascript:
```
variable.setDescription(coerceLocalizedText("Hello World"));
```
However I realized that the setDescription method was not yet exposed in the public typescript interface. Let me fix this.

erossignon, 2023-06-30:
typescript definition file has been fixed in [2.106.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.106.0)

ptorrent, 2023-06-30:
Thanks a lot ! Other question about displayName and description properties:

Do I've to use coerceLocalizedText("Hello World") for both property, actually I'm not ? 

I'm using Matrikon as OPC client and I'm not able to read them. On some others clients it's work

erossignon, 2023-06-30:
Both should work

`setDescription` takes a `LocalizedTextLike` argument 

LocalizedTextLike could be `string | { text: string, locale: string } | LocalizedText`

erossignon, 2023-08-17:
[node-opcua@2.109.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.109.0) now exposes BaseNode#setDiplayName and BaseNode#setDescription methods.

closed 2023-08-17 by erossignon