node-opcua/node-opcua #1285: Uncaught Error Error: Cannot find module 'chalk'

faljse, 2023-07-13:
Our project fails to execute, throwing:
```
Uncaught Error Error: Cannot find module 'chalk'

Require stack:
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space-base\dist\clone_helper.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space-base\dist\clone_options.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space-base\dist\index.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\ua_variable_impl.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\state_machine\ua_two_state_variable.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\condition_snapshot_impl.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\index.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\address_space.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\index_current.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\distHelpers\get_mini_address_space.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\distHelpers\index.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\testHelpers.js
- C:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua\dist\index.js
- C:\Users\martin\projects\f\opcuafrontend\build\ts_opcua_server.js
    at Module._resolveFilename (internal/modules/cjs/loader:995:15)
    at Module._load (internal/modules/cjs/loader:841:27)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space-base\dist\clone_helper.js:12:33)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space-base\dist\clone_options.js:4:24)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space-base\dist\index.js:20:14)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\ua_variable_impl.js:40:41)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\state_machine\ua_two_state_variable.js:18:28)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\condition_snapshot_impl.js:16:33)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\index.js:21:14)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
[... 149 lines cut ...]
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\node_modules\node-opcua\dist\index.js:114:21)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at Module.require (internal/modules/cjs/loader:1061:19)
    at require (internal/modules/cjs/helpers:103:18)
    at <anonymous> (c:\Users\martin\projects\f\opcuafrontend\src\ts_opcua_server.ts:1:1)
    at Module._compile (internal/modules/cjs/loader:1159:14)
    at Module._extensions..js (internal/modules/cjs/loader:1213:10)
    at Module.load (internal/modules/cjs/loader:1037:32)
    at Module._load (internal/modules/cjs/loader:878:12)
    at executeUserEntryPoint (internal/modules/run_main:81:12)
    at <anonymous> (internal/main/run_main_module:23:47)

```

- **node-opcua version:** :

```
    "node-gyp": "^9.4.0",
    "node-opcua": "^2.107.0",
    "node-opcua-file-transfer": "^2.107.0",
    "node-pre-gyp": "^0.17.0",
```

faljse, 2023-07-13:
> if you use typescript you need to use chalk 4...

Thx; seems to work with 4.1.2
However: I dont use chalk at all in my project - isnt node-opcua missing a dependency somewhere?

erossignon, 2023-07-17:
if you look into the "package.json" file of the "node-opcua" module, you'll see that "chalk" is specifically present in the "dependencies" section. I guess the problem is somewhere else, unfortunately, @faljse, we need more detailed information to reproduce the issue.

erossignon, 2023-07-17:
To answer the second part:
- chalk 5.3.0  is ESM only and is not supported by node-opcua, you need to add chalk 4.1.2 which works with CommonJS modules.

closed 2023-07-22 by erossignon

wolasss, 2023-07-28:
@erossignon 

in https://github.com/node-opcua/node-opcua/blob/c15331d5618effbfc4f5bb817784344acca5873f/packages/node-opcua-address-space-base/source/clone_helper.ts#L7 

`chalk` is imported, but not explicitly stated in `node-opcua-address-space-base` package.json (https://github.com/node-opcua/node-opcua/blob/c15331d5618effbfc4f5bb817784344acca5873f/packages/node-opcua-address-space-base/package.json). That is why this error is caused. Stumbled upon the same issue today. 

This is similar issue to https://github.com/node-opcua/node-opcua/issues/878, https://github.com/node-opcua/node-opcua/pull/1187.

sivanovvv, 2025-03-12:
Please fix that what @wolasss mentioned is still an issue.