node-opcua/node-opcua #1298: Separate LDS from the Client

ctapobep, 2023-09-08:
*Problem*: in the current JS API the same Client class is used to connect to OPC UI Server as well as to discover it. Which was very confusing for me as a beginner - first of all because it wasn't clear where to look for LDS logic (I naturally looked for classes with related names), and 2nd because you have to sets of unrelated methods in the same class (which leads to low cohesion). 

*Suggested solution*: these seem like 2 very different responsibilities, and to simplify things it'd be great to separate them into two different classes - one for discovery and one.. well, stays a client I guess.

*Possible alternatives*: I don't think there are any. Sure, you can make it 2 different static methods `Client.create()` and `Client.createLdsClient()`, but then they would probably still return the same class. Which means we still stay with low cohesion.

erossignon, 2023-09-08:
Dealing with DiscoveryServer require ... an OPCUA connection with the LDS.
 
Therefore it make sense to use a OPCUAClient to connect to the LDS and interact with it. 

LDS commands are session-less; so you don't need to create a session, just use the client to perform the getEndpoints(), findServer() , RegisterServer() command.

In fact those commands are already seperated from the full client class that provide session support 

```typescript
class OPCUAClientBase {
    connect(endpointUrl: string): Promise<void>;
    disconnect(): Promise<void>;
   
    getEndpoints(options?: GetEndpointsOptions): Promise<EndpointDescription[]>;
    findServers(options?: FindServersRequestLike): Promise<ApplicationDescription[]>;
    findServersOnNetwork(options?: FindServersOnNetworkRequestLike): Promise<ServerOnNetwork[]>;

}
class OPCUAClient extends OPCUAClientBase {
}
```

ctapobep, 2023-09-09:
I don't know the underlying reasons why LDS is also an OPC UA Server - these seem like very different beasts. But regardless of that I think it's quite easy to make the JS API separates these and make it much easier to grasp, maybe something like this:

```javascript
const client = new OpcClient(/*same params as today*/);
// now use the client for specific capabilities:
const lds = new OpcLds(client);
const servers = await lds.findServers(...);
const browser = new OpcBrowser(client, servers[0]); // or OpcBrowser.startSession(client, servers[0])?
```

This way the methods won't be intertwined, and these unrelated functionalities will be well separated. What do you think?

erossignon, 2024-02-25:
Closing due to inactivity

closed 2024-02-25 by erossignon