node-opcua/node-opcua #1300: BadCertificateHostNameInvalid error after adding UserManager to opcua server

aacano, 2023-09-21:
**Current behavior**

Hello, Im trying to add user authentication to my opcua server by using the userManager configuration. After adding this and changing the corresponding security variables i get my server working properly but when accesing to it through a client such as ua-expert i get an error saying BadCertificateHostNameInvalid. I can ignore this error and see that my server is working correctly but i  need to get rid of this message. 

**Describe the bug**

The bug appeared exactly when i added this userManager and this new securityPolicies. Basically this are the code changes on the server parameters configuration:

Code before:

allowAnonymous:True 
certificateFile: certificateFile
privateKeyFile: privateKeyFile 
securityPolicies: None
securityMode: None

Code After:

allowAnonymous: False 
certificateFile: certificateFile
privateKeyFile: privateKeyFile 
securityPolicies: Basic256Sha256 
securityMode: SignAndEncrypt 
userManager: userManager

As you can see the only changes are the ones mentioned before and i dont really understand why i get this error. I thought this error could be about how we create our certificates but they seem to be correct. also, the server works fine, i can see my messages on ua-expert after ignoring the error so its juts a configuration error.
I also checked if my certificates where correct and they seemed to. Also i see that my server creates 2 endpoint to connect to with the same ip address but looks like they have different acces configuration like the securityPolicies on none and on Basic256Sha256. Im not sure how to get rid of one of this enspoints cause if i eliminate de certificateFile parameter i also get the same error and the 2 endpoints.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Add UserManager to opcua server that uses certificates
2. Change securityPolicies and allowAnonymous
3. Connect to the server through UA Expert
4. Get the error BadCertificateHostNameInvalid 
5. Ignore the error and see that server works

**Expected behavior**

We should not have this error when connecting to the server and have it working properly as it is doing when we ignore the message

**Screenshots**

[image]

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (X) Linux : version : ubuntu 18.04.6
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: UA Expert
    - Version: 1.4.6
    - Manufacturer/Software vendor: Unified Automation GmbH
    - link to the manufacturer or product site: [https://](https://opcfoundation.org/)

- **node-opcua version:** :

  
  
2.113.0

- **Node:**

  `node --version` = v18.17.1

AndreasHeine, 2023-09-21:
UaExpert checks the Certificate of your server, it is trying to connect to and blaims the Certificate-Hostname, which does not match the Hostname of your OPC UA Server!

so first thing would be to check the Certificate (what hostnames are listed) and compare it against the OPC UA Server Config (hostname / alternateHostnames) ;)

aacano, 2023-10-03:
> UaExpert checks the Certificate of your server, it is trying to connect to and blaims the Certificate-Hostname, which does not match the Hostname of your OPC UA Server!
> 
> so first thing would be to check the Certificate (what hostnames are listed) and compare it against the OPC UA Server Config (hostname / alternateHostnames) ;)
> 
> edit:
> 
> its not really an issue... each OPC UA Application can check the certificate contents against the contents of the actual request (e.g. create session) to verify if an application is trustworthy or not. UaExpert just tells you that there is a inconsistency in your server which you can accept/ignore!

As you can see in the next picture, the certificate CN is exactly the same as the hostname of the opc ua server
[image]
[image]
Debugging the app i confirm the hostname is 192.168. .... same as the certificate hostname so thats why i cant understand why im getting this error. Any other ideas?

mikakaraila, 2023-10-03:
Or node-opcua certificate manager in not set to auto trust certificates...

aacano, 2023-10-03:
> ua expert just blames that the certificate has not been trusted yet... just means that its not in the trusted list! if you just click trust certificate it should be added to the trusted list and should not appear anymore 😉

yes, but when i click accept, is when i get the BadCertificateHostNameInvalid error, thats what makes it strange.

AndreasHeine, 2023-10-03:
just by revisiting the screenshots: common name is not hostname!

aacano, 2023-10-03:
> just by revisiting the screenshots: common name is not hostname!
> 
> https://reference.opcfoundation.org/Core/Part6/v105/docs/6.2.2
> 
> it must be in subjectAltname hostnames!
> 
> maybe this openssl conf file helps you: https://github.com/AndreasHeine/SecurePythonOpcUaClient/blob/master/x509v3/ssl.conf

From what i see my subjectAltName is "URI:urn:192.168....:TBOX_OPCUA_SERVER", isn´t it correct? I cant see no difference from what you send me.

erossignon, 2024-07-16:
Closing due to no activity

closed 2024-07-16 by erossignon