node-opcua/node-opcua #1302: Spaces not supported in translateBrowsePath

InstantlyMoist, 2023-09-25:
Hi, I'm doing an internship and I need to connect with OPC-UA. Your package has been wonderful so far.

Though I'm getting obstructed by one thing. I used this piece of code:

`    let browsePath = "/Objects/Root/Control Network"; 
    let actualBrowsePath = makeBrowsePath("RootFolder", browsePath);
    console.log("BrowsePath: ", JSON.stringify(actualBrowsePath, null, 2));
    let nodeId = await session.translateBrowsePath(actualBrowsePath);
    console.log(" FOund node id = ", JSON.stringify(nodeId, null, 2));`
    
And my result is:
`FOund node id =  {
  "statusCode": {
    "value": 2154758144
  },
  "targets": []
}`

While browsing root I get a correct result. Leading me to believe it's an error with supporting spaces. The namespace is 0.

Please let me know if I messed up somewhere, as i'm still learning :) 

**Expected behavior**

I would expect it to give me a targetPath

**Screenshots**

[image]
The /Objects/Root folder returns the correct nodeID for the corresponding folder. 

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - (X) other : nodejs

- Device: HP (Not sure which one)
- OS version: Windows 10

- Description of the other OPCUA system I am trying to connect to:

    - Name: 800xA
    - Version: 6.1.1

- **node-opcua version:** :

9.6.7

- **Node:**

  `node --version` = v18.17.1

InstantlyMoist, 2023-10-03:
Another useful thing might be that I've also tested putting \' and \" around in the string, also in combination with %20 which should be the url-encoded equivalent to a space. Neither of them worked.

InstantlyMoist, 2023-10-20:
Hi!

Hope you are doing well, I was wondering if I could do something to help solving this issue? I'm not too familiar with with typescript but I would be glad to help. Currently I'm falling back to nodeIds but for my internship projects using the browsePaths is a pretty big must.

AndreasHeine, 2023-10-20:
to be honest i have never seen spaces in a BrowseName!?
its uasually upper camelcase.

what server is it?
is it a common model (companion spec.)?
is it a custom own created model?

AndreasHeine, 2023-10-20:
> All BrowseNames should be upper camel case (also known as PascalCase), that is, all words written without spaces, and the first character of each word is upper case, the other characters are lower case.

https://opcconnect.opcfoundation.org/2020/09/opc-ua-modelling-best-practices/

mm-mbinder, 2023-10-20:
Hi, i experienced the same issue this week. 
In my case the server is created by a PLC, so it follows the [PLCopen companion spec](https://plcopen.org/) from my knowledge. The manufacturer of the [device](https://www.wago.com/us/controllers-bus-couplers-i-o/controller-pfc200/p/750-8212) uses the device model name some of the browse names. The manufacturer does this for all its devices from my knowledge.
[image]
When i try to use translateBrowsePath there it returns an error. 
For now i just browse recursively to resolve the nodes i need.

GoetzGoerisch, 2023-10-20:
What is typically shown in UI tools is the Displayname. Here spaces are allowed.
The BrowseName on the other hand need to follow the rules @AndreasHeine posted.

AndreasHeine, 2023-10-20:
it actually violates the OPC-F Moddelling best practises...

i assume wago uses the browsename like the displayname which is not according to the spec...

https://reference.opcfoundation.org/Core/Part3/v104/docs/5.2.4

> A [BrowseName](https://reference.opcfoundation.org/search/13?t=BrowseName) should never be used to display the name of a [Node](https://reference.opcfoundation.org/search/13?t=Node). The [DisplayName](https://reference.opcfoundation.org/search/13?t=DisplayName) should be used instead for this purpose.

mm-mbinder, 2023-10-20:
That may very well be the case. Just for completeness sake: yes, display name and browse name is the same for these nodes.
[image]
I am not really deep into OPC UA, so i can't really evaluate the stuff some device manufacturers do, but it seems to me that there are cases where browse names do include white space. For better or for worse.

AndreasHeine, 2023-10-20:
@GoetzGoerisch we need a tool to validate model compliance!? does the OPC-F Certification require model checks?

InstantlyMoist, 2023-10-30:
Would this be something that can be fixed code-wise? As I don't have any freedom over the systems I'm listening to :D

erossignon, 2023-11-15:
An immediate workaround is to build the BrowsePath datastructure directly without using the makeBrowsePath helper function.

erossignon, 2023-11-25:
@InstantlyMoist,

The issue is not related to the "space" , in fact I conform that `<space>` are properly handled in `makeBrowsePath`

In fact, `Root`and `Control Network` are browse Name that are defined in a namespace which is not the default namespace. For this reason, you need to prefix the text part with the namespace index and the ':' separator.

```diff
-let browsePath = "/Objects/Root/Control Network";
+let browsePath = "/Objects/2:Root/3:Control Network";
let actualBrowsePath = makeBrowsePath("RootFolder", browsePath); 
```

* check the namespaceIndex with your favorite OPCUA Client
* referred to https://reference.opcfoundation.org/Core/Part4/v104/docs/A.2 for more info about the browsePath syntax.

closed 2024-02-25 by erossignon