node-opcua/node-opcua #1303: Node-opcua does not seem to honor hostname parameter passed in when creating TCP server.

tetanw, 2023-09-26:
**Current behavior / Describe the bug**

When creating an OPCUAServer with a certain hostname, node-opcua will instead create a TCP server with host 0.0.0.0. Not honoring the `hostname` parameter in the `OPCUAServer`.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Copy files from gist
2. `npm install`
3. `node test.js`
4. Do a `netstat -an` in a cmd prompt
5. See that the application with the port 9999 has local address of 0.0.0.0

**Expected behavior**

I expect the OPCUAServer to honor the hostname being passed in, so if I specify 127.0.0.1 it should listen on that hostname.

**Screenshots**

[image]

**Gist**
\
gist: https://gist.github.com/tetanw/8697eb6bc7f0356d93b7b1f42715f306

**Suggested solution**

Pass the `hostname` to the `OPCUAServerEndPointclass` here https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-server/source/opcua_server.ts#L3499. And then use it in the TCP listen method here https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-server/source/server_end_point.ts#L513.

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Lenovo Thinkstation P700
- OS version:

    - (x) Windows : version : Windows 10 Enterprise, 21H2
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______
   
- Description of the other OPCUA system I am trying to connect to: NA

- **node-opcua version:** 2.113.0

  
  

- **Node:**  v18.16.0

iliareshetov, 2023-10-10:
Hi, @erossignon 
I would be interested to implement this enhancement. 
If no one else is working on it, please assign it to me.

Thank you very much!

closed 2023-12-05 by erossignon

erossignon, 2023-12-23:
fixed in node-opcua@2.119.0

PR #1314 (2023-12-05): feat: bind hostname to tcp server when passed to OPCUAServer
Closes #1303

Ensure that the OPCUAServer correctly binds to the provided hostname when creating the TCP server. 
Previously, it was defaulting to [unspecified IPv6 address](https://en.wikipedia.org/wiki/IPv6_address#Unspecified_address) (::) when IPv6 is available, or the [unspecified IPv4 address](https://en.wikipedia.org/wiki/0.0.0.0) (0.0.0.0) otherwise.

## Testing
- Added new test cases to validate the hostname parameter behavior.