node-opcua/node-opcua #1308: Method with more than one custom variable types fails on UA Expert GUI

jrk94, 2023-10-09:
**Current behavior**

I created two custom datatypes, Temp and Humid. I created an opc-ua server with a method that uses these two custom datatypes as inputs. When I try to call the method, UA Expert always assumes the first input when trying to write.

[image]
[image]

If I select Temperature it gives me as value the correct TemperatureDataType, but if I select humid, it also gives me HumidDataType

PS: this may be a bug on UA Expert side, but at least for me it is de de facto tool to valide everything is correct so I am afraid there's something missing on either the custom datatype creation or on the method registration, nevertheless I also opened a ticket in ua expert (https://forum.unified-automation.com/viewtopic.php?f=22&t=8137&p=13334#p13334)

**Describe the bug**
Error: "Failed with error 'BadInvalidArgument'"

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Create two or more custom variable types
2. Create an opc ua server with a method that has more than two custom variable types as inputs
3. Try calling it through UA Expert

**Expected behavior**

Be able to call the method through UA Expert

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - (x) Windows : version : __11____
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:
UA Expert

- **node-opcua version:** :
2.106.0

- **Node:**

  `node --version` = v18.11.0

jrk94, 2023-10-09:
I believe that in "node-opcua": "2.28.3", this was working

erossignon, 2023-10-09:
@jrk94 It would be cool if you could provide a full gist (nodeset2.xml + server.ts + package.json to facilitate investigation.

jrk94, 2023-10-11:
@erossignon I have done some more investigation and the problem was introduced in V2.60.0. Next step I will try to see if I can pinpoint to a particular PR.

I am showing in the gif two servers one with 2.60.0 and another with 2.57.0 and the code I am using for replicating this.
[image]
[image]
[ServerDemo.zip](https://github.com/node-opcua/node-opcua/files/12866591/ServerDemo.zip)

erossignon, 2023-10-31:
Creating extension objects requires additional work such as creating the Default Encoding nodes etc..., producing the BSD definition etc... 

I would recommend that you use a modeling tool to create a dedicated nodeset2 containing the full definition of your extensions object. 

This is easily achieved with the OPCUA Modeler developed by the author of node-opcua : 

https://opcua-modeler.doc.sterfive.com/fr/docs/nodeset-yaml-format/s41_data-types-extension-objects

closed 2024-03-10 by erossignon