node-opcua/node-opcua #1312: Server and Client - Examples not connecting

lazdinst, 2023-10-11:
**Current behavior**

**Describe the bug**
I can not connect the client to the server. There seems to be an issue with the host name and how its being resolved. I have tried using the `alternateHostname` and `hostname` both independently as arguments to the OPCUA Server config\; specifically as localhost, and 0.0.0.0. I want to test out the functionality of the OPC client as its likely a PLC based OPC server solution will be implemented.

I noticed that in this example https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_client_async_await.js, that the `endpointUrl` seems to point to your OPCUA demo server. This demo endpoint seems to work therefore I believe there is something wrong with my server configuration itself.

**Client**
```
➜  node-opc git:(main) ✗ npm run start:opc_client

> opc@1.0.0 start:opc_client
> node opcuaClient.js

using nodejs crypto (native)
Generated Host:  Taliss-MacBook-Pro.local
Port:  4334
opc.tcp://Taliss-MacBook-Pro.local:4334/UA/MyLittleServer
Client === Server endpointUrl true
still trying to connect to  opc.tcp://Taliss-MacBook-Pro.local:4334/UA/MyLittleServer : retry = 0 next attempt in  1.028 seconds
still trying to connect to  opc.tcp://Taliss-MacBook-Pro.local:4334/UA/MyLittleServer : retry = 1 next attempt in  2.14 seconds
^C
```

**Server**
```
➜  node-opc git:(main) ✗ npm run start:opc_server

> opc@1.0.0 start:opc_server
> NODEOPCUADEBUG=ALL nodemon opcuaServer.js

[nodemon] 3.0.1
[nodemon] to restart at any time, enter `rs`
[nodemon] watching path(s): *.*
[nodemon] watching extensions: js,mjs,cjs,json
[nodemon] starting `node opcuaServer.js`
using nodejs crypto (native)
initialized
13:53:49.313Z :namespace_impl                :1598  [NODE-OPCUA-W30 namespace#addVariable a getter has been specified and minimumSamplingInterval is missing.
 ...                                                MinimumSamplingInterval has been adjusted to 1000 ms
13:53:49.314Z :namespace_impl                :1598  [NODE-OPCUA-W30 namespace#addVariable a getter has been specified and minimumSamplingInterval is missing.
 ...                                                MinimumSamplingInterval has been adjusted to 1000 ms
Server is now listening ... ( press CTRL+C to stop)
port  4334
 the primary server endpoint url is  opc.tcp://Taliss-MacBook-Pro.local:4334/UA/MyLittleServer
```

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. npm install
2. npm run start:server
3. npm run start:client

**Expected behavior**
Sample Server and Client should be able to connect and transfer data; whithout being stuck in a `still trying to connect loop`

**Screenshots**

**Gist**
 
gist:
https://gist.github.com/lazdinst/fc20d42efa8e508e5a4c8b30dbc60d4f

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - (X) MacOs : version : Ventura 13.5.1
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: Node OPCUA Server
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
"node-opcua": "^2.115.0",
  
  

- **Node:**

  `node --version` =
v18.17.1

lazdinst, 2023-10-19:
Not resolved. Validated a client interface with MS OPC PLC Simulator.

erossignon, 2023-11-15:
Some Mac users found that this might be caused to a missing hostname defined in the /etc/host file.

Can you verify that Taliss-MacBook-Pro.local is present in the /etc/host file ?

```
127.0.0.1 Taliss-MacBook-Pro.local
```

lazdinst, 2023-11-15:
That hostname is not located in my /etc/hosts file; instead `127.0.0.1 localhost`. I'll make the changes and see if it resolves the issue and let you know what comes of it. cheers for the follow up.

erossignon, 2024-01-14:
Let's close for now, due to inactivity

closed 2024-01-14 by erossignon