node-opcua/node-opcua #1320: node-opcua will not instantiate a child variable when it has the same name as its parent objectType

tetanw, 2023-11-16:
**Current behavior / Describe the bug**

If I create an objectType named Test which has a child variable named Test then the child variable will never be instantiated. Other child variables are instantiated because I added an extra one named Test2 that is indeed instantiated.

See:
[image]

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Copy files from gist into a folder
2. Copy the standard nodeset `Opc.Ua.NodeSet2.xml` into the folder as well
3. `npm install`
4. `node server.js`
5. Open any OpcUa browser and see that child variable Test is not created.

**Expected behavior**

I expect the child variable to be created. It is not good practice to name the variable the same, but I do think this is allowed by the OpcUa specification.

**Gist**
\
gist: https://gist.github.com/tetanw/c1c2a92dbe77e20efebba90ad057a195

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Lenovo Thinkstation P700
- OS version:

    - (x) Windows : version : Windows 10 Enterprise, 21H2
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______
   
- Description of the other OPCUA system I am trying to connect to: NA

- **node-opcua version:** 2.115.0

  
  

- **Node:**  v18.16.0

tetanw, 2023-12-12:
Found the issue. The optional filter thinks it is already in while it isn't, because the `references` list contains a `HasTypeDefinition` reference with a matching `browseName`:
https://github.com/node-opcua/node-opcua/blob/734c0d9ea3a1a5bae5fec4fd795838d9847620f9/packages/node-opcua-address-space/src/_mandatory_child_or_requested_optional_filter.ts#L32-L47

I think the solution is to not check all references, but, instead only check all `Aggregates` type references here: 
https://github.com/node-opcua/node-opcua/blob/734c0d9ea3a1a5bae5fec4fd795838d9847620f9/packages/node-opcua-address-space/src/_mandatory_child_or_requested_optional_filter.ts#L26

erossignon, 2024-02-25:
Fixed in [node-opcua@2.121.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.121.0)

closed 2024-03-06 by erossignon