node-opcua/node-opcua #1322: Node-OPCUA Issue: 'Acknowledge Method Not Found' Error with Alarm Acknowledgment Functions

AcureroAdrian, 2023-11-23:
**I'm submitting a ...**

• (x) bug report
• ( ) feature request
• (x) support request

**Current behavior**
I successfully connect to my local Codesys PLC, I can see and modify values, and I am able to view alarms and see their current status.

**Describe the bug**
When using the `acknowledgeCondition` or `confirmCondition` function to acknowledge an alarm, I encounter an error saying Error: cannot find Acknowledge Method. This shouldn't be possible as I am passing the same values provided by `EventMonitor.on()`, which are the '**ConditionId**' and '**EventId**'.

**Expected behavior**
It should acknowledge the alarm and I should see it reflected in my Codesys. When using UAExpert, I can both see my alarms and acknowledge and confirm them, so I think the error is in node-opcua.

Any help is greatly appreciated!

**CODE:**
`        import express from "express"; 
        import { createServer } from 'node:http'; 
        import { AttributeIds, OPCUAClient, TimestampsToReturn, constructEventFilter, ObjectIds } from "node-opcua";         
                
                
        const app = express();
        const server = createServer(app);

        const URL = "opc.tcp://ADRIAN-ASUS:4840";

        (async () => {
        try {
             const client = OPCUAClient.create();
            client.on("backoff", (retry, delay) => {
             console.log("Retying to connect to ", URL, " attempt ", retry);
            });
       
        console.log("connecting to ", URL);
        await client.connect(URL);
        console.log("connected to ", URL);

       const session = await client.createSession();
        console.log("session initialized");

        const subscripcion = await session.createSubscription2({
            requestedPublishingInterval: 50,
            requestedMaxKeepAliveCount: 20,
            publishingEnabled: true,
        });

        const fields = [
            "EventId",
            "AckedState",
            "AckedState.Id",
            "ConfirmedState",
            "ConfirmedState.Id",
            "ConditionId",
        ];

        const eventFilter = constructEventFilter(fields);

        const itemToMonitor = {
            nodeId: ObjectIds.Server,
            attributeId: AttributeIds.EventNotifier,
        };

        const parameters = {
            filter: eventFilter,
            discardOldest: true,
            queueSize: 100,
        };

        const EventMonitor = await subscripcion.monitor(
            itemToMonitor,
            parameters,
            TimestampsToReturn.Both
        )

        const alarmData = {
        }

        EventMonitor.on("changed", (events) => {
            for (let i = 0; i < events.length; i++) {
                alarmData[fields[i]] = events[i].value
            }
        });

        setTimeout( async () => {

            try {
                const comment = 'Comment random';

                console.log({conditionId: alarmData.ConditionId, eventId: alarmData.EventId})

                session.acknowledgeCondition(alarmData.ConditionId, alarmData.EventId, comment, (err) => {
                    console.log({ err })
                }) 
            } catch (error) {
                console.log(error)
            }

        }, 5000)

        let running = true;
        process.on("SIGINT", async () => {
            if (!running) {
                return; // avoid calling shutdown twice
            }
            console.log("shutting down client");
            running = false;
            await subscripcion.terminate();
            await session.close();
            await client.disconnect();
            console.log("Done");
            process.exit(0);
        });
    } catch (error) {
        console.log("ERROR: ", error.message);
        console.log(error);
    }

    server.listen(4000, () => {
        console.log('server running at http://localhost:4000');
    })
         })()
`

**ERROR**
[image]

      
----------------------------------------------------------------------------------------------------------------------------------------------------

**console.log({ conditionId: alarmData.ConditionId, eventId: alarmData.EventId }):**
[image]

      
Context

• (x ) my request is related to node-opcua acting as an OPCUA CLIENT
• () my request is related to node-opcua acting as an OPCUA SERVER
• (x) I have installed node-opcua as a package ( using npm install )

• Description of the other OPCUA system I am trying to connect to:
    Name: CODESYS Control Win V3 x64
    Version: 3.5.19.3 (x64)

• Device: ASUS ZenBook

• OS version:
    Windows : version : Windows 11 Pro, 10.0.22631 Build 22631

• node-opcua version: 2.115.0

• Node: `nvm use` = 20.10.0

AcureroAdrian, 2023-12-01:
While debugging the code, I observed that in the **'result'** parameter of the callback, which is passed as an argument to `this.translateBrowsePath`, the **StatusCode** is:

```typescript
 statusCode: BadNodeIdUnknown (0x80340000) {
      _description: 'The node id refers to a node that does not exist in the server address space.',
      _name: 'BadNodeIdUnknown',
      _value: 2150891520,
}
```

[image]

AcureroAdrian, 2023-12-01:
I successfully acknowledged the alarm by employing a different method, which I discovered in a node-opcua issue discussion:
[How to acknowledge alarm using Cogent DataHub #1317](https://github.com/node-opcua/node-opcua/issues/1317#:~:text=on%20Oct%2018-,Awesome,-%2C%20finally%20gettin)

However, the standard or 'correct' method for event acknowledgment proved entirely ineffective in my case. Consequently, it's fair to conclude that this issue remains unresolved.

This situation is particularly puzzling since I've been providing precisely the same parameters that worked with `session.acknowledgeCondition`.

Here's how the code currently stands:
[image]

Additionally, I am eager to share the complete code for the benefit of anyone looking to implement a similar solution.

```Typescript
import express from "express";
import { createServer } from 'node:http'
import { AttributeIds, OPCUAClient, TimestampsToReturn, constructEventFilter, ObjectIds, callConditionRefresh, MethodIds, Variant, DataType } from "node-opcua-client";

const app = express();
const server = createServer(app);

const URL = "opc.tcp://ADRIAN-ASUS:4840";

(async () => {
    try {
        const client = OPCUAClient.create();

        client.on("backoff", (retry, delay) => {
            console.log("Retying to connect to ", URL, " attempt ", retry);
        });

        console.log("connecting to ", URL);
        await client.connect(URL);
        console.log("connected to ", URL);

        const session = await client.createSession();
        console.log("session initialized");

        const subscripcion = await session.createSubscription2({
            requestedPublishingInterval: 50,
            requestedMaxKeepAliveCount: 20,
            publishingEnabled: true,
        });

        const fields = [
            "EventId",
            "AckedState",
            "AckedState.Id",
            "ConfirmedState",
            "ConfirmedState.Id",
            "ConditionId",
        ];

        const eventFilter = constructEventFilter(fields);

        const itemToMonitor = {
            nodeId: ObjectIds.Server,
            attributeId: AttributeIds.EventNotifier,
        };

        const parameters = {
            filter: eventFilter,
            discardOldest: true,
            queueSize: 100,
        };

        const EventMonitor = await subscripcion.monitor(
            itemToMonitor,
            parameters,
            TimestampsToReturn.Both
        )

        const alarmData = {
        }

        callConditionRefresh(subscripcion) // Give us all events!!!

        EventMonitor.on("changed", (events) => {
            for (let i = 0; i < events.length; i++) {
                alarmData[fields[i]] = events[i].value
            }
            console.log(alarmData)
        });

        setTimeout(async () => {
            try {

                /* session.acknowledgeCondition(alarmData.ConditionId, alarmData.EventId, comment, (err) => {
                    console.log({ err })
                }) */

                const methodToCall = {
                    objectId: alarmData.ConditionId,
                    methodId: MethodIds.AcknowledgeableConditionType_Acknowledge, // TODO: Replace to MethodIds
                    inputArguments: [
                        new Variant({ dataType: DataType.ByteString, value: alarmData.EventId }),
                        new Variant({ dataType: DataType.LocalizedText, value: "test comment!" })
                    ]
                }

                session.call(methodToCall, (err, result) => {
                    if (err) return console.log(err); // Currently not important
                    console.log("Method call result: " + JSON.stringify(result, null, 2));
                })
            } catch (error) {
                console.log(error)
            }

        }, 5000)

        let running = true;
        process.on("SIGINT", async () => {
            if (!running) {
                return; // avoid calling shutdown twice
            }
            console.log("shutting down client");
            running = false;
            await subscripcion.terminate();
            await session.close();
            await client.disconnect();
            console.log("Done");
            process.exit(0);
        });
    } catch (error) {
        console.log("ERROR: ", error.message);
        console.log(error);
    }

    server.listen(4000, () => {
        console.log('server running at http://localhost:4000');
    });
})();

```

erossignon, 2024-02-25:
It should be fixed now in [node-opcua@2.120](https://github.com/node-opcua/node-opcua/releases/tag/v2.120.0). 
@AcureroAdrian  can you confirm ?

AcureroAdrian, 2024-05-09:
> It should be fixed now in [node-opcua@2.120](https://github.com/node-opcua/node-opcua/releases/tag/v2.120.0). @AcureroAdrian can you confirm ?

I'm not working on this project right now, but I think I'll pick it up again soon. I'll confirm it later.