node-opcua/node-opcua #1323: MultiStateValueDiscrete_ValueAsText gives "Invalid" when Variable value is 0

muddasir-barnes, 2023-11-24:
1. instantiated an object which has a child node of type MultiStateValueDiscreteType. 
2. Initialized the EnumValues property in the server
[image]
3. Initialized the getter and setter functions for the MultiStateValueDiscrete Variable using bindVariable Call. 
4. When 0 is written, the expected value of ValueAsText property should be "First", however, it shows "Invalid".( see screenshot below)
[image]
5. When other values are written (1,2,3), the ValueAsText works as expected (see screenshots below)
[image] 
[image]
[image]

It seems like it works partially, only that 0 index value is not detected and value as text is set to Invalid 

#### Version
node-opcua": "^2.115.0"

erossignon, 2023-11-25:
Please provide a gist to ensure that the issue can be reproduced.

erossignon, 2023-12-23:
fixed in node-opcua@2.119.0

closed 2023-12-23 by erossignon