node-opcua/node-opcua #1326: Node-opcua inherits variables from objectType with same name as variable browseName.

tetanw, 2023-12-08:
**Current behavior / Describe the bug**

When a variable's BrowseName on one objectType matches the name of another objectType that inherits from BaseObjectType then it will take over the reference from that objectType when instantiating.

See the configuration in the Siemens modelling editor:
[image]
[image]

See how the instantiated node looks like:
[image]

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Copy files from gist into a folder
2. Copy the standard nodeset `Opc.Ua.NodeSet2.xml` into the folder as well
3. `npm install`
4. `node server.js`
5. Open any OpcUa browser and see the issue

**Expected behavior**

In the example scenario in the image I would expect only the Name variable to be there.

**Gist**

gist: https://gist.github.com/tetanw/248223774efafe9a5593dbd55f063145

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Lenovo Thinkstation P700
- OS version:

    - (x) Windows : version : Windows 10 Enterprise, 21H2
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______
   
- Description of the other OPCUA system I am trying to connect to: NA

- **node-opcua version:** 2.118.0

  
  

- **Node:**  v18.16.0

tetanw, 2023-12-08:
Might actually be related to #1320.

tetanw, 2023-12-08:
Just to clarify this only goes wrong when doing dynamic object instantiation, i.e. `objectType.instantiate(...)`. When I add a xml file with the objectType already instantiated by the Siemens OPCUA modelling editor, this just works correctly.

tetanw, 2023-12-11:
Problem is probably caused by this line of code:

https://github.com/node-opcua/node-opcua/blob/734c0d9ea3a1a5bae5fec4fd795838d9847620f9/packages/node-opcua-address-space/src/base_node_private.ts#L545

The while loop keeps iterating until it reaches `BaseObjectType` where it finds a child with the name `Other` even though `Other` is just a subType of `BaseObjectType`. I think this line needs to be changed into just searching among the nodes with a `Aggregates` relation instead of a `Child` relation.

erossignon, 2024-02-25:
fixed in [node-opcua@2.121.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.121.0)

closed 2024-03-06 by erossignon