node-opcua/node-opcua #1331: Delay after connection_reestablished before OPCUAClient.isReconnecting is back to false

ePluvinage, 2024-01-03:
**Current behavior**
Whenever the client is disconnected from the server, for exemple because the server was shutdown and is restarting, client.isReconnecting is set to true. Then, I capture the event "connection_reestablished" (also tried with "after_reconnection") to know that the connection is back on and I can resume whatever I was doing. However, if I check client.isReconnecting it is still true.

**Describe the bug**
Unless I missed something, it would make sense that once the connection was reestablished, isReconnecting is instantly set to false.

**Step by step instruction to reproduce**

Please see the code below:
`      opcuaClient
      .on("connection_reestablished", () => {
        this.logger.error("OCPUA Connection re-established.")
        this.logger.error(opcuaClient.isReconnecting);
        setTimeout(() => { 
          this.logger.error(opcuaClient.isReconnecting);
          this.logger.error("Emptying buffer...")
          this.emptyBuffer();
        }, 100);
      })
`

The first time I log opcuaClient.isReconnecting it is still true, whilst 100ms later it is finally false. I use this workaround of adding a 100ms delay for now and it works (not very clean though).

**Screenshots**

[image]
[image]

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

- **Node:**

  `node --version` = v18.17.1

erossignon, 2024-01-05:
connection_reestablished only signifies that the secure channel has been re-established.  However, the session may still need to be restored at this stage. There is much more to do to reestablish a session, rebuild the subscription, and reinstate the monitored items; only then is the reconnection process completed.

It would be best if you listened to the events emitted by the session

```
  // session_restored is raised when the session and related subscription
  // have been fully repaired after a reconnection.
  session.on("session_restored", ()=>{});
```

ePluvinage, 2024-01-05:
Thanks a lot for you answer.
 I had tried that as well and it doesn't solve the issue since: 
- Session_restored event is fired before Connection_reestablished
- When session_restored is fired, both session.isReconnecting and client.isReconnecting are still true. 

Please see the code and screen shot below : 
```
      opcuaSession.on("session_restored", () => {
        this.logger.error('Session restored')
        this.logger.error(opcuaClient.isReconnecting)
        this.logger.error(opcuaSession.isReconnecting)
      })
```
[image]

erossignon, 2024-01-05:
@ePluvinage the current behavior is that the client.isReconnecting flag goes back to true straight after the "connection_reestablished" event is raised. 
Let's change the behavior as you suggested as it make more sense that isReconnecting=true at the time the connection_reestablished event is raised.

erossignon, 2024-01-21:
fixed in [node-opcua@2.200.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.120.0)

closed 2024-01-21 by erossignon