node-opcua/node-opcua #1332: Allow the nodeIdNameSeparator in the NamespaceOptions of the node-opcua-address-space to be configurable

jrk94, 2024-01-03:
Currently, the NamespaceOptions has the nodeIdNameSeparator hardcoded. This is a problem, because when we call the installFileType in order to host a file, it is creating all the commands for the file with i.e <nodeId>-Open and we would like to have i.e <nodeId>.Open

https://github.com/node-opcua/node-opcua/blob/cb434af1ed2fbe463822898df4addc2711abc462/packages/node-opcua-address-space/src/nodeid_manager.ts#L14

https://github.com/node-opcua/node-opcua/blob/cb434af1ed2fbe463822898df4addc2711abc462/packages/node-opcua-address-space/src/nodeid_manager.ts#L190

I would like to have it configurable, where we could have it whatever separator we like.

erossignon, 2024-01-05:
It is already configuration
```

export const NamespaceOptions = {
    nodeIdNameSeparator: "-"
}

```

```

import { NamespaceOptions } from "node-ocpua";
NamespaceOptions.nodeIdNameSeparator = "whatever";
```

jrk94, 2024-01-12:
Thanks!! Sorry, stupid on my part

closed 2024-01-12 by jrk94