node-opcua/node-opcua #1334: Client is connected but no more changes are retrived from monitored items after some hours

robertsLando, 2024-01-08:
**Current behavior**

I have an OPCUA client correctly connected, everything works for some hours but then suddenly I stop receiving updates from monitored items. No error/warning shown in console except this warning that is shown immediatly after creating the subscription and starting monitoring items:

```
04:52:16.776Z :client_session_impl           :709   [NODE-OPCUA-W21] Pending transactions:  CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest
04:52:16.776Z :client_session_impl           :710   [NODE-OPCUA-W22] Warning : your opcua client is sending multiple requests simultaneously to the server CreateMonitoredItemsRequest 
...                                                  please fix your application code
```

**Describe the bug**

See above

**Step by step instruction to reproduce**

Not able to provide instructions to reproduce the issues.

**Expected behavior**

The client to keep sending uypdates of monitored items

**Screenshots**

No screenshots

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Intel NUC
- OS version: Ubuntu 22

    - ( ) Windows : version : ______
    - (x) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: Siemens
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :

  2.118.0

- **Node:**

  `node --version` =18

AndreasHeine, 2024-01-08:
maybe the lifetimecount or the max keepalivecount!

https://reference.opcfoundation.org/Core/Part4/v104/docs/5.13.1

i assume 500ms publishing interval is fast for yor count values

AndreasHeine, 2024-01-08:
you can test by just adding the server time as a item it should produce frequent notifications! if that does not help its something else 🤔

robertsLando, 2024-01-08:
Hi @AndreasHeine, thanks for your fast response.

> maybe the lifetimecount or the max keepalivecount! [reference.opcfoundation.org/Core/Part4/v104/docs/5.13.1](https://reference.opcfoundation.org/Core/Part4/v104/docs/5.13.1) i assume 500ms publishing interval is fast for yor count values

Could you explain what you mean more specifically? What values should I set for `requestedLifetimeCount` and `requestedMaxKeepAliveCount`? I have events listeners on client for `keepalive` and `keepalive_failure`, shouldn't them be triggered in case of keepalive errors?  Should I also use an higher sampling interval so? I checked PLC server setting and minumum sampling interval is set to 100ms so that should not be a problem I think.

Any other suggestions about how I could recover from this?

AndreasHeine, 2024-01-08:
sampling just means checking for changes! a notification is only generated if a item change. if there is no change after a while the keepalive counter inc's if the counter is full the subscription dies on clientside. the lifetimecount works the other way around if there is no publishrequest within a publishinterval but notifications the liftemecount inc's. if exeded the server can delete the subscription. so one defines when the server can delete the sub and the other is for the client

robertsLando, 2024-01-08:
Ok what if I want to create a subscription that never dies so?

AndreasHeine, 2024-01-08:
add ServerTime to the subscription sampleinterval 5s or so

robertsLando, 2024-01-09:
What about listening for `terminated` event and recreate the subscription if needed?

AndreasHeine, 2024-01-09:
if the subscription is terminated there is a small timewindow in which you might miss a event...

-try higher values for the counts
-add all eventlisteners for the subscription just to see what happens if you dont recv changes anymore

robertsLando, 2024-01-09:
> -try higher values for the counts

Number.MAX_SAFE_INTEGER ?

erossignon, 2024-01-11:
> What about listening for `terminated` event and recreate the subscription if needed? Also what about the warning posted above? It's not clear what I should fix on code

The subscription should never die and should help you to keep the OPC UA connection alive.

Do not try to interfere and recreate a subscription if the terminate event is raised, it might not work as you expect.

```
    requestedPublishingInterval: 500,
    requestedLifetimeCount: 1000,
    requestedMaxKeepAliveCount: 12,
```

The parameters you are using impose that the server will send a keepalive publish response every 500*12 ms = 6 seconds, if none of the variables that you are monitoring is changing.

 

Regarging 

> 04:52:16.776Z :client_session_impl           :709   [NODE-OPCUA-W21] Pending transactions:  CreateMonitoredItemsRequest CreateMonitoredItemsRequest CreateMonitoredItemsRequest [...]
04:52:16.776Z :client_session_impl           :710   [NODE-OPCUA-W22] Warning : your opcua client is sending multiple requests simultaneously to the server CreateMonitoredItemsRequest 
...                                                  please fix your application code

You may be calling `subscription.monitor` without waiting for the previous call to be completed.

Use this instead:

```typescript
const monitoredItem = await this.subscription.monitor(
			{
				nodeId: nodeId,
				attributeId: AttributeIds.Value
			},
			{
				samplingInterval: 500,
				discardOldest: true,
				queueSize: 1
			},
			TimestampsToReturn.Both,
			MonitoringMode.Reporting);

```

Alternatively, if you need to monitor multiple item in one go, use a monitored item group.
( as explain in [node-opcua by example](https://leanpub.com/node-opcuabyexample-edition2024)

robertsLando, 2024-01-12:
> The parameters you are using impose that the server will send a keepalive publish response every 500*12 ms = 6 seconds, if none of the variables that you are monitoring is changing.

Ok so would setting the keepalive to `Number.MAX_SAFE_INTEGER` be a possible solution if I want a never ending subscription then?

> You may be calling subscription.monitor without waiting for the previous call to be completed.

Thanks for that, makes sense, didn't thought that could cause problems...

> Alternatively, if you need to monitor multiple item in one go, use a monitored item group.

Was wondering what are the benefits of using the monitoItems in place of doing single monitors, is that more efficent or what?

erossignon, 2024-01-14:
> What does happen when the subscription reaches the max keepalive? 

The subscription sends an empty PublishResponse with no data change notification into it,
The client then sends a new PublishRequest, so the server always has some PublishRequest in stock to send back data change notifications.

The consequence is that there is always communication between the client and the server, the subscription remains active, and the channel doesn't time out.

Having a small maxKeepalive is beneficial, it ensure that the session will not timeout.

As a rule of thumb,  make sure that publishingInterval*maxKeepAlive < session timeout.

With carefully chosen subscription parameters, the client and server communication should always be up and running, and your subscription will never die.

https://reference.opcfoundation.org/Core/Part4/v104/docs/5.13.1

> Was wondering what are the benefits of using the monitoItems in place of doing single monitors, is that more efficent or what?

Yes, grouping monitored items is more efficient as there are fewer client<->server round trips.

> I was just wondering how I could recover from a state where the subscription is not working and needs to be refreshed someway.

Subscriptions could timeout and be discarded on the server side if the client and server cannot communicate for an extended period of time 

The Subscription is discarded on the server side, if the client hasn't been able to communicate with the server within publishingInterval * maxLifeTimecount.

You need to ensure that maxLifeTimecount is big enough, 

After a network outage, an OPCUA client reconnects automatically with the server and will automatically reconstruct any subscription that has a timeout). This requires that the client and servers are OPC UA complaints and implement the reconnection mechanism adequately.

We fixed one recently in 2.117 to fix subscription repair with some servers that do not fully implements the recommended services.

It is possible that we have still case that are not handle properly. This would requires a deeper investigation between the client and server, done under professional service.  Please reach out to me in private to see how we can investigate this.

erossignon, 2025-03-09:
Closing due to more than 12 months inactivity

closed 2025-03-09 by erossignon