node-opcua/node-opcua #1335: [NODE-OPCUA-E11] message size 16385 exceeds the negotiated message size 16384 nb chunks 1

szelski, 2024-01-10:
Problem:
I cannot establish connection to a certain OPC UA Server with my OPC UA Client. (Connections to other tested OPC UA Servers do work)
I have a similar Issue like #1263 which is already closed. But the replys there, which i tested could not help me. In my case there seems to be a mismatch of 1 Byte only: 

[NODE-OPCUA-E11] message size 16385 exceeds the negotiated message size 16384 nb chunks 1

Details (process.env.NODEOPCUADEBUG = "TRANSPORT{HELACK-CHUNK}";):

10:17:38.521Z :client_tcp_transport          :285   sending Hello
 ...                                                 protocolVersion   = 0
 ...                                                receiveBufferSize = 655360
 ...                                                sendBufferSize    = 655360
 ...                                                maxMessageSize    = 0
 ...                                                maxChunkCount     = 0
 ...                                                endpointUrl       = opc.tcp://172.18.4.25:48020
 ...
10:17:38.523Z :client_tcp_transport          :263   receiving Ack
 ...                                                 protocolVersion   = 0
 ...                                                receiveBufferSize = 16384
 ...                                                sendBufferSize    = 16384
 ...                                                maxMessageSize    = 16384
 ...                                                maxChunkCount     = 1
 ...
10:17:38:524    MESSAGE BUILDER LIMITS maxMessageSize =  16384 maxChunkCount =  1 maxChunkSize =  16384 ( 16384 )
10:17:38.527Z :message_chunker               :117   [NODE-OPCUA-E11] message size 16385 exceeds the negotiated message size 16384 nb chunks 1

erossignon, 2024-01-10:
Please forward this request to nodered-contrib-opcua forum, if you need a community answer.

Alternatively, we encourage you to use [`@opcua/for-node-red`](https://flows.nodered.org/node/@opcua/for-node-red) ,  a industrial-grade OPC UA node for node-red developed and supported by the NodeOPCUA team at Sterfive.

szelski, 2024-01-11:
Why you just rename my Issue like that?
I do not have anything to do with node-red!

erossignon, 2024-01-11:
It was mentioned the issue was similar to #1263, which relates to node-red-contrib-opcua. Hence the confusion. Sorry for this. 

Please adjust the initial issue description by filling carefully the Bug reporting template , providing  clear step by step instructions to reproduce, version of node-opcua involved, describing the third party system etc ....

szelski, 2024-01-11:
-->

**Current behavior**

I have an OPC UA Client created with the node-opcua library. The connection to other OPC UA servers generally works.
But with a specific device (labeling system), this connection does not work. However,
there are other OPC UA clients that can connect to this labeling system, as I have tested myself.
The connection works with the OPC UA client "UaExpert" from Unified Automation, and you can browse through the OPC UA structure of the labeling system.

This is evidence that an OPC UA client-server connection with the labeling system generally works.

**Describe the bug**

First, I have enabled the following for an extended debug output.

```javascript
process.env.NODEOPCUADEBUG = "TRANSPORT{HELACK-CHUNK}";
```

Then I used this connection routine.

```javascript
this.client = opcua.OPCUAClient.create({
  endpointMustExist: true, // also tried false and different transportSettings{...}
});
```

I received the following outputs.

´´´
10:17:38.521Z :client_tcp_transport :285 sending Hello
... protocolVersion = 0
... receiveBufferSize = 655360
... sendBufferSize = 655360
... maxMessageSize = 0
... maxChunkCount = 0
... endpointUrl = opc.tcp://172.18.4.25:48020
...
10:17:38.523Z :client_tcp_transport :263 receiving Ack
... protocolVersion = 0
... receiveBufferSize = 16384
... sendBufferSize = 16384
... maxMessageSize = 16384
... maxChunkCount = 1
...
10:17:38:524 MESSAGE BUILDER LIMITS maxMessageSize = 16384 maxChunkCount = 1 maxChunkSize = 16384 ( 16384 )
10:17:38.527Z :message_chunker :117 [NODE-OPCUA-E11] message size 16385 exceeds the negotiated message size 16384 nb chunks 1
´´´

Subsequently, no connection is established.

As I understand it, the client and server agree on the server's maxMessageSize of 16384 bytes. However, the client violates this agreement by sending a message with 16385 bytes (one byte more than agreed upon).

**Step by step instruction to reproduce**

Actually i followed the advice:

https://github.com/node-opcua/node-opcua/issues/1263#issuecomment-1502928573

Steps to reproduce the behavior:

1.  Activate:

```javascript
process.env.NODEOPCUADEBUG = "TRANSPORT{HELACK-CHUNK}";
```

2.  Try to connect:

```javascript
[... 130 lines cut ...]
  "type": "git",
  "url": "git://github.com/node-opcua/node-opcua.git"
  },
  "keywords": [
  "OPCUA",
  "opcua",
  "m2m",
  "iot",
  "opc ua",
  "internet of things"
  ],
  "funding": {
  "url": "https://github.com/sponsors/erossignon"
  },
  "homepage": "http://node-opcua.github.io/",
  "gitHead": "36db335391fedd39726990a1b37f7768da16466a",
  "files": [
  "dist",
  "source"
  ]
  }

- **Node:**

  `node --version` = v20.5.1

szelski, 2024-01-12:
The other OPCUA system I am trying to connect to:

Name: Alpha HSM (https://www.bluhmsysteme.com/etikettierung/etikettenspender/alpha-hsm.html)
Version: (have to look up later)
Manufacturer/Software vendor: https://s-q-f.com/

So you want to buy this system to reproduce the problem?

I think the Debug log of the connection process that i provided is the most important information.

erossignon, 2024-01-12:
I should be able to investigate without .... stay tune

erossignon, 2024-01-21:
fixed in [node-opcua@2.200.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.120.0)

closed 2024-01-21 by erossignon

szelski, 2024-01-22:
Thanks a lot, i tested it and it works now!