node-opcua/node-opcua #1342: Timestamps not updating in latest versions

mb-software, 2024-02-23:
Hi @erossignon ,

I am using the latest version (2.118.0) of node-opcua with [this](https://github.com/node-opcua/node-opcua-sampleserver) sample server.
The sourceTimestamp and/or serverTimestamp are not updating for me.
But the same is working for the node-opcua version 2.75.0.
I attached the image below.

Image for V 2.118.0 :
[image]

Image for V 2.75.0 :
[image]

Would you please check and update?

Thanks for the awesome lib !!

_Originally posted by @Akeel95 in https://github.com/node-opcua/node-opcua/issues/651#issuecomment-1852142246_

erossignon, 2024-02-25:
fixed in (node-opcua@2.121.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.121.0)

closed 2024-03-06 by erossignon

Akeel95, 2024-08-30:
Hi @erossignon 

With respect to the timestamps there are 2 problems I faced which are explained below:

1. When constant data is coming on server continuously, the source and server timestamp is not updating in data access view. But the same is updating in Attributes Section on Right side.

[image]

2. When continuous data is updating, it is not updating the server timestamp correctly. (ie. library is showing the sourceTimestamp in serverTimestamp)

[image]

In any of the case it should always change the serverTimestamp, but its not happening.

To check the scenarios I used the node-opcua library provided server [examples](https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_server.js).

I used the method to update the sourceTimestamp:
      node.**setValueFromSource**(variant, StatusCodes.Good, new Date(timestamp));

Similar issue is reported here:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/611

Please let us know about the fix or work around.

Thanks!!

erossignon, 2024-09-02:
Hi Akeel95](https://github.com/Akeel95),

This ticket is closed and resolved and unlikely to be answered by anyone anymore.
 

Consider opening an new isssue to may be get a answer from the community. 

Also whilst node-opcua is open-source, we only do support through paid services.  (https://support.sterfive.com).