node-opcua/node-opcua #1345: Slow require time in jest

BartaG512, 2024-03-01:
**Current behavior / Describe the bug**

The node-opcua package require/import takes too much time loading. 
I measure around 800ms. 

For a jest test environment with a lot of file, it adds up very quickly, since jest re-requiring it in each test file.

**Step by step instruction to reproduce**

1. Create a node project `npm init -y`
2. Install jest and node-opcua  `npm i jest node-opcua`
3. Create a simple test file with content: 
```js
console.time('loadTime');
require('node-opcua');
console.timeEnd('loadTime');

describe('Name of the group', () => {
	it('should ', () => {
                expect(true).toBe(true)
	});
});
```

6. Run test `npx jest`
7.  console.time
    loadTime: 809 ms

**Expected behavior**
Faster require times.

**Context**  
- (x) I have installed node-opcua as a package ( using npm install ) 
 
- OS version: Microsoft Windows 11 Pro 

    - (x ) Windows : version : 10.0.22631 build 22631 
    - 
- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** : "^2.122.0"

  
  

- **Node:**

  `node --version` = 20.11.0

erossignon, 2024-03-05:
node-opcua loads approximatively in 300ms on my computer

If you are only using the client API, you could gain 25% to 30% load time by just using 
"node-opcua-client" instead of "node-opcua". 
There is no need to load the server and address-space API if you are just using the Client API.

Alternatively, you could use a different test framework ( mocha or vitest)

* node-opcua is using mocha and we can parallel test effectively, 
* I am been experiencing with vitest lately and find it amazing.

BartaG512, 2024-03-06:
In integration testing I am using both client and server. 
Thanks for the tips. 

I have found a jest runtime which will disable the isolation between test files: 

https://www.npmjs.com/package/@m-abboud/hella-fast-jest-runtime

From its description:
```The core problem that this is addressing is that jest runs test suites in isolation from each other. So for every test file you have jest has to reload many modules and setup a new test env which can be very slow at least on some projects especially ones using jsdom. Running test suites in isolation is of course helpful because it helps prevent tests interfering with each other which can be a real pain to debug and track down, but I've found that this type of problem is quite rare. The trade off of having way faster tests over these occasional test bugs is way worth it to me and probably other people too.```

erossignon, 2024-03-06:
Thank for the tip. Let's close for now.

closed 2024-03-06 by erossignon