node-opcua/node-opcua #1346: After updating node-opcua, an error during client.createSession process occurs. Old versions works fine.

sdarou, 2024-03-03:
**Current behavior**

After updating to node-opcua@2.122, creating a session throws an error in node-opcua-secure-channel. 
After going back to older node-opcua@2.44 version, the session is established without any problem.

**Describe the bug**

When creating a session, the client seems not to be able to handle the response coming from the server. 
Here is the output from the console:

**Step by step instruction to reproduce**

The bug appeared in my self written script, but also in one of the examples provided: 

cd xxx/node/node_modules/node-opcua-samples/dist/
node simple_client_ts.js -e opc.tcp://192.168.xxx.xxx:4840 -P=Basic256Sha256 -s=SignAndEncrypt -u xxxx -p xxxx

**Expected behavior**

The session should be established like it did in prevous versions

**Screenshots**

$node simple_client_ts.js -e opc.tcp://192.168.xxx.xxx:4840 -P=Basic256Sha256 -s=SignAndEncrypt -u xxxx -p xxxx
using nodejs crypto (native)
securityMode        =  3
securityPolicy      =  http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
timeout             =  20000
 monitoring node id =  NodeId { identifierType: 1, value: 2258, namespace: 0 }
 connecting to  opc.tcp://192.168.xxx.xxx:4840
    strategy {
  initialDelay: 2000,
  maxDelay: 10000,
  maxRetry: 10,
  randomisationFactor: 0
}
 Connected ! exact endpoint url is  opc.tcp://192.168.xxx.xxx:4840
endpoint                       Application URI                             Product URI                      Application Name                        Security Mode   securityPolicyUri                                          Type    certificate  discoveryUrls
-----------------------------  ------------------------------------------  -------------------------------  --------------------------------------  --------------  ---------------------------------------------------------  ------  -----------  -----------------------------
opc.tcp://192.168.xxx.xxx:4840  urn:SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  https://www.siemens.com/s7-1500  SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  None            http://opcfoundation.org/UA/SecurityPolicy#None            Server  ...          opc.tcp://192.168.xxx.xxx:4840
opc.tcp://192.168.xxx.xxx:4840  urn:SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  https://www.siemens.com/s7-1500  SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  SignAndEncrypt  http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256  Server  ...          opc.tcp://192.168.xxx.xxx:4840

Identify Token for : Security Mode= 1  Policy= http://opcfoundation.org/UA/SecurityPolicy#None
policyId  tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
--------  ---------  ---------------  -----------------  --------------------------------------------------------
Username  1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15

Identify Token for : Security Mode= 3  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
policyId  tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
--------  ---------  ---------------  -----------------  --------------------------------------------------------
Username  1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15

Server Certificate :
xxxxx (Deleted, but it was recieved and worked fine)

 .... ( 1029)
adjusted endpoint Url = opc.tcp://192.168.xxx.xxx:4840
Options =  3 http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
 --------------------------------- Now connecting again to  opc.tcp://192.168.xxx.xxx:4840
 Connected ! exact endpoint url is  opc.tcp://192.168.xxx.xxx:4840
 now creating Session !
xxx/node/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:54
        err = new Error(" serviceResult = " + response.responseHeader.serviceResult.toString());

Error:  serviceResult = Bad (0x80000000)
    at process_request_callback (xxx/node/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:54:15)
    at ClientSecureChannelLayer._on_message_received (xxx/node/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:579:9)
    at MessageBuilder.<anonymous> (xxx/node/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:177:18)
    at MessageBuilder.emit (node:events:517:28)
    at MessageBuilder._decodeMessageBody (xxx/node/node_modules/node-opcua-secure-channel/dist/source/message_builder.js:216:26)
    at MessageBuilder._feed_messageChunk (xxx/node/node_modules/node-opcua-transport/dist/source/message_builder_base.js:207:40)
    at PacketAssembler.<anonymous> (xxx/node/node_modules/node-opcua-transport/dist/source/message_builder_base.js:64:66)
    at PacketAssembler.emit (node:events:517:28)
    at PacketAssembler.feed (xxx/node/node_modules/node-opcua-transport/node_modules/node-opcua-packet-assembler/dist/packet_assembler.js:75:18)
    at MessageBuilder.feed (xxx/node/node_modules/node-opcua-transport/dist/source/message_builder_base.js:94:35) {
response: ServiceFault {
    responseHeader: ResponseHeader {
      timestamp: 2024-03-03T11:04:50.345Z {
        high_low: [ 31092058, -1666849567 ],
        picoseconds: 289700000
      },
      requestHandle: 4,
      serviceResult: ConstantStatusCode {
        _value: 2147483648,
        _description: 'The operation failed.',
        _name: 'Bad'
      },
      serviceDiagnostics: DiagnosticInfo {
        symbolicId: -1,
        namespaceURI: -1,
        locale: -1,
        localizedText: -1,
        additionalInfo: null,
        innerStatusCode: ConstantStatusCode {
          _value: 0,
          _description: 'The operation succeeded.',
          _name: 'Good'
        },
        innerDiagnosticInfo: null
      },
      stringTable: [],
      additionalHeader: null
    }
  },
  request: ActivateSessionRequest {
    requestHeader: RequestHeader {
      authenticationToken: NodeId { identifierType: 1, value: 14715782, namespace: 0 },
      timestamp: 2024-03-03T10:45:22.182Z {
        high_low: [ 31092055, -463580576 ],
        picoseconds: 0
      },
      requestHandle: 4,
      returnDiagnostics: 0,
      auditEntryId: '',
      timeoutHint: 15000,
      additionalHeader: null
    },
    clientSignature: SignatureData {
      algorithm: 'http://www.w3.org/2001/04/xmldsig-more#rsa-sha256',
      signature: Buffer(256) [Uint8Array] [ xxxxx ]
    },
    clientSoftwareCertificates: [],
    localeIds: [],
    userIdentityToken: UserNameIdentityToken {
      policyId: 'Username',
      userName: 'xxxxx',
      password: Buffer(256) [Uint8Array] [ xxxxx ],
      encryptionAlgorithm: 'http://www.w3.org/2001/04/xmlenc#rsa-1_5'
    },
    userTokenSignature: SignatureData { algorithm: null, signature: null }
  }
}

**Gist**

gist:

cd xxx/node/node_modules/node-opcua-samples/dist/
node simple_client_ts.js -e opc.tcp://192.168.xxx.xxx:4840 -P=Basic256Sha256 -s=SignAndEncrypt -u xxxx -p xxxx

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: 

    - ( ) Windows : version : ______
    - (x) Linux : version : Ubuntu 20.04.6 LTS
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: SIMATIC.S7-1500.OPC-UAServer:PLC_1517F
    - Version: ______
    - Manufacturer/Software vendor: Siemens
    - link to the manufacturer or product site: https://siemens.com

- **node-opcua version:** :

2.121.0

- **Node:**

  `node --version` = node v18.19.1

sdarou, 2024-03-03:
After additional tests I realized that node-opcua@2.120.0 is the latest version where the bug does not occur, versions 2.121 and 2.122 are affected.

erossignon, 2024-03-04:
node-opcua@2.121.0 remove support for  http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 as nodejs@21.0.0 is not supporting it anymore.

Basic128Rsa15 is subject to Marvin Attack (CVE-2023-46809) because it is using PKCS#1  padding and PKCS#1 padding has been deprecated in recent nodeJS version

See:

https://github.com/nodejs/node/blob/main/doc/changelogs/CHANGELOG_V20.md#20.11.1
[[7079c062bb](https://github.com/nodejs/node/commit/7079c062bb)] - crypto: disable PKCS#1 padding for privateDecrypt (Michael Dawson) [nodejs-private/node-private#525](https://github.com/nodejs-private/node-private/pull/525)

Alternative:
  - downgrade node-opcua and wait for a future resolution
  - or activate an alternate securityPolicyUri in the token Policy on the PLC side if you can.

If you are a member of the NodeOPCUA Support subscription, you can [contact the NodeOPCUA Support team](contact@sterfive.com) to consider mitigation if the issue is pressing for you.

sdarou, 2024-03-05:
Thank you very much. I will ask our PLC programmer to change the setting and test again as soon as he is available (1-2 weeks)
Actually I´m not actively using Basic128Rsa15, but the PLC is providing it as an alternative method, so it should be a good idea to remove it anyway.

closed 2024-03-05 by sdarou

erossignon, 2024-03-05:
The settings could be changed to something like:

```

endpoint                       Application URI                             Product URI                      Application Name                        Security Mode   securityPolicyUri                                          Type    certificate  discoveryUrls
-----------------------------  ------------------------------------------  -------------------------------  --------------------------------------  --------------  ---------------------------------------------------------  ------  -----------  -----------------------------
opc.tcp://192.168.xxx.xxx:4840  urn:SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  https://www.siemens.com/s7-1500  SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  None            http://opcfoundation.org/UA/SecurityPolicy#None            Server  ...          opc.tcp://192.168.xxx.xxx:4840
opc.tcp://192.168.xxx.xxx:4840  urn:SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  https://www.siemens.com/s7-1500  SIMATIC.S7-1500.OPC-UAServer:PLC_1517F  SignAndEncrypt  http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256  Server  ...          opc.tcp://192.168.xxx.xxx:4840

Identify Token for : Security Mode= 1  Policy= http://opcfoundation.org/UA/SecurityPolicy#None
policyId  tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
--------  ---------  ---------------  -----------------  --------------------------------------------------------
Username  1          null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256

Identify Token for : Security Mode= 3  Policy= http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
policyId  tokenType  issuedTokenType  issuerEndpointUrl  securityPolicyUri
--------  ---------  ---------------  -----------------  --------------------------------------------------------
Username  1          null             null               http://opcfoundation.org/UA/SecurityPolicy#None
```

* use securityPolicyUri Basic256Sha256 for unencrypted channel (MessageSecurity.None) to ensure that password is strongly crypted when communicated to server.
* use securityPolicyUri None for encrypted channel (MessageSecurity.SignAndEncrypt) as there is no need to re-encrypt the password within an already encrypted channel.