node-opcua/node-opcua #1349: Difference in clocks between client and server cause problems

EirikVea, 2024-03-08:
**Current behavior**
When connected to a OPC UA server where the clock is out of sync with the client (node-opcua) there seems to be a problem when connecting to the device and when renewing the security token (see details below). The client reports the error message: `requestData not found for requestId =  <n> try with  <n+1>`

**Describe the bug**
How the clocks differ seems to matter, meaning if the device clock is *behind* my client's clock by more than 30 minutes, I am unable to connect at all. If the device clock is either *ahead* or *behind* my device by a couple of minutes, I am able to establish a session with the server and it works for a while, then seemingly when the security token is renewed, the error message appear and the client crashes.

**Step by step instruction to reproduce**

Immediate crash:
1. Make sure to set the clock of the device > 30 minutes *behind* that on your PC.
2. Connect to the server with message security enabled.
3. You should see the error.

Crash after security token renewal:
1. Make sure to set the clock of the device such that it differs by more than about 2 minutes from your PC.
2. Connect to the server with message security enabled.
3. Wait until security token is renewed
4. You should see the error.

**Expected behavior**
I expect the client not to crash even though the clocks between client and server are different (even by a lot). UA Expert seems to handle the situation without issue.

Log output leading up to crash
[node-opcua-crash.log](https://github.com/node-opcua/node-opcua/files/14537763/node-opcua-crash.log)

Stack trace:
```
C:\my-project\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:534
                const callback = requestData.callback;
                                             ^
TypeError: Cannot read properties of undefined (reading 'callback')
    at MessageBuilder.<anonymous> (C:\my-project\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:534:46)
    at MessageBuilder.emit (node:events:514:28)
    at MessageBuilder.emit (node:domain:488:12)
    at MessageBuilder._report_error (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:228:14)
    at MessageBuilder._append (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:267:25)
    at MessageBuilder._feed_messageChunk (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:318:22)
    at PacketAssembler.<anonymous> (C:\my-project\node_modules\node-opcua-transport\source\message_builder_base.ts:155:66)
    at PacketAssembler.emit (node:events:514:28)
    at PacketAssembler.emit (node:domain:488:12)
    at PacketAssembler.feed (C:\my-project\node_modules\node-opcua-packet-assembler\source\packet_assembler.ts:127:18)
```

**Context**

- My request is related to node-opcua acting as an OPCUA CLIENT
- I have installed node-opcua as a package ( using npm install )

My setup:
- Microsoft Windows 10 Business
- Version	10.0.19045 Build 19045
- node-opcua v2.122.0 (19397cb2584438293a85aedc5965fe294b489f18)
- `node --version` = `v20.9.0`

Device info:
- Device: Beckhoff CX5230 (although the behaviour is similar accross all Beckhoff models I've used)
- OS version: FreeBSD 14.0-RELEASE-p4 (BHF) n265489-b2dcdf4e1d1- (behaviour is observed in other versions as well)

OPCUA system I am trying to connect to:
- Name: TF6100-OPC-UA-Server
- Version: 5.2.11_1
- Manufacturer/Software vendor: Beckhoff
- link to the manufacturer or product site: [here](https://www.beckhoff.com/en-en/products/automation/twincat/tfxxxx-twincat-3-functions/tf6xxx-connectivity/tf6100.html)

AndreasHeine, 2024-03-08:
https://reference.opcfoundation.org/Core/Part4/v104/docs/5.5.2

 Clients should request a new SecurityToken after 75 % of its lifetime has elapsed. This should ensure that Clients will receive the new SecurityToken before the old one actually expires. 

in opc ua utc is used!
did you use NTP time sync?

erossignon, 2024-03-09:
> https://reference.opcfoundation.org/Core/Part4/v104/docs/5.5.2

Also interesting article:

https://reference.opcfoundation.org/Core/Part6/v105/docs/6.3

The server that cannot connect to the NTP time sync could interact with the Discovery server to update its time and correct time drift. I haven't seen this implemented

erossignon, 2024-03-10:
Should be fixed in node-opcua@2.113.0, please confim

closed 2024-03-11 by EirikVea