node-opcua/node-opcua #1353: maxStringLength

ptorrent, 2024-03-21:
Hello !

I'm not able to get data from a OPC UA:

```
:message_builder               :489   Decode message error :  maxStringLength(16777216) has been exceeded in BinaryStream.readString len=2918097563
12:39:03.655Z :message_builder               :231   cannot decode message  for valid object of type ns=0;i=634 ReadResponse
12:39:03.681Z :message_builder               :234   { /*ReadResponse*/
                                  responseHeader                /*  ResponseHeader                            */: {
                                     timestamp                   /*  DateTime                                  */: 2024-03-20T10:03:57.554Z
                                    requestHandle               /*  UInt32                                    */: 7               0x7
                                    serviceResult               /*  StatusCode                                */: Good (0x00000000)
                                    serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
                                   namespaceUri                  /*  Int32                                     */: null
                                  symbolicId                    /*  Int32                                     */: -1
                                   locale                        /*  Int32                                     */: -1
                                   localizedText                 /*  Int32                                     */: -1
                                  additionalInfo                /*  String                                    */: null
                                   innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
                                  innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
                                  };
                                    stringTable                 /*  String                                []  */: [ /* empty*/ ]
                                     additionalHeader            /*  ExtensionObject                           */: null
                                   }
                                   results                       /*  DataValue                             []  */: [ /* length =2*/

                                     { /* DataValue */
                                       value: Variant(Scalar<UInt32>, value: 10000)
                                        statusCode:      Good (0x00000000)
                                        serverTimestamp: null
                                        sourceTimestamp: null
                                    },

                                   .... TRUNCATED ..... (NODEOPCUA_DEBUG_MAXLINE_PER_MESSAGE=25
```

I'm running node-opcua  2.123.0 on node v18.19.1

Any idea ?

ptorrent, 2024-03-21:
Maybe a link with https://github.com/node-opcua/node-opcua/issues/655 ?

erossignon, 2024-03-23:
One of the variables you are crawling exceeds the built-in Buffer max Length

2918097563  = 2783 MB i.e 2.7 Gb ! That is quite a lot.

Even if you reduce the maxNodesPerRead value, this would not be enough to read the entire value in  
The old crawler doesn't handle this case. A different strategy is required.

[Sterfive (the company behind NodeOPCUA)](https://www.sterfive.com/contact/) provides a Professional Crawler module with additional options that allow you to separate browsing the address space structure from reading the variable IDs.
This might be helpful in your case.

ptorrent, 2024-03-24:
Thanks for your answer ! I will take a look into it.

closed 2024-03-24 by ptorrent