node-opcua/node-opcua #1361: Client fails to generate default certificate when connecting to server in Electron main process.

mazzal1, 2024-04-19:
Within an Electron app, after instantiating a client, trying to connect to an endpoint fails with errors related to certificates.
Starting from a bare Quickstart electron application, install node-opcua. Then import OPCUAClient and create it with default options.
Call a function that tries to connect to a server (like connect or withSessionAsync), passing any URL, e.g. `"opc.tcp://opcuademo.sterfive.com:26543"`.

The connection fails raising this error:
`
npm start

> electron-quick-start@1.0.0 start
> electron .

using nodejs crypto (native)
15:03:18.129Z :client_base_impl              :583   Creating default certificate ... please wait
ERROR!
DOMException [DataError]: Invalid keyData
    at new DOMException (node:internal/per_context/domexception:53:5)
    at __node_internal_ (node:internal/util:695:10)
    at Object.rsaImportKey (node:internal/crypto/rsa:233:15)
    at SubtleCrypto.importKey (node:internal/crypto/webcrypto:615:10)
    ... 4 lines matching cause stack trace ...
    at fulfilled (/Users/user/Projects/electron-quick-start/node_modules/node-opcua-pki/dist/lib/toolbox/without_openssl/create_self_signed_certificate.js:5:58) {
  [cause]: Error: error:0c0000be:ASN.1 encoding routines:OPENSSL_internal:WRONG_TAG
      at createPrivateKey (node:internal/crypto/keys:632:12)
      at Object.rsaImportKey (node:internal/crypto/rsa:227:21)
      at SubtleCrypto.importKey (node:internal/crypto/webcrypto:615:10)
      at derToPrivateKey (/Users/user/Projects/electron-quick-start/node_modules/node-opcua-crypto/dist/index.js:1807:31)
      at pemToPrivateKey (/Users/user/Projects/electron-quick-start/node_modules/node-opcua-crypto/dist/index.js:1829:10)
      at /Users/user/Projects/electron-quick-start/node_modules/node-opcua-pki/dist/lib/toolbox/without_openssl/create_self_signed_certificate.js:67:74
      at Generator.next (<anonymous>)
      at fulfilled (/Users/user/Projects/electron-quick-start/node_modules/node-opcua-pki/dist/lib/toolbox/without_openssl/create_self_signed_certificate.js:5:58) {
    opensslErrorStack: [
      'error:0c00009e:ASN.1 encoding routines:OPENSSL_internal:NESTED_ASN1_ERROR',
      'error:0c00009e:ASN.1 encoding routines:OPENSSL_internal:NESTED_ASN1_ERROR'
    ],
    library: 'ASN.1 encoding routines',
    function: 'OPENSSL_internal',
    reason: 'WRONG_TAG',
    code: 'ERR_OSSL_ASN1_WRONG_TAG'
  }
}
`

Steps to reproduce the behavior:

1. git clone https://github.com/electron/electron-quick-start
2. npm install node-opcua
3. edit main.js:
`const { app, BrowserWindow } = require('electron')
const path = require('node:path')

// NEW
const {
  OPCUAClient,
} = require('node-opcua');
//

function createWindow () {
  const mainWindow = new BrowserWindow({
    width: 800,
    height: 600,
    webPreferences: {
      preload: path.join(__dirname, 'preload.js')
    }
  })

  mainWindow.loadFile('index.html')
}

app.whenReady().then(() => {

  // NEW
  const ENDPOINT = "opc.tcp://opcuademo.sterfive.com:26543";
  const client = OPCUAClient.create();
  client.withSessionAsync(ENDPOINT, (session)=>{})
    .then(()=>{
      console.log("ok")
    }).catch((error)=>{
      console.error("ERROR!")
      console.log(error)
    });
  //

  createWindow()

  app.on('activate', function () {
    if (BrowserWindow.getAllWindows().length === 0) createWindow()
  })
})

app.on('window-all-closed', function () {
  if (process.platform !== 'darwin') app.quit()
})
`

The client connects without throwing an error.

gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: MacBook Air M1
- OS version: macOS Sonoma 14.3.1

- **node-opcua version:** :

2.124.0
"integrity": "sha512-VQ4G8RNICR0Itk4J7xDa0SU2TMhcVTeGIm8LbTHjTvbImo84dxb1PD8SSIEflonpKkSn12uEw3ATUxJgeMsq4Q==",

- **Node:**

  `node --version` =v18.19.0
  `console.log(process.versions.node)` =20.11.1
   electron = 30.0.0

erossignon, 2025-04-07:
May be you could raise the issue to the electron team to understand why nodejs SubtleCrypto module behalve differently.

closed 2025-04-07 by erossignon

mendoza-dev, 2026-01-07:
I have the exact same issue but by trying to test the "sample_server.js" and "sample_client_with_callback.ks" from the documentation:

server: https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_server.md

client: https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_client_callback.md

[image]