node-opcua/node-opcua #1374: Bind variable error:   : the getter must return a Variant or a StatusCode

AndreasHeine, 2024-06-18:
https://github.com/AndreasHeine/SampleServer-node-opcua/blob/f9b42caad83c2aacf5108490589a8b9b67f3c01b/src/serveraddressspace/serveraddressspace.ts#L300C1-L312C11
```ts
        const myVar = namespace.addVariable({
            browseName: 'MyVar',
            componentOf: showcaseAC,
            dataType: DataType.Double,
            value: {
                get: function (this) {
                    return new Variant({
                        value: myValue,
                        dataType: DataType.Double
                })},
            },
            eventSourceOf: showcaseAC,
        })
```

results in:
```
11:05:52.486Z :ua_variable_impl              :1617   Bind variable error:   : the getter must return a Variant or a StatusCode
 ...                                                value_check.constructor.name  Variant
11:05:52.493Z :address_space                 :1094  modelChangeTransaction  bindVariable : the value getter function returns a invalid result 
( expecting a Variant or a StatusCode !!!
 LOG:   error: Error:  Creating AddressSpace failed: Error:  bindVariable : the value getter function returns a invalid result ( expecting a Variant or a StatusCode !!!
```

AndreasHeine, 2024-06-18:
@erossignon has something changed in the last versions?

erossignon, 2024-06-18:
Probably a module mistmatch; please rebuild your node_modules and ensure node-opcua-* version consistency.

AndreasHeine, 2024-06-19:
@erossignon you where right ;) thanks!

closed 2024-06-19 by AndreasHeine