node-opcua/node-opcua #1375: Node-opcua produces error messages related to `MinSupportedSampleRate` on connection with a client when using a plain server

tetanw, 2024-06-26:
**Current behavior / Describe the bug**

Following error messages occurs when installing with the new version (this is one of them):
[image]

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Copy files from gist into a folder
3. `npm install`
4. `node index.js`
5. Connect an OPC UA client to the server (I checked with UaExpert)
6. See error messages 

**Expected behavior**

Not produce that specific error.

**Suggested solution**

Change dataType to `DataType.Double` in https://github.com/node-opcua/node-opcua/blob/81b467f284c41813ab8caf17f6d20322d4734fdc/packages/node-opcua-server/source/server_engine.ts#L1095. That seemed to work for me locally.

**Gist**

https://gist.github.com/tetanw/4fc5e6f4888c6be7c0232791c9bb37ee

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Lenovo Thinkstation P700
- OS version:

    - (x) Windows : version : Windows 10 Enterprise, 21H2
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______
   
- Description of the other OPCUA system I am trying to connect to: NA

- **node-opcua version:** 2.126.0

- **Node:**  v20.12.0

AndreasHeine, 2024-06-26:
according to the OPC UA Spec. -> https://reference.opcfoundation.org/Core/Part5/v104/docs/6.3.2
it has to be of Type "Duration" -> https://reference.opcfoundation.org/Core/Part3/v105/docs/8.13

so double sounds kind of valid!?

but actually its a subtype of "Double" -> https://reference.opcfoundation.org/Core/Part5/v105/docs/12.2.9.3#Table164

so it shall be Duration !!!

@erossignon

tetanw, 2024-06-26:
You are correct. Just checked, duration seems to work as well. If we change it to duration, then we need to make sure to change this line as well. Because that is then also wrong: https://github.com/node-opcua/node-opcua/blob/81b467f284c41813ab8caf17f6d20322d4734fdc/packages/node-opcua-server/source/server_engine.ts#L1022

erossignon, 2024-06-26:
I spotted this as well, it will be fixed in next build.

erossignon, 2024-06-27:
Should be fixed in node-opcua@2.217.0 
this bug only affected  node-opcua@2.216.0 due to a incorrect duplicated binding of the minSampleRate variable in the server engine.

closed 2024-06-27 by erossignon